lopsided98/nix-ros-overlay #268: rosPackages.noetic.laser-scan-matcher fails to build

Smona, 2023-06-27:
On latest master, I'm getting this error when trying to include `laser-scan-matcher` in `buildEnv`'s paths argument:

```
CMake Error at /nix/store/y2dbjs0hpkdmra1jcnvih069nq27j9ka-ros-noetic-catkin-0.8.10-r1/share/catkin/cmake/test/tests.cmake:18 (message):
  () is not available when tests are not enabled.  The CMake code should only
  use it inside a conditional block which checks that testing is enabled:

  if(CATKIN_ENABLE_TESTING)

    (...)

  endif()

Call Stack (most recent call first):
  CMakeLists.txt:71 (add_rostest)
```

I've attempted to enable testing for this package specifically but despite being a relatively experienced nix user, both ROS and overlays are pretty new to me, so I haven't been able to make much progress here.

lopsided98, 2023-06-27:
That's a bug in that package that should be fixed upstream. As the error says, the package needs to make calls to [`add_rostest`](https://github.com/CCNYRoboticsLab/scan_tools/blob/219c31965ff904c59072389d1873a21bcb433e81/laser_scan_matcher/CMakeLists.txt#L71) conditional on `CATKIN_ENABLE_TESTING`.

Smona, 2023-06-27:
Makes sense, I've [submitted an upstream PR](https://github.com/CCNYRoboticsLab/scan_tools/pull/90).

In the meantime I suppose I can just supply my fork in place of the overlay package. Thanks!

closed 2023-06-27 by Smona

Smona, 2023-06-27:
Welp, I've been defeated. Even after resolving that error, the package is still failing to build:

```
In file included from /nix/store/p0pv66dhnlgxvigh797jby2id4wsfaci-ros-noetic-roscpp-serialization-0.7.2-r1/include/ros/serialization.h:37,
                 from /nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/publisher.h:34,
                 from /nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/node_handle.h:32,
                 from /nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/ros.h:45:
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h: In instantiation of 'static const char* ros::message_traits::MD5Sum<M>::value() [with M = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]':
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h:227:102:   required from 'const char* ros::message_traits::md5sum() [with M = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]'
/nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/subscribe_options.h:89:49:   required from 'void ros::SubscribeOptions::initByFullCallbackType(const std::string&, uint32_t, const boost::function<void(P)>&, const boost::function<boost::shared_ptr<typename ros::ParameterAdapter<P>::Message>()>&) [with P = const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&; std::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int; typename ros::ParameterAdapter<P>::Message = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]'
/nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/node_handle.h:406:43:   required from 'ros::Subscriber ros::NodeHandle::subscribe(const std::string&, uint32_t, void (T::*)(M), T*, const ros::TransportHints&) [with M = const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&; T = scan_tools::LaserScanMatcher; std::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]'
/build/scan_tools-release-release-noetic-laser_scan_matcher-0.4.0-1/src/laser_scan_matcher.cpp:104:38:   required from here
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h:120:28: error: '__s_getMD5Sum' is not a member of 'std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >'
  120 |     return M::__s_getMD5Sum().c_str();
      |            ~~~~~~~~~~~~~~~~^~
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h: In instantiation of 'static const char* ros::message_traits::DataType<M>::value() [with M = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]':
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h:236:104:   required from 'const char* ros::message_traits::datatype() [with M = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]'
/nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/subscribe_options.h:90:53:   required from 'void ros::SubscribeOptions::initByFullCallbackType(const std::string&, uint32_t, const boost::function<void(P)>&, const boost::function<boost::shared_ptr<typename ros::ParameterAdapter<P>::Message>()>&) [with P = const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&; std::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int; typename ros::ParameterAdapter<P>::Message = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >]'
/nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/node_handle.h:406:43:   required from 'ros::Subscriber ros::NodeHandle::subscribe(const std::string&, uint32_t, void (T::*)(M), T*, const ros::TransportHints&) [with M = const std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&; T = scan_tools::LaserScanMatcher; std::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]'
/build/scan_tools-release-release-noetic-laser_scan_matcher-0.4.0-1/src/laser_scan_matcher.cpp:104:38:   required from here
/nix/store/yc5bwr4zfpnc8cqjx912yw89kafa765z-ros-noetic-roscpp-traits-0.7.2-r1/include/ros/message_traits.h:137:30: error: '__s_getDataType' is not a member of 'std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >'
  137 |     return M::__s_getDataType().c_str();
      |            ~~~~~~~~~~~~~~~~~~^~
/nix/store/p0pv66dhnlgxvigh797jby2id4wsfaci-ros-noetic-roscpp-serialization-0.7.2-r1/include/ros/serialization.h: In instantiation of 'static void ros::serialization::Serializer<T>::read(Stream&, typename boost::call_traits<T>::reference) [with Stream = ros::serialization::IStream; T = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >; typename boost::call_traits<T>::reference = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >&]':
/nix/store/p0pv66dhnlgxvigh797jby2id4wsfaci-ros-noetic-roscpp-serialization-0.7.2-r1/include/ros/serialization.h:163:22:   required from 'void ros::serialization::deserialize(Stream&, T&) [with T = std::shared_ptr<const pcl::PointCloud<pcl::PointXYZ> >; Stream = IStream]'
/nix/store/wa1ss4aapssimxawgknbkzf2mvdi2x65-ros-noetic-roscpp-1.16.0-r1/include/ros/subscription_callback_helper.h:136:21:   required from 'ros::VoidConstPtr ros::SubscriptionCallbackHelperT<P, Enabled>::deserialize(const ros::SubscriptionCallbackHelperDeserializeParams&) [with P = const std::shared_ptr<const pcl::P [... cut ...]
       > make[2]: *** [CMakeFiles/laser_scan_matcher.dir/build.make:76: CMakeFiles/laser_scan_matcher.dir/src/laser_scan_matcher.cpp.o] Error 1
       > make[1]: *** [CMakeFiles/Makefile2:2526: CMakeFiles/laser_scan_matcher.dir/all] Error 2
       > make: *** [Makefile:136: all] Error 2
       For full logs, run 'nix-store -l /nix/store/jn06f21hd9sfl59ayiff87vskhvm1616-ros-noetic-laser-scan-matcher-0.4.0-r1.drv'.
error: 1 dependencies of derivation '/nix/store/md2x16ri9dgsfq6ml754kys2c0i4dgzs-ros-env.drv' failed to build
```

lopsided98, 2023-06-27:
That may be a simple issue such as missing an include that used to be included through some other dependency. Unfortunately, I'm not sure what include might be missing.