lopsided98/nix-ros-overlay #273: ros2 run not working with python package executable not found error

Elston10, 2023-07-02:
[develop]
script_dir=$base/lib/my_py_pkg
[install]
install_scripts=$base/lib/my_py_pkg

my setup.cfg file

from setuptools import setup
package_name = 'my_py_pkg'

setup(
    name=package_name,
    version='0.0.0',
    packages=[package_name],
    data_files=[
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
    ],
    install_requires=['setuptools'],
    zip_safe=True,
    maintainer='victus',
    maintainer_email='victus@todo.todo',
    description='TODO: Package description',
    license='TODO: License declaration',
    tests_require=['pytest'],
    entry_points={
        'console_scripts': ['py_node=my_py_pkg.my_first_node:main'
        ],
    },
)

my setup.py file
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
class MyNode(Node):
    def __init__(self):
        super().__init__("my_node")
        self.get_logger().info("Initializing")
def main(args=None):
    rclpy.init(args=args)
    node=MyNode()
    rclpy.spin(node)
    rclpy.shutdown()

if __name__=="__main__":
    main()

my node program

Sorry to post it like this

hacker1024, 2023-07-02:
Duplicate of #230... Which you already know as you commented there first...

Elston10, 2023-07-02:
But I kinda didn't understand that it's why I posted differently

Elston10, 2023-07-02:
Truth is I've been looking at this error for days and i just can't figure it out

closed 2023-09-12 by lopsided98