lopsided98/nix-ros-overlay #277: Can I cross-compile packages?

jackyliu16, 2023-07-27:
Recently I've been trying to **port ROS1Master nodes to embedded bare boards** *(basically kernel-only boards)*, and I've tried to do a lot of dependency porting, but it doesn't work very well. 
Well, luckily, I have seen that Nix can develop software on the embedded board by cross-compilation. I want to try to generate a cross-compiled package through this overlay and move it to the board to use.

But when am I trying, I have on ideas about how to use this kind of operation in other machine, neither I'm simply trying to change the system argument in flakes or change example env to using pkgsCross, is it any example or recommendation of doing this ?

hacker1024, 2023-08-01:
I believe using `pkgsCross` or setting the Nixpkgs `hostSystem` should both work. Are you facing issues with these?

jackyliu16, 2023-08-01:
Could you be more specific? I don't quite understand the operation of setting hostPlatform.

 I have used two methods. One is to modify devShells directly so that it can generate the corresponding Shell through cross-compilation (although this attempt is obviously wrong). The other is to use `nix build .# pkgsCross.riscv64.rosPackages.noetic.ros-core`to try to generate the corresponding content through cross-compilation, but failure with this:
 ```
nix-repl> pkgsCross.riscv64.rosPackages.noetic.ros-core
error:
       … while calling the 'derivationStrict' builtin

         at //builtin/derivation.nix:9:12: (source not available)

       … while evaluating derivation 'ros-noetic-ros-core-riscv64-unknown-linux-gnu-1.5.0-r1'
         whose name attribute is located at /nix/store/sqc5viwwy1g1c9hrm93r9561af8fkqsl-source/pkgs/stdenv/generic/make-derivation.nix:303:7

       … while evaluating attribute 'propagatedBuildInputs' of derivation 'ros-noetic-ros-core-riscv64-unknown-linux-gnu-1.5.0-r1'

         at /nix/store/sqc5viwwy1g1c9hrm93r9561af8fkqsl-source/pkgs/stdenv/generic/make-derivation.nix:357:7:

          356|       depsHostHostPropagated      = lib.elemAt (lib.elemAt propagatedDependencies 1) 0;
          357|       propagatedBuildInputs       = lib.elemAt (lib.elemAt propagatedDependencies 1) 1;
             |       ^
          358|       depsTargetTargetPropagated  = lib.elemAt (lib.elemAt propagatedDependencies 2) 0;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: assertion '((builtins).hasAttr (stdenv).hostPlatform.system options)' failed

       at /nix/store/sqc5viwwy1g1c9hrm93r9561af8fkqsl-source/pkgs/development/compilers/sbcl/bootstrap.nix:49:1:

           48| in
           49| assert builtins.hasAttr stdenv.hostPlatform.system options;
             | ^
           50| stdenv.mkDerivation rec {
```

The `nix build .#noetic.ros-core` is useable, it seems will generated a file which contains all dependence...

hacker1024, 2023-08-01:
Ah. `pkgsCross` should work, but RISC-V is very new and you'll find many things don't support it yet. If you go to the SBCL bootstrap file shown in the error message, you'll see that it's missing RISC-V.

jackyliu16, 2023-08-01:
I see. Is there any other way to achieve this goal? The list of technical documents I found seems to be all about cross-compiling for ubuntu or debian (which speeds up compilation).

hacker1024, 2023-08-01:
If you want to use Nix, you'll need to either find a way to remove the dependency on broken packages, or fix the broken packages themselves. Both of these things can be done through overlays and overrides.

It's up to you to decide if it's worth your time trying to fix Nixpkgs rather than finding a different (non-Nix) solution.

jackyliu16, 2023-08-01:
Thank you very much for your advice, I'll try to fix nixpkgs first. If not working then continue to find non-Nix solution.

By the ways, did you think it's processible to add another toolchain into nix for doing this ? (this is a long-term plan）

closed 2023-08-03 by jackyliu16