lopsided98/nix-ros-overlay #285: Cannot install rclnodejs with npm: g++: fatal error: cannot execute '/nix/store/.../cc1plus’: execv: Argument list too long

hacker1024, 2023-08-06:
When attempting to install [rclnodejs](https://github.com/RobotWebTools/rclnodejs) with npm, `node-gyp` fails to build the native add-on.

```
$ nix-shell \
  -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/develop.tar.gz \
  --option extra-substituters 'https://ros.cachix.org' \
  --option extra-trusted-public-keys 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' \
  -E 'with import <nix-ros-overlay> { }; with pkgs.rosPackages.humble; (buildEnv { paths = [ ros-core nodejs ]; }).env'

$ npm i rclnodejs
...
npm ERR! g++: fatal error: cannot execute '/nix/store/1x1q5sqa0ilbi8fz7aayk02pjy5g7jhh-gcc-12.3.0/libexec/gcc/x86_64-unknown-linux-gnu/12.3.0/cc1plus': execv: Argument list too long
npm ERR! compilation terminated.
...
```

strace (`strace --follow-forks --trace=execve --string-limit 2048 -- npm i rclnodejs`) indicates that `execve` is being called with the entire contents of `AMENT_PREFIX_PATH`, which is too long for the kernel to handle in an argument list.

Other than patching the kernel to allow longer argument lists, is there any way to work around this issue?

hacker1024, 2023-08-06:
With further experimentation, it is possible to construct an environment with only the packages explicitly needed by rclnodejs. This keeps `AMENT_PREFIX_PATH` to a reasonable size.

```
nix-shell \
  -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/develop.tar.gz \
  --option extra-substituters 'https://ros.cachix.org' \
  --option extra-trusted-public-keys 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' \
  -E 'with import <nix-ros-overlay> { }; with pkgs.rosPackages.humble; (buildEnv {
    paths = [
      ros-environment
      rclcpp
      rcl-action
      rcl-lifecycle
      nodejs
    ];
  }).env'
```

This is not a sustainable solution, though, as rclnodejs could add more dependencies in the future. The official installation requirements just list ROS 2 as a dependency, which means at least `ros-core`.

lopsided98, 2023-08-06:
This type of error was the primary reason for creating the ROS specific `buildEnv` in the first place. With complex ROS 1 environments, `ROS_PACKAGE_PATH` and `CMAKE_PREFIX_PATH` would exceed the maximum size. When using `buildEnv` with ROS 1, only a single path for the entire environment ends up in these variables. Unfortunately, with ROS 2 the ament setup hooks add each package to `AMENT_PREFIX_PATH` anyway. I need to find a way to convince ament_cmake to treat the entire environment like a single workspace.

hacker1024, 2023-08-08:
> I need to find a way to convince ament_cmake to treat the entire environment like a single workspace.

I don't think `ament_cmake` is actually the problem here (unless you're referring to setup hooks). `node-gyp` does not seem to be using CMake at all in this instance, though I'm not sure what build system it is using. CMake is intelligent enough to only add necessary libraries to the compiler commands, rather than the entire contents of `AMENT_PREFIX_PATH`.

lopsided98, 2023-08-08:
Yes, I am talking about the setup hooks, or more specifically the `local_setup.sh` scripts in each package. There are executed within Nix using the `ament-cmake-core` setup hook, and normally responsible for constructing `AMENT_PREFIX_PATH`, but they have no way of knowing this is not necessary when inside `buildEnv`.

wentasah, 2024-01-02:
> Unfortunately, with ROS 2 the ament setup hooks add each package to `AMENT_PREFIX_PATH` anyway. I need to find a way to convince ament_cmake to treat the entire environment like a single workspace.

I think I found the way in #342. Not sure whether it fixes this issue, though.

lopsided98, 2024-03-23:
This is probably fixed by #342.

closed 2024-03-23 by lopsided98