lopsided98/nix-ros-overlay #286: navigation2 does not build

muellerbernd, 2023-08-08:
The navigation2 package does not build because `nav2_mppi_controller` is missing. In the last _rosdistro sync_ there is a dependency (`xtensor`) of `nav2_mppi_controller` listed as missing. But `xtensor` is available in nixpkg (see [search](https://search.nixos.org/packages?channel=23.05&show=xtensor&from=0&size=50&sort=relevance&type=packages&query=xtensor)). 

Because of this `rtabmap-ros` does also not compile. 

I want to contribute more to this repo is there a guideline on how to do so?

muellerbernd, 2023-08-08:
@hacker1024 thanks for this information. I will make a PR to rosdistro

lopsided98, 2023-08-08:
It looks like `nav2_mppi_controller` needs `xsimd` as well.

muellerbernd, 2023-08-09:
I created a [PR](https://github.com/ros/rosdistro/pull/38184) for rosdep to include `xsimd`

lopsided98, 2023-09-09:
`nav2-mppi-controller` is now generated but fails to build due to some non-obvious issue with the xsimd/xtensor headers. Do you need that specific package or were you just trying to use the `navigation2` meta package which happens to include that package?

muellerbernd, 2023-09-12:
I was trying to build `rtabmap-ros` and it seemed like navigation2 was a dependency.

hacker1024, 2023-09-24:
> `nav2-mppi-controller` is now generated but fails to build due to some non-obvious issue with the xsimd/xtensor headers. Do you need that specific package or were you just trying to use the navigation2 meta package which happens to include that package?

Now we need it! I'll look into it. A bunch of nav2 packages also need `-Wno-error=maybe-uninitialized`. I'll submit a PR with overrides sometime this week.

hacker1024, 2023-09-24:
Alright, got it all working:

## Issues

a) The following packages need to be compiled with `-Wno-error=maybe-uninitialized`: `nav2-behaviors`, `nav2-constrained-smoother`, `nav2-planner`, `nav2-smoother`, `nav2-waypoint-follower`, `dwb-critics`, `dwb-plugins`.
b) OMPL requires a fix upstream: https://github.com/ompl/ompl/pull/1101
c) xtensor needs upgrading, but xsimd needs downgrading. This handles the former so far: https://github.com/NixOS/nixpkgs/pull/225904

## Solutions

a) can be fixed here in an overlay, though it's probably best fixed upstream. That seems like a fair bit of effort, though.
b) can be applied in an overlay as well, though it's probably best to just wait for it to get merged.
c) is trickier. There's no easy solution yet. Apparently waiting on https://github.com/xtensor-stack/xtensor/issues/2721 to make a decision. For now, I just downgraded it globally.

doronbehar, 2023-09-27:
Hey all, feel free to suggest a solution to the `xtensor` update going on at #2721 that will add xsimd support to `xtensor`. An option I was also thinking about is creating a dedicated `xsimd10` attribute in `all-packages.nix`.

thearthur, 2024-04-29:
Has anyone gotten navigation2 to build? 
just checking if this is the same issue:
```
error: builder for '/nix/store/j161k57jy8mlgfrxybhqj6498n0ga61h-ros-iron-nav2-lifecycle-manager-1.2.7-r1.drv' failed with exit code 2;
       last 10 log lines:
       >    92 |     level = lvl;
       >       |     ~~~~~~^~~~~
       > /build/navigation2-release-release-iron-nav2_lifecycle_manager-1.2.7-1/src/lifecycle_manager.cpp: In member function 'void nav2_lifecycle_manager::LifecycleManager::CreateDiagnostic(diagnostic_updater::DiagnosticStatusWrapper&)':
       > /build/navigation2-release-release-iron-nav2_lifecycle_manager-1.2.7-1/src/lifecycle_manager.cpp:161:17: note: 'error_level' was declared here
       >   161 |   unsigned char error_level;
       >       |                 ^~~~~~~~~~~
       > cc1plus: all warnings being treated as errors
```

wentasah, 2024-04-30:
Some (if not all) these errors have already been fixed upstream, but probably are not yet released. The problem you mention can be addressed by [this commit](https://github.com/lopsided98/nix-ros-overlay/commit/7fc6eb501e16f6a213fb9de7e14d8498ddeda4c1) (in my branch). 

If you apply it, you will see the other problems, which are addressed by https://github.com/open-navigation/navigation2/commit/fbec0fa68a23e8257e1420075745aba0735b07c4. You can apply the changes from that commit similarly as above. The complication is that the commit is a squashed commit and you need to split it to individual packages by adding `stripLen` and `includes` arguments to `fetchpatch` as shown below:

```nix
          (self.fetchpatch {
            url = "https://github.com/open-navigation/navigation2/commit/fbec0fa68a23e8257e1420075745aba0735b07c4.patch";
            stripLen = 1;
            includes = [ "*/subdir..../*" ];
            hash = "";
          })
```
Beware that the value of those arguments influences the `hash`.

To be honest, I'm not sure whether it is worth the effort. Especially if the package will be released with these fixes sometimes soon, it may be better to wait :-/. However, I don't use these packages and don't know what's their release policy.

muellerbernd, 2025-05-14:
with the merge of #591 this can be closed.

closed 2025-05-14 by muellerbernd