lopsided98/nix-ros-overlay #288: how to setup autocomplete?

muellerbernd, 2023-08-09:
I am using this overlay for ros1 and ros2 development. For ros1 autocompletion was enable after sourcing `devel/setup.zsh` . For ros2 someting like `eval "$(register-python-argcomplete colcon)"` would be needed ([see this issue](https://github.com/ros2/ros2cli/issues/534)). 
What is the best way to setup autocompletion for ros1 and ros2? Can I setup autocompletion in the `mkshell` shell-hook?

hacker1024, 2023-08-09:
Yep, in the shell hook will work nicely. Be aware that `argcomplete` was only listed as a ROS 2 CLI dependency fairly recently, so anything older than Humble will need an override to add it.

muellerbernd, 2023-08-09:
and how would I do that? For ros2 I tried to use the following in my setup
```nix
  shell-hook = ''
    eval ${pkgs.python310Packages.argcomplete}/bin/register-python-argcomplete ros2
    eval ${pkgs.python310Packages.argcomplete}/bin/register-python-argcomplete colcon
    eval ${pkgs.python310Packages.argcomplete}/bin/register-python-argcomplete rosidl
  '';
```

hacker1024, 2023-08-09:
That looks good to me. I have a more developed setup here: https://github.com/hacker1024/nix-ros-workspace/blob/master/packages/ros/build-ros-workspace/default.nix

muellerbernd, 2023-08-15:
@hacker1024 I got it working using the setup script from your setup. But it only works with `bash` when I use `zsh` it does not work. Do you have any tips on how to get it working with `zsh`?

hacker1024, 2023-08-20:
> It works when I call eval "$(mk-workspace-shell-setup)" after I got into zsh.

That is expected. The `shellHook` does not run directly in Zsh, but rather in Bash before launching Zsh. This allows environment variables to be set, but does not allow any shell-specific setup to be done. This means that completion cannot be set up automatically.

liarokapisv, 2024-12-28:
Leaving a few tips on how to properly do this here.

As hacker1024 said, this is a general issue with non-bash shells integration with `shellHook`.
`shellHook` is actually run in a bash environment and then the environment is exported and then sourced from the 
active shell. This obviously breaks autocomplete functionality.

What you want to do is rely on native shell auto-complete script discovery. This is usually achieved through 
`XDG_DATA_DIRS` and `FPATH`. 

`stdenv`'s setup hook will take care to properly set the `XDG_DATA_DIRS` env variable from `$out/share` folders present in dependencies. This means that adding a simple derivation like below will properly create the directories and set the relevant env variables:

```nix
  (pkgs.runCommand "ros-autocompletions" { } ''
    for dir in {bash-completion/completions,zsh/site-functions}; do
        mkdir -p $out/share/$dir
        for program in {ros2,colcon,rosidl}; do
            ${pkgs.python3.pkgs.argcomplete}/bin/register-python-argcomplete $program > $out/share/$dir/_$program
        done
    done
  '')
```

An issue is that this discovery process is usually only performed once when starting the shell.
Explicitly calling `compinit` after entering a nix shell (or after a direnv reload) will take care of this.

You can avoid having to call `compinit` by relying on plugins like https://github.com/BronzeDeer/zsh-completion-sync.

muellerbernd, 2024-12-28:
@liarokapisv Thanks for your reply on this issue. Can you give a minimal working example?

liarokapisv, 2024-12-28:
Basically, just put the above derivation in your `mkShell`'s packages. Below is an example flake:

```nix
{
  inputs = {
    ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs.follows = "ros-overlay/nixpkgs";
  };

  outputs = { self, nixpkgs, ros-overlay, ... }:
    let
      forAllSystems = (f: nixpkgs.lib.genAttrs [ "x86_64-linux" ] (system:
        let
          pkgs = import nixpkgs {
            inherit system;
            overlays = [ self.overlays.default ];
          };
          ros-distro = pkgs.rosPackages.jazzy;
        in
        f {
          inherit system pkgs ros-distro;
        })
      );
    in
    {
      overlays.default = ros-overlay.overlays.default;

      devShells = forAllSystems
        ({ pkgs, ros-distro, ... }:
          {
            default =
              pkgs.mkShell
                {
                  name = "ros-shell";
                  packages = with pkgs; [
                    ros-distro.ros-core
                    colcon
                    (pkgs.runCommand "ros-autocompletions" { } ''
                      for dir in {bash-completion/completions,zsh/site-functions}; do
                          mkdir -p $out/share/$dir
                          for program in {ros2,colcon,rosidl}; do
                              ${pkgs.python3.pkgs.argcomplete}/bin/register-python-argcomplete $program > $out/share/$dir/_$program
                          done
                      done
                    '')
                  ];
                };
          });
    };
}
```

Then just add the [zsh-completion-sync](https://github.com/BronzeDeer/zsh-completion-sync) plugin to your zsh config.

muellerbernd, 2024-12-28:
> Basically, just put the above derivation in your `mkShell`'s packages. Below is an example flake:
>
> ```nix
> {
>   inputs = {
>     ros-overlay.url = "github:lopsided98/nix-ros-overlay/master"; [...]

Works for me. Thanks for this tip.

GeFrIt42, 2025-01-11:
hi,
if you use pkgs.mkShell you could just add the shellHook, so when the shell starts it will initialize the autocomplete for you.
I have tested this with the flake.nix on the root of this repo.

`
          shellHook = ''
            eval "$(register-python-argcomplete ros2)"
            eval "$(register-python-argcomplete colcon)"
          '';

`

liarokapisv, 2025-01-11:
> hi, if you use pkgs.mkShell you could just add the shellHook, so when the shell starts it will initialize the autocomplete for you. I have tested this with the flake.nix on the root of this repo. [...]

This only works if you are using bash as a shell.

GeFrIt42, 2025-01-11:
> This only works if you are using bash as a shell.

You are right:
- It is not clear to me how  **argcomplete** is implemented, but it seems that the registered completion do not get carried to the child shells, which means that the register commands need to be executed per shell.
- At the moment there is no way to tell to nix shell to use a different shell other than bash [link](https://discourse.nixos.org/t/using-nix-develop-opens-bash-instead-of-zsh/25075) . in order to swap shell you have to start it from bash in the shellHook.

I see also that [The previous solutions](https://github.com/lopsided98/nix-ros-overlay/issues/288#issuecomment-2564302147) have a dependencies to the **zsh-completion-sync** that is not part of default omz plugins, it requires a little more configuration on the system level.

Would it be a good thing to have the shellHook solution merged in the [README.md](https://github.com/lopsided98/nix-ros-overlay/blob/develop/README.md) ? 
It is not perfect, but it is more self contained and short, it would allow beginners to have the completion.

liarokapisv, 2025-01-11:
If we add the autocomplete support to the appropriate packages (by adding the `argcomplete` files in the output), then it should automatically work for bash users without them having to add anything to the `shellHook.` Zsh / fish users can optionally add the corresponding sync package to their configuration to also take advantage of the support.

Pleune, 2025-02-24:
> Basically, just put the above derivation in your `mkShell`'s packages. Below is an example flake:
> 
> {
>   inputs = {
>     ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
>     nixpkgs.follows = "ros-overlay/nixpkgs";
>   };
> 
>   outputs = { self, nixpkgs, ros-overlay, ... }:
>     let
>       forAllSystems = (f: nixpkgs.lib.genAttrs [ "x86_64-linux" ] (system:
>         let
>           pkgs = import nixpkgs {
>             inherit system;
>             overlays = [ self.overlays.default ];
>           };
>           ros-distro = pkgs.rosPackages.jazzy;
>         in
>         f {
>           inherit system pkgs ros-distro;
>         })
>       );
>     in
>     {
>       overlays.default = ros-overlay.overlays.default;
> 
>       devShells = forAllSystems
>         ({ pkgs, ros-distro, ... }:
>           {
>             default =
>               pkgs.mkShell
>                 {
>                   name = "ros-shell";
>                   packages = with pkgs; [
>                     ros-distro.ros-core
>                     colcon
>                     (pkgs.runCommand "ros-autocompletions" { } ''
>                       for dir in {bash-completion/completions,zsh/site-functions}; do
>                           mkdir -p $out/share/$dir
>                           for program in {ros2,colcon,rosidl}; do
>                               ${pkgs.python3.pkgs.argcomplete}/bin/register-python-argcomplete $program > $out/share/$dir/_$program
>                           done
>                       done
>                     '')
>                   ];
>                 };
>           });
>     };
> }
> Then just add the [zsh-completion-sync](https://github.com/BronzeDeer/zsh-completion-sync) plugin to your zsh config.

Has something changed with Nix since this? This seems like it would work, but the autocompletions package won't get added to my `XDG_DATA_DIRS`. Although somehow patchelf and lndir have share folders that are added...

closed 2025-09-14 by muellerbernd