lopsided98/nix-ros-overlay #292: Is it possible to keep generating derivations for EOL distros?

hacker1024, 2023-08-21:
Our organisation is in the process of migrating from Foxy to Humble. While Foxy packages are not getting updated anymore, Super Flore can still benefit from improvements to rosdep.

Would it be possible to keep generating derivations for EOL distros, but perhaps not include them in the binary cache to save resources?

lopsided98, 2023-09-08:
From a technical perspective, superflore automatically stops updating the distro when it goes EOL. I've never looked into how it determines when a distro is EOL.

I'm also not sure this is something I want to start doing. The distro has to go away at some point, and the logical time for that seems to be the official EOL date that was published years in advance. It is also not simply a matter of continuing to update the distro with superflore, I also have to put in work (which tends to increase as the distro gets older) to maintain the distro as nixpkgs and the rest of the overlay changes, otherwise it will quickly end up broken and useless.

hacker1024, 2023-09-12:
Fair enough. For anyone else curious, the patch removing a distro can be reverted, though this can of course break in the future.

```nix
pkgs.applyPatches {
  src = pkgs.fetchFromGitHub {
    owner = "lopsided98";
    repo = "nix-ros-overlay";
    # ...
  };
  patches = [
    # Bring back Foxy
    (pkgs.fetchpatch {
      url = "https://github.com/lopsided98/nix-ros-overlay/commit/113129c9daf2edfcb4a12593a60e918116dee671.patch";
      revert = true;
      hash = "sha256-73+rJj9c34cHQkDqxHzSe0HDAf+3zMJj4g+04y25xAU=";
    })
  ];
}
```

lopsided98, 2023-09-12:
I used to try to keep old distros around for ~6 months after EOL (without regenerating them, which is different from what you were originally asking). I removed melodic soon after EOL because enough core packages were broken that it seemed pretty useless and I didn't want to put maintenance effort into an EOL distro. I don't remember how broken Foxy was, but I might have been a little quick in removing it.

Thinking about this more, if you want to take over maintaining foxy, I'll revert the removal and you can submit PRs for the changes you want. Since superflore isn't generating the distro any more, you can modify the generated files as well.

hacker1024, 2023-09-13:
Thanks. Foxy is not very broken yet, so keeping it is still quite useful for the time being. My main interests in keeping it in the repository were to:

a) Prevent it breaking from any future structural changes
b) Allow it to benefit from improvements to Super Flore and rosdep
c) Keep it in the binary cache

In its current form, a and c still stand.

As for submitting PRs, I'm not sure that I'll really be making that many - I'll submit the overrides we have so far, but as we're migrating away from Foxy, we'll be tending to upgrade to avoid problems rather than patching Foxy more.

lopsided98, 2023-09-14:
I think your best bet is to make a PR to apply the rosdep updates manually to the generated files. It's just temporary life support at this point; I don't think we need to automate it.

closed 2023-09-17 by lopsided98