lopsided98/nix-ros-overlay #299: humble.rviz2 not working, Unable to create glx visual

etherswangel, 2023-09-07:
Do I need to install nixgl? qt5.base can't be installed manually
```
> Error: detected mismatched Qt dependencies:
       >     /nix/store/7m1px7m2zvidm7w3s93qjgvpkvdzwp8c-qtbase-5.15.9-dev
       >     /nix/store/rvbhi2xv6kjzrrdvz0swdvcp8v9sw91b-qtbase-5.15.10-dev
       For full logs, run 'nix log /nix/store/cva9nsday84fk10pj763dmi8lkcvfqbl-nix-shell-env.drv'.
```

my `flake.nix`:
```nix
rec {
  description = "ROS";

  inputs = {
    nixpkgs.url = github:nixos/nixpkgs/nixpkgs-unstable;
    rospkgs.url = github:lopsided98/nix-ros-overlay;
  };

  outputs = { self, nixgl, nixpkgs, rospkgs }:
  let
    distro = "humble";
    system = "x86_64-linux";
    pkgs = nixpkgs.legacyPackages.${system};
    ros  = rospkgs.legacyPackages.${system}.${distro};
  in
  {
    devShells.${system}.default =
      pkgs.mkShell
        {
          nativeBuildInputs = [
            pkgs.glibcLocales
            (ros.buildEnv {
              paths = with ros; [
                ros-core
                rviz2
              ]; })
          ];
        };
  };
}

```

rviz2 output:
```bash
$ rviz2
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Unable to create glx visual
[ERROR] [1694091880.851399154] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /build/rviz-release-release-humble-rviz_ogre_vendor-11.2.7-1/build/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)

...

[ERROR] [1694091880.854538968] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
fish: Job 1, 'rviz2' terminated by signal SIGABRT (Abort)
```

wentasah, 2023-09-07:
There seems to be at least two problems:

1. If you use non-NixOS Linux distro, you should use nixGL or something similar. This is the command that I use. It works on Wayland desktops:

       QT_QPA_PLATFORM=xcb nix run --impure github:guibou/nixGL -- rviz2

2. Your flake uses two different versions of `nixpkgs`. One version is a dependency of `nix-ros-overlay`, the second is your `nixpkgs` input. The best is, in most cases, to use the same version as nix-ros-overlay, i.e., instead of `pkgs = nixpkgs.legacyPackages.${system};` use `pkgs = nix-ros.inputs.nixpkgs.legacyPackages.${system};`

etherswangel, 2023-09-07:
Thanks for your reply! 

I fixed 2 and tried nixGL, but

```bash
$ QT_XCB_GL_INTEGRATION=none nix run --impure --override-input nixpkgs nixpkgs/nixos-unstable github:guibou/nixGL -- rviz2
Unable to create glx visual
[ERROR] [1694099755.698413320] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /build/rviz-release-release-humble-rviz_ogre_vendor-11.2.7-1/build/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
[ERROR] [1694099755.698475136] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /build/rviz-release-release-humble-rviz_ogre_vendor-11.2.7-1/build/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
...
```

This might be related to Nvidia driver https://github.com/NixOS/nixpkgs/issues/85866#issuecomment-1003618138

wentasah, 2023-09-07:
I have the same error if I don't have `QT_QPA_PLATFORM=xcb` in my environment. Try adding it as in my example above.

etherswangel, 2023-09-08:
I've exported `QT_QPA_PLATFORM=xcb` actually, `glxinfo` also fails so it's probably not about QT or ros-overlay.

```bash
$ nix run --impure --override-input nixpkgs nixpkgs/nixos-unstable github:guibou/nixGL -- gl
xinfo
warning: not writing modified lock file of flake 'github:guibou/nixGL':
• Updated input 'nixpkgs':
    'github:nixos/nixpkgs/441dc5d512153039f19ef198e662e4f3dbb9fd65' (2022-08-15)
  → 'github:NixOS/nixpkgs/0bffda19b8af722f8069d09d8b6a24594c80b352' (2023-09-06)
name of display: :0
Error: couldn't find RGB GLX visual or fbconfig
```

wentasah, 2023-09-08:
In this case I'd need more information to help. What's your Linux distribution, which GPU (and drivers) do you have and which desktop environment do you use? Also, does accelerated graphics (e.g. glxgears) work in the underlaying distro?

etherswangel, 2023-09-08:
1. Void Linux
2. `AMD Radeon Vega Mobile Series (Ryzen 5900hx)` and `Nvidia GeForce RTX 3070 Mobile / Max-Q (Driver Version: 535.98)`
3. i3wm
4. works fine

Thank you very much for your help. Should I raise an issue on guibou/nixGL instead of this? 

```bash
$ nix-shell -p nix-info --run "nix-info -m"
 - system: `"x86_64-linux"`
 - host os: `Linux 6.3.13_1, Void, noversion, nobuild`
 - multi-user?: `no`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.17.0`
```

```bash
$ glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
1544 frames in 5.0 seconds = 308.755 FPS
1501 frames in 5.0 seconds = 300.012 FPS
```

```bash
$ glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon Graphics (renoir, LLVM 15.0.7, DRM 3.52, 6.3.13_1) (0x1638)
    Version: 23.1.3
    Accelerated: yes
    Video memory: 512MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 70 MB, largest block: 70 MB
    VBO free aux. memory - total: 7503 MB, largest block: 7503 MB
    Texture free memory - total: 70 MB, largest block: 70 MB
    Texture free aux. memory - total: 7503 MB, largest block: 7503 MB
    Renderbuffer free memory - total: 70 MB, largest block: 70 MB
    Renderbuffer free aux. memory - total: 7503 MB, largest block: 7503 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 512 MB
    Total available memory: 8219 MB
    Currently available dedicated video memory: 70 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon Graphics (renoir, LLVM 15.0.7, DRM 3.52, 6.3.13_1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 23.1.3
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 23.1.3
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 23.1.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

etherswangel, 2024-08-03:
I no longer work on a nvidia machine, closing this issue

closed 2024-08-03 by etherswangel