lopsided98/nix-ros-overlay #301: rqt-robot-plugins is pulling in multiple python-qt-binding packages and failing to build

kjeremy, 2023-09-08:
I threw the following under `examples/` and did a `nix-build ./ros-blah.nix`:

```nix
{ pkgs ? import ../. {} }:
with pkgs;
with rosPackages.noetic;

mkShell {
  nativeBuildInputs = [
    (buildEnv {
      paths = [
        rqt-robot-plugins
      ];
    })
  ];
}
```

I get this error:

```shellsession
building
error: collision between `/nix/store/rhl1f35rrqgm6kz8yjvijdxb17x3h0fv-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py' and `/nix/store/hk76klg7vwy20n2kpagmp16d3f508kp3-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py'
error: builder for '/nix/store/ivdccsrwk3m9pa18mbb18v4ds62lxr21-ros-env.drv' failed with exit code 25;
       last 2 log lines:
       > building
       > error: collision between `/nix/store/rhl1f35rrqgm6kz8yjvijdxb17x3h0fv-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py' and `/nix/store/hk76klg7vwy20n2kpagmp16d3f508kp3-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py'
       For full logs, run 'nix log /nix/store/ivdccsrwk3m9pa18mbb18v4ds62lxr21-ros-env.drv'.
error: 1 dependencies of derivation '/nix/store/advq9q3s7hhsr4q0cib548wznsgyga11-nix-shell.drv' failed to build
```

kjeremy, 2023-09-12:
This is from a much larger derivation:

```nix
rqtScript =
    let
      env = (with rosDistro; buildEnv {
        name = "configured-rqt-env";
        paths = [
          ros-environment
          rospack

          rqt-gui-bin
          # rqt-robot-plugins seems to pull in conflicting versions of python-qt5 and python-qt-binding
          # rqt-common-plugins does something similar.
          # https://github.com/lopsided98/nix-ros-overlay/issues/301
          #rqt-common-plugins
          #rqt-robot-plugins
        ] ++ extraMsgs;
      }).override ({ paths, ... }: {
        paths = paths ++ (with rosDistro.python.pkgs; [
          # many plugins require these
          pycryptodomex
          python-gnupg
          defusedxml

          # rqt-bag
          pillow
          pycairo
          cairocffi
          cffi

          # process monitor
          psutil

          # rqt-graph
          pydot

          # rqt-plot requires one of these:
          pyqtgraph
          matplotlib

          # navigation viewer
          pyopengl

          # web
          # JAK: Disable to avoid collisions with gazebo_sims. Maybe there's an alternative for this.
          # pyqt5_with_qtwebkit
        ]
        # We no longer install dev outputs by default, which means we don't get
        # propagatedBuildInputs from non-ROS packages in buildEnv.  Propagate
        # pyqt's PyQt5.sip dependency manually.
        #++ pyqt5_with_qtwebkit.propagatedBuildInputs
        );
      });

      defaultArgs = if (defaultRqtConfig == null) then [ ] else [ defaultRqtConfig ];
    in
    writeConfiguredScriptBin "configured-rqt" ''
      # Fix up the python path so rqt can find its plugins
      export PYTHONPATH="${env}/${rosDistro.python.sitePackages}''${PYTHONPATH:+:$PYTHONPATH}"

      # If no args are explicitly given, use a platform-specific config file
      if [[ $# == 0 ]]; then set -- ${lib.escapeShellArgs defaultArgs}; fi

      exec ${env}/bin/rqt "$@"
    '';
```

That makes it's way into a `.desktop` file eventually and is installer in a per-robot development environment.

wentasah, 2026-08-29:
This seems related to Noetic, which is EOL. Closing.

closed 2026-08-29 by wentasah