lopsided98/nix-ros-overlay #305: grid-map does not build

2m, 2023-09-16:
First of all I would like to give a big thank you for this project. :) I am new to the ROS world and was having troubles installing noetic packages on my ArchLinux system. I was really glad when I found this repository. 

I am working on a flake for OpenMower project and noticed that one of the dependencies `grid-map` does not build with the following error:

```
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /nix/store/fcz98r5q08naprpcb2gm1s059iz5msly-ros-noetic-grid-map-loader-1.6.4-r>
  Project 'grid_map_loader' specifies 'include' as an include dir, which is
  not found.  It does not exist in
  '/nix/store/fcz98r5q08naprpcb2gm1s059iz5msly-ros-noetic-grid-map-loader-1.6.4-r1/include'.
  Check the issue tracker 'http://github.com/anybotics/grid_map/issues' and
  consider creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /nix/store/y0nqm2cxrmrig4gl98438k0gpm20bpjk-ros-noetic-catkin-0.8.10-r1/share/catkin/cmake/>
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
error (ignored): error: interrupted by the user
```

Not sure if its a bug in the package or something else is going on.

lopsided98, 2023-09-16:
This is a CMake bug in the package, which normally goes unnoticed in a traditional Linux distribution. `grid_map_loader` calls [`include_directories(include ...)`](https://github.com/ANYbotics/grid_map/blob/0b889ca222a4233bb533cec51d3bca062f76aaf8/grid_map_loader/CMakeLists.txt#L40C10-L40C10), which makes the include directory part of the public interface (i.e. downstream packages will try to add that directory to their include directories). The problem is that the package doesn't actually install any headers, so that directory isn't created in the install prefix. On traditional Linux distributions all packages are installed into the same tree and therefore some other package will have created the directory (which I think ends up being something like `/opt/ros/noetic/include`), so downstream packages don't run into any problems.

The fix is to replace the `include_directories(include)` call with `target_include_directories(${PROJECT_NAME} PRIVATE include)` to prevent the directory from becoming part of the public interface.

2m, 2023-09-16:
Thank you for the explanation. I will create a PR to the upstream project.

2m, 2023-09-21:
While the PR has not yet been merged, I was able to create a flake witch pulls grid_map_loader from my PR (https://github.com/2m/open_mower_ros/blob/fix/nix-flake-2m/flake.nix#L14-L24)

However if I add grid_map which depends on grid_map_demos which depend on grid_map_loader, it still takes the original non-patched version. I was not able to figure out how to patch grid_map_loader transitively. Whats the trick here? :)

wentasah, 2026-08-29:
This is related to Noetic, which is EOL. Closing.

closed 2026-08-29 by wentasah