lopsided98/nix-ros-overlay #311: OMPL on Rolling fails to build

werner291, 2023-10-03:
Hi, unfortunately this flake doesn't build:

```nix
{
  description = "My ROS Project Build Environment";
  nixConfig.bash-prompt = "[ros] ";
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-unstable";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nix-ros-overlay }:
    let
      system = "x86_64-linux";
      pkgs = import nixpkgs { inherit system; overlays = [ nix-ros-overlay.overlays.default ]; };
    in
    {
      devShells.x86_64-linux.default = pkgs.mkShell {
        name = "My ROS Project Build Environment";
        buildInputs = with pkgs; with pkgs.rosPackages.rolling; [
#           moveit-core
# #           eigen3
#           jsoncpp
#           fcl
#           boost
#           bullet
#           ompl
#           range-v3
#           qhull
#           cgal
#           ament-cmake
#           rclcpp
#           moveit-msgs
#           moveit-ros-planning
          moveit-planners-ompl
#           shape-msgs
        ];
      };
    };
}
```

Here's my lockfile: [flake.lock](https://github.com/lopsided98/nix-ros-overlay/files/12796840/flake.lock.txt)

It fails with this error:

```
error: builder for '/nix/store/28yl58cy37igw9wzwgz4qs83dj7f1hji-ros-rolling-ompl-1.5.2-r3.drv' failed with exit code 1;
       last 10 log lines:
       > checking for references to /build/ in /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3...
       > gzipping man pages under /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/share/man/
       > patching script interpreter paths in /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3
       > stripping (with command strip and flags -S -p) in  /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/lib
       > Broken paths found in a .pc file! /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/lib/pkgconfig/ompl.pc
       > The following lines have issues (specifically '//' in paths).
       > 4:libdir=${prefix}//nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/lib
       > It is very likely that paths are being joined improperly.
       > ex: "${prefix}/@CMAKE_INSTALL_LIBDIR@" should be "@CMAKE_INSTALL_FULL_LIBDIR@"
       > Please see https://github.com/NixOS/nixpkgs/issues/144170 for more details.
       For full logs, run 'nix log /nix/store/28yl58cy37igw9wzwgz4qs83dj7f1hji-ros-rolling-ompl-1.5.2-r3.drv'.
error: 1 dependencies of derivation '/nix/store/dvhm9c6gsfy0y0pw3jcaqiwbrn6nknig-ros-rolling-moveit-planners-ompl-2.8.0-r2.drv' failed to build
error: 1 dependencies of derivation '/nix/store/xspvdan026gqrycagfy5dffnyajlg75l-My-ROS-Project-Build-Environment-env.drv' failed to build
```

werner291, 2023-10-03:
Here's the file it's complaining about:

```
cat /nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/lib/pkgconfig/ompl.pc
# This file was generated by CMake for ompl
prefix=/nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3
exec_prefix=${prefix}
libdir=${prefix}//nix/store/4jzs6xfbgs9xk71nlpz1wljvr1apwpj3-ros-rolling-ompl-1.5.2-r3/lib
includedir=${prefix}/include/ompl-1.5

Name: ompl
Description: The Open Motion Planning Library
Version: 1.5.2
Requires: eigen3 ode
Libs: -L${libdir} -lompl -L/nix/store/hz6hdj4d12da362chq3r8hhznlj6qnlj-flann-1.9.1/lib -L/nix/store/w208hh73hwj2nbn2yqmxiy1vplxnkz8d-lz4-1.9.4/lib -L/nix/store/xxydgbry9aqfvigg7qkzgb80ppi1qvb7-ode-0.16.4/lib -lboost_serialization -lboost_filesystem -lboost_system
Cflags: -std=c++14 -I${includedir} -I/nix/store/4yjad2mn8xnr1gc20bisyllgml8pl08g-boost-1.81.0-dev/include
```

werner291, 2023-10-03:
Commenting this to link the issues: https://github.com/NixOS/nixpkgs/issues/144170

werner291, 2023-10-03:
Just ompl from Rolling and 23.05 instead of unstable; same issue:
```
{
  description = "My ROS Project Build Environment";
  nixConfig.bash-prompt = "[ros] ";
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-23.05";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nix-ros-overlay }:
    let
      system = "x86_64-linux";
      pkgs = import nixpkgs { inherit system; overlays = [ nix-ros-overlay.overlays.default ]; };
    in
    {
      devShells.x86_64-linux.default = pkgs.mkShell {
        name = "My ROS Project Build Environment";
        buildInputs = with pkgs; with pkgs.rosPackages.rolling; [
#           moveit-core
# #           eigen3
#           jsoncpp
#           fcl
#           boost
#           bullet
#           ompl
#           range-v3
#           qhull
#           cgal
#           ament-cmake
#           rclcpp
#           moveit-msgs
#           moveit-ros-planning
#           moveit-planners-ompl
ompl
#           shape-msgs
        ];
      };
    };
}
```

```
warning: Git tree '/home/werner/workspace/Multigoal-Orchard-Drone-Planning-Library' is dirty
warning: updating lock file '/home/werner/workspace/Multigoal-Orchard-Drone-Planning-Library/flake.lock':
• Updated input 'nixpkgs':
    'github:nixos/nixpkgs/f5892ddac112a1e9b3612c39af1b72987ee5783a' (2023-09-29)
  → 'github:nixos/nixpkgs/32dcb45f66c0487e92db8303a798ebc548cadedc' (2023-09-30)
warning: Git tree '/home/werner/workspace/Multigoal-Orchard-Drone-Planning-Library' is dirty
error: builder for '/nix/store/c8nawz7rzv9zz16wdi4f1ywpc5y12dh9-ros-rolling-ompl-1.5.2-r3.drv' failed with exit code 1;
       last 10 log lines:
       > checking for references to /build/ in /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3...
       > gzipping man pages under /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/share/man/
       > patching script interpreter paths in /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3
       > stripping (with command strip and flags -S -p) in  /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib
       > Broken paths found in a .pc file! /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib/pkgconfig/ompl.pc
       > The following lines have issues (specifically '//' in paths).
       > 4:libdir=${prefix}//nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib
       > It is very likely that paths are being joined improperly.
       > ex: "${prefix}/@CMAKE_INSTALL_LIBDIR@" should be "@CMAKE_INSTALL_FULL_LIBDIR@"
       > Please see https://github.com/NixOS/nixpkgs/issues/144170 for more details.
       For full logs, run 'nix log /nix/store/c8nawz7rzv9zz16wdi4f1ywpc5y12dh9-ros-rolling-ompl-1.5.2-r3.drv'.
error: 1 dependencies of derivation '/nix/store/wj6svbwnjvy1gjk0xy4svdbvmibl86c0-My-ROS-Project-Build-Environment-env.drv' failed to build
```

werner291, 2023-10-03:
`pkgs.rosPackages.foxy.ompl` is affected too.

werner291, 2023-10-03:
Related? https://github.com/ompl/ompl/pull/1101

werner291, 2023-10-03:
Yup, that PR fixes the issue. This flake builds correctly:

```nix
{
  description = "My ROS Project Build Environment";
  nixConfig.bash-prompt = "[ros] ";
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-23.05";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nix-ros-overlay }:
  let
    system = "x86_64-linux";
    pkgs = import nixpkgs { inherit system; overlays = [ nix-ros-overlay.overlays.default ]; };
  in
  {
    devShells.x86_64-linux.default = pkgs.mkShell {
      name = "My ROS Project Build Environment";
      buildInputs = with pkgs; with pkgs.rosPackages.rolling; [
        (pkgs.rosPackages.rolling.ompl.overrideAttrs(oldAttrs: {
          src = builtins.fetchGit {
            url = "https://github.com/hacker1024/ompl.git";
            rev = "1ddecbad87b454ac0d8e1821030e4cf7eeff2db2";
            allRefs = true;
          };
        }))
      ];
    };
  };
}
```

werner291, 2023-10-03:
Add that to the overlays in {
  description = "My ROS Project Build Environment";
  nixConfig.bash-prompt = "[ros] ";
  inputs = {
    nixpkgs.url = "github:nixos/nixpkgs/nixos-23.05";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, nix-ros-overlay }:
    let
      system = "x86_64-linux";
      pkgs = import nixpkgs { inherit system; overlays = [ nix-ros-overlay.overlays.default ]; };
    in
    {
      devShells.x86_64-linux.default = pkgs.mkShell {
        name = "My ROS Project Build Environment";
        buildInputs = with pkgs; with pkgs.rosPackages.rolling; [
          (pkgs.rosPackages.rolling.ompl.overrideAttrs(oldAttrs: {
            src = builtins.fetchGit {
        url = "https://github.com/hacker1024/ompl.git";
        rev = "1ddecbad87b454ac0d8e1821030e4cf7eeff2db2";
        allRefs = true;
        };
      }))
        ];
      };
    };
}

hacker1024, 2023-10-04:
It's often more convenient to apply the commit as a patch - that way, the sources will always be up-to-date.

```
ompl.overrideAttrs ({ patches ? [ ], ... }: {
  patches = [
    # Use full install paths for pkg-config
    (self.fetchpatch {
      url = "https://github.com/hacker1024/ompl/commit/1ddecbad87b454ac0d8e1821030e4cf7eeff2db2.patch";
      hash = "sha256-sAQLrWHoR/DhHk8TtUEy8E8VXqrvtXl2BGS5UvElJl8=";
    })
  ];
});
```

werner291, 2023-10-04:
Oh, a *patch*, that's cool.

Here's the overlay that pulls in your PR commit, just because I already had it:

```
# See: https://github.com/lopsided98/nix-ros-overlay/issues/311

self: super:

let
  myOmpl = super.rosPackages.rolling.ompl.overrideAttrs (oldAttrs: {
    src = builtins.fetchGit {
      url = "https://github.com/hacker1024/ompl.git";
      rev = "1ddecbad87b454ac0d8e1821030e4cf7eeff2db2";
    };
  });
in
{
  rosPackages = super.rosPackages // {
    rolling = super.rosPackages.rolling // {
      ompl = myOmpl;
    };
  };
}
```

werner291, 2023-10-04:
Wait, how do I use that?

I'm actually trying to use `moveit-planners-ompl` which depends on ompl. I figured that applying the overlay like I'm doing would override the ompl being pulled in here, but... it isn't?

```
error: builder for '/nix/store/c8nawz7rzv9zz16wdi4f1ywpc5y12dh9-ros-rolling-ompl-1.5.2-r3.drv' failed with exit code 1;
       last 10 log lines:
       > checking for references to /build/ in /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3...
       > gzipping man pages under /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/share/man/
       > patching script interpreter paths in /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3
       > stripping (with command strip and flags -S -p) in  /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib
       > Broken paths found in a .pc file! /nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib/pkgconfig/ompl.pc
       > The following lines have issues (specifically '//' in paths).
       > 4:libdir=${prefix}//nix/store/8kxzs91f4zsnb0vwjbwwc1zcpbrkd2sv-ros-rolling-ompl-1.5.2-r3/lib
       > It is very likely that paths are being joined improperly.
       > ex: "${prefix}/@CMAKE_INSTALL_LIBDIR@" should be "@CMAKE_INSTALL_FULL_LIBDIR@"
       > Please see https://github.com/NixOS/nixpkgs/issues/144170 for more details.
       For full logs, run 'nix log /nix/store/c8nawz7rzv9zz16wdi4f1ywpc5y12dh9-ros-rolling-ompl-1.5.2-r3.drv'.
error: 1 dependencies of derivation '/nix/store/dxy85ns0jpryl076ahl929va4jywqz3z-ros-rolling-moveit-planners-ompl-2.8.0-r2.drv' failed to build
error (ignored): error: cannot unlink '/tmp/nix-build-ros-rolling-geometry-msgs-5.2.0-r1.drv-0': Directory not empty
error: 1 dependencies of derivation '/nix/store/vvgzlbpaj3pnna2kicsd0yyaalwc5nhw-My-ROS-Project-Build-Environment-env.drv' failed to build
```

werner291, 2023-10-04:
Ok, I'm getting some CMake errors that  I can only assume is related:

```
CMake Error in CMakeLists.txt:
  Imported target "moveit_planners_ompl::moveit_ompl_interface" includes
  non-existent path

    "/nix/store/mg432iy0715cc7fylwg10bsdhsg4qy30-ros-rolling-moveit-planners-ompl-2.8.0-r2/include/moveit_planners_ompl"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "moveit_planners_ompl::moveit_ompl_interface" includes
  non-existent path

    "/nix/store/mg432iy0715cc7fylwg10bsdhsg4qy30-ros-rolling-moveit-planners-ompl-2.8.0-r2/include/moveit_planners_ompl"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

This occurs when I invoke this in my CMakeLists:

```
set(AMENT_DEPS "moveit_core" "moveit_msgs" "shape_msgs" "rclcpp" "ompl" "moveit_planners_ompl")

ament_target_dependencies(${PROJECT_NAME}_shared ${AMENT_DEPS})
```

lopsided98, 2024-03-23:
The patch is now applied in the overlay.

closed 2024-03-23 by lopsided98