lopsided98/nix-ros-overlay #315: Turtlebot3 Gazebo fails to build in ROS Iron

Educorreia932, 2023-10-11:
Trying to use `ros-iron-turtlebot3-gazebo` package results in the following error:

```sh
error: builder for '/nix/store/k6gqq6vm8zh9528d76aqq413api4jl6f-ignition-msgs5-5.11.0.drv' failed with exit code 2;
       last 10 log lines:
       > build flags: -j12 SHELL=/nix/store/vqvj60h076bhqj6977caz0pfxs6543nb-bash-5.2-p15/bin/bash
       > [  0%] Building CXX object src/CMakeFiles/ign_msgs_gen.dir/Generator.cc.o
       > [  0%] Building CXX object src/CMakeFiles/ign_msgs_gen.dir/generator_main.cc.o
       > [  0%] Linking CXX executable ../bin/ign_msgs_gen
       > /nix/store/rhhll3vwpj38ri72ahrrrvcbkhz4fhh6-binutils-2.40/bin/ld: CMakeFiles/ign_msgs_gen.dir/Generator.cc.o: undefined reference to symbol '_ZN4absl12lts_2023012513hash_internal15MixingHashState5kSeedE'
       > /nix/store/rhhll3vwpj38ri72ahrrrvcbkhz4fhh6-binutils-2.40/bin/ld: /nix/store/ci3nvrcx269dws6yr1nckxs752myi9if-abseil-cpp-20230125.3/lib/libabsl_hash.so.2301.0.0: error adding symbols: DSO missing from command line
       > collect2: error: ld returned 1 exit status
       > make[2]: *** [src/CMakeFiles/ign_msgs_gen.dir/build.make:115: bin/ign_msgs_gen] Error 1
       > make[1]: *** [CMakeFiles/Makefile2:258: src/CMakeFiles/ign_msgs_gen.dir/all] Error 2
       > make: *** [Makefile:146: all] Error 2
 ```

rgrunbla, 2023-10-20:
Likely, "nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo" fails here, with those logs : 
[logs.txt](https://github.com/lopsided98/nix-ros-overlay/files/13051602/logs.txt)

wentasah, 2023-11-04:
> Likely, "nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo" fails here, with those logs : [logs.txt](https://github.com/lopsided98/nix-ros-overlay/files/13051602/logs.txt)

@rgrunbla This command works for me. Your log contains `g++: fatal error: Killed signal terminated program cc1plus`, which can be caused by out-of-memory situation.

wentasah, 2023-11-04:
@Educorreia932 You should not build `nix-ros-overlay` against some random commit from nixpkgs. Use the same version of nixpkgs used by this overlay. You can do that by replacing:

    nixpkgs.url = "github:nixos/nixpkgs/nixos-unstable";

with 

    nixpkgs.follows = "ros-overlay/nixpkgs";

If I do that, everything builds just fine.

hacker1024, 2023-11-16:
> I changed `nixpkgs` to follow `ros-overlay`, removed unnecessary packages and tried using Humble instead:
> 
> ```nix
> {
>   inputs = {
>     nixpkgs.follows = "ros-overlay/nixpkgs";
>     flake-utils.url = "github:numtide/flake-utils";
>     ros-overlay.url = "github:lopsided98/nix-ros-overlay";
>   };
> 
>   outputs = { nixpkgs, flake-utils, ros-overlay, ... }:
>     with nixpkgs.lib;
>     with flake-utils.lib;
> 
>     eachSystem allSystems (system:
>       let
>         pkgs = import nixpkgs {
>           inherit system;
>           overlays = [ ros-overlay.overlays.default ];
>         };
>       in
>       {
>         devShells.default = with pkgs; mkShell {
>           buildInputs = with rosPackages.humble; [
>             (buildEnv {
>               paths = [
>                 gazebo
>                 ros-core
>               ];
>             })
>           ];
>         };
>       });
> }
> ```
> 
> But now I get a different error:
> 
> ```shell
> /build/ignition-common3_3.15.1-source/av/src/AudioDecoder.cc:348:41: error: 'AV_CODEC_CAP_TRUNCATED' was not declared in this scope; did
>  you mean 'AV_CODEC_CAP_HARDWARE'?
>   348 |   if (this->data->codec->capabilities & AV_CODEC_CAP_TRUNCATED)
>       |                                         ^~~~~~~~~~~~~~~~~~~~~~
>       |                                         AV_CODEC_CAP_HARDWARE
> /build/ignition-common3_3.15.1-source/av/src/AudioDecoder.cc:349:36: error: 'AV_CODEC_FLAG_TRUNCATED' was not declared in this scope; di
> d you mean 'AV_CODEC_FLAG_DROPCHANGED'?
>   349 |     this->data->codecCtx->flags |= AV_CODEC_FLAG_TRUNCATED;
>       |                                    ^~~~~~~~~~~~~~~~~~~~~~~
>       |                                    AV_CODEC_FLAG_DROPCHANGED
> ```

I fixed this with an overlay (see #327)

Educorreia932, 2023-11-17:
With or without the `ignition` overlay, my computer is freezing when trying to build `ignition-msgs`

hacker1024, 2023-11-19:
> With or without the `ignition` overlay, my computer is freezing when trying to build `gazebo`

Sounds like a RAM issue. Have you got swap space?
I've also found the Zen kernel to be more responsive during large builds.