lopsided98/nix-ros-overlay #318: Can't build moveit-ros-planning-interface

tmayoff, 2023-10-13:
This is the error

```
error: builder for '/nix/store/ggqw3bamz3p7jrjhkf7paxab6hsl4xxk-ros-humble-moveit-kinematics-2.5.5-r1.drv' failed with exit code 1;
       last 10 log lines:
       >     moveit_ros_planningConfig.cmake
       >     moveit_ros_planning-config.cmake
       >
       >   Add the installation prefix of "moveit_ros_planning" to CMAKE_PREFIX_PATH
       >   or set "moveit_ros_planning_DIR" to a directory containing one of the above
       >   files.  If "moveit_ros_planning" provides a separate development package or
       >   SDK, be sure it has been installed.
       >
       >
       > -- Configuring incomplete, errors occurred!
       For full logs, run 'nix log /nix/store/ggqw3bamz3p7jrjhkf7paxab6hsl4xxk-ros-humble-moveit-kinematics-2.5.5-r1.drv'.
error: 1 dependencies of derivation '/nix/store/xhxnfjw53v5hx174mydxjm9g9xipywjg-ros-humble-moveit-ros-move-group-2.5.5-r1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/ablw1xm27cirrbfhjc89dwh5y3f4z51p-ros-humble-moveit-ros-planning-interface-2.5.5-r1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/d9lnd12n4al52j3gsgwzb1wb0p536dvx-ros-devenv-env.drv' failed to build
```

This is my flake.
```nix
{
  description = "Testing ROS on NixOS";

  nixConfig = {
    substituters = [ 
      "https://nix-community.cachix.org"
	  "https://cache.nixos.org/"
      "https://ros.cachix.org"
     ];
    trusted-public-keys = [
	  "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY="
      "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
      "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo="
    ];
  };

  inputs = {
    flake-utils.url = "github:numtide/flake-utils";
    ros-flake.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, flake-utils, ros-flake }: 
    # flake-utils.lib.eachDefaultSystem (system:
      let
        system = "x86_64-linux";
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ros-flake.overlays.default (import ./replace-ompl.nix)];
        };
      in
      {
        devShells.x86_64-linux.default = pkgs.mkShell {
          name = "ros devenv";
          buildInputs = with pkgs; with pkgs.rosPackages.humble; [
            pkg-config
            cairo
            eigen
            pcl
            pcl-conversions

            ament-cmake-core
            ros-core
            ros2run
            ros2launch
            rmw-cyclonedds-cpp
            xacro
            joint-state-publisher 
            rqt-tf-tree
            # rtabmap-ros
            # navigation2
            # nav2-bringup

            ros2-control
            ros2-controllers

            teleop-twist-keyboard
            joint-state-publisher-gui
            rqt-robot-steering
            rqt-joint-trajectory-controller

            # Gazebo
            # gz-fortress
            # ros-gz-bridge
            gazebo-ros

            # Image Pipeline
            image-pipeline

            # MoveIt
            pluginlib
            graph-msgs
            moveit-core
            moveit-ros-planning
            moveit-planners-ompl
            moveit-planners-chomp
            moveit-visual-tools
            moveit-ros-planning-interface
            # moveit
          ];
          RMW_IMPLEMENTATION = "rmw_cyclonedds_cpp";
        };
      };
    # );
}
```

Any help is appreciated.

Educorreia932, 2023-10-13:
Add `moveit_ros_planning` package to your `buildInputs`

tmayoff, 2023-10-13:
It's already there

Educorreia932, 2023-10-13:
> It's already there

I'm sorry, I hadn't noticed. Then try to encapsulate your packages inside a `buildEnv` statement, like so:

```nix
buildInputs = with pkgs; with pkgs.rosPackages.humble; [
  (buildEnv {
    paths = [
      # packages
      # ...
    ];
  })
]
```

tmayoff, 2023-10-13:
Still not working

```nix
{
  description = "Testing ROS on NixOS";

  nixConfig = {
    substituters = [ 
      "https://nix-community.cachix.org"
	  "https://cache.nixos.org/"
      "https://ros.cachix.org"
     ];
    trusted-public-keys = [
	  "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY="
      "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
      "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo="
    ];
  };

  inputs = {
    flake-utils.url = "github:numtide/flake-utils";
    ros-flake.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, flake-utils, ros-flake }: 
    # flake-utils.lib.eachDefaultSystem (system:
      let
        system = "x86_64-linux";
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ros-flake.overlays.default (import ./replace-ompl.nix)];
        };
      in
      {
        devShells.x86_64-linux.default = pkgs.mkShell {
          name = "ros devenv";
          buildInputs = with pkgs; with pkgs.rosPackages.humble; [
            (buildEnv {
              paths = [
                pkg-config
                cairo
                eigen
                pcl
                pcl-conversions

                ament-cmake-core
                ros-core
                ros2run
                ros2launch
                rmw-cyclonedds-cpp
                xacro
                joint-state-publisher 
                rqt-tf-tree
                # rtabmap-ros
                # navigation2
                # nav2-bringup

                ros2-control
                ros2-controllers

                teleop-twist-keyboard
                joint-state-publisher-gui
                rqt-robot-steering
                rqt-joint-trajectory-controller

                # Gazebo
                # gz-fortress
                # ros-gz-bridge
                gazebo-ros

                # Image Pipeline
                image-pipeline

                # MoveIt
                pluginlib
                graph-msgs
                moveit-core
                moveit-ros-planning
                moveit-planners-ompl
                moveit-planners-chomp
                moveit-visual-tools
                moveit-ros-planning-interface
                # moveit
              ];
            })
          ];
          RMW_IMPLEMENTATION = "rmw_cyclonedds_cpp";
        };
      };
    # );
}
```

movefasta, 2023-10-15:
We also had some issues on building moveit here https://github.com/lopsided98/nix-ros-overlay/issues/244
Now we have a working devshell via flake with some overrides
https://gitlab.com/robossembler/nix-robossembler-overlay/-/blob/master/flake.nix?ref_type=heads#L39

Maybe it can hepls.

tmayoff, 2023-10-15:
That seems to have been it, thanks!

movefasta, 2024-01-26:
@lopsided98 i think we could close that

closed 2026-08-29 by wentasah