lopsided98/nix-ros-overlay #319: ros-gz Fails to build, ros-gz-bridge ros-gz-images, etc.. don't exist

tmayoff, 2023-10-15:
It seems for every 

```
error:
       … while calling the 'derivationStrict' builtin

         at /builtin/derivation.nix:9:12: (source not available)

       … while evaluating derivation 'ros-devenv'
         whose name attribute is located at /nix/store/rkkiiyv317i5c9iwqz8cz7prvd04k08k-source/pkgs/stdenv/generic/make-derivation.nix:300:7

       … while evaluating attribute 'buildInputs' of derivation 'ros-devenv'

         at /nix/store/rkkiiyv317i5c9iwqz8cz7prvd04k08k-source/pkgs/stdenv/generic/make-derivation.nix:347:7:

          346|       depsHostHost                = lib.elemAt (lib.elemAt dependencies 1) 0;
          347|       buildInputs                 = lib.elemAt (lib.elemAt dependencies 1) 1;
             |       ^
          348|       depsTargetTarget            = lib.elemAt (lib.elemAt dependencies 2) 0;

       error: evaluation aborted with the following error message: 'Function called without required argument "ros-gz-bridge" at /nix/store/dc7sviyvcgif1ss22kgx5pvvx28dks32-source/distros/humble/ros-gz/default.nix:5, did you mean "ros-ign-bridge"?'
```

lopsided98, 2023-11-26:
New Gazebo (as opposed to classic) and all its Ignition dependencies need to be packaged manually. This is quite a bit of work, and I started it at one point but haven't had time to finish it.

muellerbernd, 2023-12-15:
I've tried to get ignition gazebo running. You can find my sources [here](https://github.com/muellerbernd/gazebo-sim-overlay). In that repo I tried to mimic the structure we are using in the nix-ros-overlay. My repo is still WIP but I managed to compile [Gazebo Sim Harmonic](https://github.com/gazebosim/gz-sim/releases/tag/gz-sim8_8.0.0) and all it's dependencies. The GUI is not working right now:
```bash
gz sim 
Library error: libgz-tools2-backward.so not found. Improved backtrace generation will be disabled
Library error for [/nix/store/g13mc8m3dx5insvl5kz5pxp0igm7kgfg-gazebo-sim-gz-sim8_8.0.0/nix/store/g13mc8m3dx5insvl5kz5pxp0igm7kgfg-gazebo-sim-gz-sim8_8.0.0/lib/libgz-sim8-gz.so.8.0.0]: /nix/store/g13mc8m3dx5insvl5kz5pxp0igm7kgfg-gazebo-sim-gz-sim8_8.0.0/nix/store/g13mc8m3dx5insvl5kz5pxp0igm7kgfg-gazebo-sim-gz-sim8_8.0.0/lib/libgz-sim8-gz.so.8.0.0: cannot open shared object file: No such file or directory
```
@lopsided98 maybe you have an idea how to fix this.

muellerbernd, 2023-12-29:
Little Update to my previous post. In my [repo](https://github.com/muellerbernd/gazebo-sim-overlay) I have managed to pull gazebo-classic out of the nix-ros-overlay repo and I was able to compile [Gazebo Sim Harmonic](https://github.com/gazebosim/gz-sim/releases/tag/gz-sim8_8.0.0), but I still have the error I mentioned in my previous post when I try to use the gui from the new Gazebo-Sim.

muellerbernd, 2024-02-17:
Update: in my [gazebo overlay repo](https://github.com/muellerbernd/gazebo-sim-overlay) you get an overlay for gazebo-classic and gazebo-sim-harmonic. gazebo-classic is running fine but gazebo-sim-harmonic is not working. When I try to run `gz gui I get QML runtime errors:
```module "QtQuick.Controls" is not installed```
I am not so familiar with QT and maybe someone can help with that? @lopsided98  @wentasah  @hacker1024

wentasah, 2024-02-18:
Hi!

I, on the other hand, am not familiar with gazebo :-). When using your flake as follows:
```
nix develop
gz gui -v
```
I get:
```
Library error: libgz-tools2-backward.so not found. Improved backtrace generation will be disabled
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Aborted (core dumped)
```
This is different than your `module "QtQuick.Controls" is not installed`. I'm on NixOS 23.11.

However, I looked at you code and it seems to me that you need to apply the Qt wrapper to the `gz` binary, which seems to run everything. I managed to do that with the following change:
```diff
diff --git a/pkgs/ignition/tools/default.nix b/pkgs/ignition/tools/default.nix
index 7caf997..1eef6b4 100644
--- a/pkgs/ignition/tools/default.nix
+++ b/pkgs/ignition/tools/default.nix
@@ -29,13 +29,17 @@ stdenv.mkDerivation rec {
     hash = srcHash;
   };
 
-  nativeBuildInputs = [ cmake pkg-config ];
+  nativeBuildInputs = [ cmake pkg-config wrapQtAppsHook ];
   # pkg-config is needed to use some CMake modules in this package
   # propagatedBuildInputs = [ pkg-config ];
   propagatedNativeBuildInputs = [ ignition-cmake ];
   buildInputs = [ ignition-cmake ruby ronn ];
 
-  # dontWrapQtApps = true;
+  dontWrapQtApps = true;
+
+  postFixup = ''
+    wrapQtApp $out/bin/gz
+  '';
 
   # makeWrapperArgs = [
   #   "\${qtWrapperArgs[@]}"
```
Note that `gz` is not wrapped automatically by the `wrapQtAppsHook`, because [it's not an ELF binary](https://github.com/NixOS/nixpkgs/blob/5863c27340ba4de8f83e7e3c023b9599c3cb3c80/pkgs/development/libraries/qt-5/hooks/wrap-qt-apps-hook.sh#L88), and we must do it manually in the `postFixup` hook.

With this change, it is still not running, but I get the same error that you report in your `README.md`:
```
bsh ▶ gz gui -v
Library error: libgz-tools2-backward.so not found. Improved backtrace generation will be disabled
[GUI] [Wrn] [Application.cc:904] [QT] QQmlApplicationEngine failed to load component
[GUI] [Wrn] [Application.cc:904] [QT] qrc:qml/Main.qml:25:1: Type ApplicationWindow unavailable
[GUI] [Wrn] [Application.cc:904] [QT] file:///nix/store/3g6524r2b1ns5p9kdfpccxq8s08m9syx-qtquickcontrols2-5.15.12-bin/lib/qt-5.15.12/qml/QtQuick/Controls.2/Material/qmldir: plugin cannot be loaded for module ".nix.store.3g6524r2b1ns5p9kdfpccxq8s08m9syx-qtquickcontrols2-5.15.12-bin.lib.qt-5.15.12.qml.QtQuick.Controls.Material": Module namespace 'QtQuick.Controls.Material' does not match import URI '.nix.store.3g6524r2b1ns5p9kdfpccxq8s08m9syx-qtquickcontrols2-5.15.12-bin.lib.qt-5.15.12.qml.QtQuick.Controls.Material'
[GUI] [Err] [MainWindow.cc:116] Internal error: Failed to instantiate QML file [qrc:qml/Main.qml]
[GUI] [Err] [Application.cc:207] Failed to initialize main window.
```
I can get a bit further when I `LD_PRELOAD` the Material plugin:
```
LD_PRELOAD=/nix/store/3g6524r2b1ns5p9kdfpccxq8s08m9syx-qtquickcontrols2-5.15.12-bin/lib/qt-5.15.12/qml/QtQuick/Controls.2/Material/libqqc2materialstyleplugin.so gz gui -v
```
Then, I'm getting different errors depending on whether I'm on NixOS stable or unstable. With stable, I get:
```
Library error: libgz-tools2-backward.so not found. Improved backtrace generation will be disabled
Library error: [/nix/store/kvgfg5z6rd7j7jf64ddkcy4pp378nl4p-gz-gui8-8.0.0/lib/libgz-gui8.so.8.0.0] not found.
```
(and preloading also `libgz-gui8.so.8.0.0` results in ``version `GLIBCXX_3.4.32' not found`` – likely caused by OpenGL/Vulkan usespace drivers)

With unstable:
```
LD_PRELOAD=/nix/store/63hphasrgfkc79qfavrhrz0sc5sqn8xz-qtquickcontrols2-5.15.12-bin/lib/qt-5.15.12/qml/QtQuick/Controls.2/Material/libqqc2materialstyleplugin.so gz gui -v
Library error: libgz-tools2-backward.so not found. Improved backtrace generation will be disabled
[GUI] [Wrn] [Application.cc:904] [QT] QQmlApplicationEngine failed to load component
[GUI] [Wrn] [Application.cc:904] [QT] qrc:qml/Main.qml:186:7: Type ToolButton unavailable
[GUI] [Wrn] [Application.cc:904] [QT] file:///nix/store/63hphasrgfkc79qfavrhrz0sc5sqn8xz-qtquickcontrols2-5.15.12-bin/lib/qt-5.15.12/qml/QtQuick/Controls.2/Material/qmldir: plugin cannot be loaded for module ".nix.store.63hphasrgfkc79qfavrhrz0sc5sqn8xz-qtquickcontrols2-5.15.12-bin.lib.qt-5.15.12.qml.QtQuick.Controls.Material": Module namespace 'QtQuick.Controls.Material' does not match import URI '.nix.store.63hphasrgfkc79qfavrhrz0sc5sqn8xz-qtquickcontrols2-5.15.12-bin.lib.qt-5.15.12.qml.QtQuick.Controls.Material'
[GUI] [Err] [MainWindow.cc:116] Internal error: Failed to instantiate QML file [qrc:qml/Main.qml]
[GUI] [Err] [Application.cc:207] Failed to initialize main window.
```
As can be seen, the error is reported at `Main.qml:186` whereas previously, it was at `Main.qml:25` (progress!). I don't know why the material plugin is not loaded automatically. I've found similar problems reported on the internet ([here](https://stackoverflow.com/questions/55439010/how-to-configure-qml-import-path-when-using-qt5-from-nixpkgs) and [here](https://github.com/NixOS/nixpkgs/issues/111501)) but without a solution.

I think I'd need to look at Qt source code what the error `Module namespace 'QtQuick.Controls.Material' does not match import URI` actually means. But it's too late now and I'll be busy a few next days. Maybe, somebody else can figure out what to do next.

muellerbernd, 2024-02-19:
@wentasah thanks for your work. I have managed to get it working :+1: , which means that the gui is running as intended. I had to add , `qtquickcontrols2 qtgraphicaleffects` as propagatedNativeBuildInputs to `ignition-tools` (more or less via trial and error). Please take a look at [the develop branch](https://github.com/muellerbernd/gazebo-sim-overlay/tree/develop). I think some more work is needed, but for now the gui parts are running.

muellerbernd, 2024-03-24:
@wentasah @lopsided98 I have managed to get gz-harmonic working. See my [gazebo overlay repo](https://github.com/muellerbernd/gazebo-sim-overlay). Some small adjustments are still needed. For example I had to set a special variable `GZ_CONFIG_PATH` like documented [here](https://github.com/gazebosim/gz-gui#known-issue-of-command-line-tools). I set this variable in the shell.nix for gz-harmonic (see [here](https://github.com/muellerbernd/gazebo-sim-overlay/blob/main/shells/gazebo-ignition.nix)). Do you know a better solution for that?

muellerbernd, 2024-08-18:
With #422 it is possible to build the gz-packages for jazzy and rolling, because the new gazebo and all its Ignition dependencies are packaged as ROS-packages. But then there is still a problem with the `GZ_CONFIG_PATH` variable, it is not set. 
For humble there is still no way for building `ros-gz`.
As mentioned in my previous comment, I have managed to get gz-harmonic working in a separate flake and we could use that in nix-ros-overlay. I think it would be better to keep gazebo outside of ROS, meaning no `gz-*-vendor` package.
What do you guys think about it?

wentasah, 2026-08-29:
I'm closing this. We have `gz-*` working for long time. If someone is interested, try the following:
```sh
nix-shell examples/ros2-gz.nix --argstr rosDistro jazzy
gz sim
```

closed 2026-08-29 by wentasah