lopsided98/nix-ros-overlay #323: How to transitively override packages in this overlay

Hugal31, 2023-10-27:
Hi, Thanks for all your work!

I was trying to patch an [infinite loop in rosgraph](https://github.com/ros/ros_comm/issues/2352) and could fix it in the package itself, but not its dependent packages.That is, if I have a script using rosgraph, it uses the patched one, however if I use ros-core, it uses the unpatched version.

Is there a way to apply the patch and make it affect every dependents?

I am not very fluet in nix, sorry if this is a known pattern.

My flake.nix:
```nix
{
  inputs = {
    flake-utils.url = github:numtide/flake-utils;
    nixpkgs.url = github:NixOS/nixpkgs;
    nix-ros-overlay.url = github:lopsided98/nix-ros-overlay;
  };

  outputs = { self, nixpkgs, flake-utils, nix-ros-overlay }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        fix-ros-logging = (final: prev: {
          rosPackages = nixpkgs.lib.recursiveUpdate prev.rosPackages {
            noetic.rosgraph = prev.rosPackages.noetic.rosgraph.overrideAttrs ({
              patches ? [], ...
            }: {
              patches = patches ++ [
                ./rosgraph-2353.patch
              ];
            });
          };
        });
        overlays = [ nix-ros-overlay.overlays.default fix-ros-logging ];
        pkgs = import nixpkgs {
          inherit system overlays;
        };
        ros = pkgs.rosPackages.noetic;
      in {

        defaultPackage = pkgs.stdenv.mkDerivation {
          name = "hello ROS";
          src = self;
        };

        devShells.default = pkgs.mkShell {
          packages = [
            ros.ros-core
          ];
        };
      }
    );
}
```

```
> nix develop path:.
> roscore
# Infinite loop
```

<details>
<summary>The patch, if needed</summary>

```
diff --git a/ros_comm-release-release-noetic-rosgraph-1.16.0-1/src/rosgraph/roslogging.py b/ros_comm-release-release-noetic-rosgraph-1.16.0-1-patched/src/rosgraph/roslogging.py
index 9ecc121..5adc95f 100644
--- a/src/rosgraph/roslogging.py
+++ b/src/rosgraph/roslogging.py
@@ -69,6 +69,8 @@ class RospyLogger(logging.getLoggerClass()):
                 break
             if f.f_back:
                 f = f.f_back
+            else: # Reached the last stack frame and found no matching one.
+                raise ValueError("Could not find function [%s] on the framestack"%func_name)
 
         # Jump up two more frames, as the logger methods have been double wrapped.
         if f is not None and f.f_back and f.f_code and f.f_code.co_name == '_base_logger':
```

</details>

hacker1024, 2023-10-28:
You're looking for `overrideScope`. Unlike `recursiveUpdate`, this changes the entire fixed point package set.

```nix
rosPackages.noetic.overrideScope (rosSelf: rosSuper: {
  rosgraph = rosSuper.rosgraph.overrideAttrs ({
    patches ? [], ...
  }: {
    patches = patches ++ [
      ./rosgraph-2353.patch
    ];
  });
})
```

Hugal31, 2023-10-28:
That worked! Many thanks.

closed 2023-10-28 by Hugal31