lopsided98/nix-ros-overlay #324: Can't play bags on Humble - No module named 'yaml'

Programmerino, 2023-11-03:
Running a command like `ros2 bag play file.mcap` causes:
```
Failed to load entry point 'play': No module named 'yaml'
Traceback (most recent call last):
  File "/nix/store/j9z06lirncbj5xrgj2i6x1w4sabbpj40-python3.11-ros-humble-ros2cli-0.18.7-r1/bin/.ros2-wrapped", line 34, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.7', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/j9z06lirncbj5xrgj2i6x1w4sabbpj40-python3.11-ros-humble-ros2cli-0.18.7-r1/lib/python3.11/site-packages/ros2cli/cli.py", line 50, in main
    add_subparsers_on_demand(
  File "/nix/store/j9z06lirncbj5xrgj2i6x1w4sabbpj40-python3.11-ros-humble-ros2cli-0.18.7-r1/lib/python3.11/site-packages/ros2cli/command/__init__.py", line 250, in add_subparsers_on_demand
    extension.add_arguments(
  File "/nix/store/x7kg09f2y5fdcnkb4qs9q64id6fyad1k-python3.11-ros-humble-ros2bag-0.15.8-r1/lib/python3.11/site-packages/ros2bag/command/bag.py", line 26, in add_arguments
    add_subparsers_on_demand(
  File "/nix/store/j9z06lirncbj5xrgj2i6x1w4sabbpj40-python3.11-ros-humble-ros2cli-0.18.7-r1/lib/python3.11/site-packages/ros2cli/command/__init__.py", line 237, in add_subparsers_on_demand
    extension = command_extensions[name]
                ~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: 'play'
```

I have `with pkgs.rosPackages.humble; [ros2cli ros2run rviz2 rosbag2]` added to my devShell packages.

wentasah, 2023-11-04:
Yes, the package lacks the dependency on Python's `yaml` package. I've [submitted an upstream PR](https://github.com/ros2/rosbag2/pull/1490) to add it. Before it propagates to ROS releases, you can just add `python3Packages.pyyaml` to you devShell to resolve this issue.

Programmerino, 2023-11-07:
Thank you for taking the initiative! I'll use your workaround until then.

closed 2023-11-07 by Programmerino