lopsided98/nix-ros-overlay #326: RQT-graph not building due to python binding

vBruegge, 2023-11-06:
I'm currently running nixos-unstable and wanted to install rqt for my ROS project. Whenever I add the package to the flake, I get the error:
```  error: pyside2 not supported for interpreter python3.11 ```
This issue occured also for freecad and was there fixed [here](https://github.com/NixOS/nixpkgs/pull/263599). Since I'm not sure, how you can change this for the ros-overlay, it would be great if somebody could either help me. Here is my flake for installing ros:
```
{
  description = "ROS-Project";

  inputs = {
    ros.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
  };

  outputs = { self, nixpkgs, ros }: {
    devShells.x86_64-linux.default =
      let pkgs = import nixpkgs { overlays = [ ros.overlays.default ]; system = "x86_64-linux"; };
      in pkgs.mkShell {
        nativeBuildInputs = with pkgs; [
          glibcLocales

	  (rosPackages.humble.buildEnv {
	     paths = with rosPackages.humble; [
		rqt
               ros2cli
               colcon
                ];
	  })
	];
      };
   };
}
```
Thanks in advance!

wentasah, 2023-11-06:
If you just want to have your flake running, use older nixpkgs version stored in nix-ros-overlay flake, as I suggest in https://github.com/lopsided98/nix-ros-overlay/issues/315#issuecomment-1793575795.

wentasah, 2023-11-06:
BTW, the fix mentioned by @vBruegge is already in `nixos-unstable`. I can now build the `develop` branch against the latest `nixpkgs/nixos-unstable` without problems.

closed 2023-11-07 by vBruegge