lopsided98/nix-ros-overlay #340: Devshell with `noetic.ros-core` fails to build

damb, 2023-12-31:
First of all, thanks for the great overlay and all the work.

This is rather a question than a real issue. Also, I'm fairly new to Nix and the Nix Language. Besides, I'm not on NixOS.

My goal is to set up a development shell with the following `flake.nix`:

```nix
{
  inputs = {
    nixpkgs.url = "nixpkgs/nixpkgs-unstable";
    flake-utils.url = github:numtide/flake-utils;
    nix-ros-overlay.url = github:lopsided98/nix-ros-overlay;
  };

  outputs = { self, nixpkgs, flake-utils, nix-ros-overlay }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [nix-ros-overlay.overlays.default];
        };

        ros = pkgs.rosPackages.noetic;
      in {
        devShells.default = pkgs.mkShell {
          buildInputs = [
            ros.ros-core
          ]; 

          ROS_HOSTNAME = "localhost";
          ROS_MASTER_URI = "http://localhost:11311";
        };
      }
    );
}
```

Though, when running `nix develop` buillding the environment fails with

```
error: builder for '/nix/store/si87ak705ibkqdn5q44mdgvbwzi5fdn1-ros-noetic-std-msgs-0.5.13-r1.drv' failed with exit code 2;
       last 10 log lines:
       > [ 83%] Generating EusLisp manifest code for std_msgs
       > [ 83%] Generating Lisp code from std_msgs/UInt64MultiArray.msg
       > [ 83%] Generating Lisp code from std_msgs/UInt64.msg
       > [ 84%] Generating Lisp code from std_msgs/UInt8.msg
       > [ 84%] Generating Lisp code from std_msgs/UInt8MultiArray.msg
       > [ 84%] Built target std_msgs_generate_messages_nodejs
       > [ 84%] Built target std_msgs_generate_messages_py
       > [ 84%] Built target std_msgs_generate_messages_lisp
       > [ 84%] Built target std_msgs_generate_messages_eus
       > make: *** [Makefile:136: all] Error 2
       For full logs, run 'nix log /nix/store/si87ak705ibkqdn5q44mdgvbwzi5fdn1-ros-noetic-std-msgs-0.5.13-r1.drv'.
error: 1 dependencies of derivation '/nix/store/xch809l9z5j5dqb6ms5f7xz0nnk7h0rx-ros-noetic-ros-core-1.5.0-r1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/krbjx0zzz0cx6l1zy9g4zgpnrj0sg7xl-nix-shell-env.drv' failed to build
```

Again, the log states that `empy-4.0` is used whereas https://github.com/ros/genmsg/issues/63#issuecomment-1865890638 requires `empy-3.3.4`.
```
[...]

-- Found PythonInterp: /nix/store/5k91mg4qjylxbfvrv748smfh51ppjq0g-python3-3.11.6/bin/python (found version "3.11.6")
-- Using PYTHON_EXECUTABLE: /nix/store/5k91mg4qjylxbfvrv748smfh51ppjq0g-python3-3.11.6/bin/python
-- Using default Python package layout
-- Found PY_em: /nix/store/7vjnfzcrrwmbf9j2xmv7j6w60gdyxixf-python3.11-empy-4.0/lib/python3.11/site-packages/em.py
-- Using empy: /nix/store/7vjnfzcrrwmbf9j2xmv7j6w60gdyxixf-python3.11-empy-4.0/lib/python3.11/site-packages/em.py

[...]

Traceback (most recent call last):
  File "/nix/store/p4iqnydqrk3cly3ffvvx1knsvs9g3iwv-ros-noetic-gencpp-0.7.0-r1/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>
    genmsg.template_tools.generate_from_command_line_options(
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 213, in generate>
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 154, in generate>
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 93, in _generate>
    _generate_from_spec(input_file,
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 77, in _generate>
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
                                                                   ^^^^^^^^^^
AttributeError: module 'em' has no attribute 'RAW_OPT'
Traceback (most recent call last):
  File "/nix/store/p4iqnydqrk3cly3ffvvx1knsvs9g3iwv-ros-noetic-gencpp-0.7.0-r1/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>
    genmsg.template_tools.generate_from_command_line_options(
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 213, in generate>
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 154, in generate>
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 93, in _generate>
    _generate_from_spec(input_file,
  File "/nix/store/2fbx2ibj57r5l4cc46f7hkr8g9yl93fw-ros-noetic-genmsg-0.6.0-r1/lib/python3.11/site-packages/genmsg/template_tools.py", line 77, in _generate>
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})

[...]
```

Therefore, my question is how to fix my `flake.nix` to correctly build my dev shell?

Thanks for any hint.

damb, 2023-12-31:
Note that following the [`turtlebot3-gazebo.nix` example](https://github.com/lopsided98/nix-ros-overlay/blob/42cae93f39f0d26134a884a1ec20d1e36dd14f66/examples/turtlebot3-gazebo.nix) and replacing the `buildInputs` with
```nix
          buildInputs = [
            (ros.buildEnv {
              paths = [
                ros.ros-core
              ];
             })
          ];
```

leads to the same result.

lopsided98, 2024-01-01:
It looks like is caused by a breaking change in empy 4.0. I'll need to look into what is required to fix this, but for now you can use an older nixpkgs version. Generally, I'd recommend using the nixpkgs version pinned in the flake to avoid this type of issue.

damb, 2024-01-01:
Ok, thank you. So for now I'll go with the nixpkgs version pinned in the `flake.nix`.

wentasah, 2024-01-02:
I wanted to try something, which required newer nixpkgs and was hit by this issue too. My findings are available in the [rosidl repo](https://github.com/ros2/rosidl/issues/779).

I'm not sure how to deal with it here. Patching rosidl would require different patches for different ROS distros. EmPy author [suggests that distributions should ship both 3.x and 4.x versions simultaneously](http://www.alcyone.com/software/empy/README.html#for-package-maintainers). Maybe, it would be easiest if nixpkgs ships both empy versions for some time.

muellerbernd, 2024-01-30:
I am currently also running on newer nixpkgs. To deal with the empy issue I simply created a custom empy package in my [nix-ros-overlay fork](https://github.com/muellerbernd/nix-ros-overlay/tree/develop/pkgs/empy).

commit d0ed0ee3 (2024-02-09): Override to use empy 3.3.4

Empy 4.0 has many breaking changes and is not supported by ROS.

Fixes: #340
closed 2024-02-09 by lopsided98