lopsided98/nix-ros-overlay #343: Difficulty running turtlebot examples

grnnja, 2024-01-08:
Hi! I'm having difficulty running the example turtlebot flake command. It tries to find `opencv.cxxdev` but can't. Not sure if this is a problem on my end.

```bash
[prem@morokei:~]$ nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo
error:
       … while calling the 'derivationStrict' builtin

         at /derivation-internal.nix:9:12:

            8|
            9|   strict = derivationStrict drvAttrs;
             |            ^
           10|

       … while evaluating derivation 'nix-shell'
         whose name attribute is located at /nix/store/im9ymqc98dwazijqagl057yalg78lxx6-source/pkgs/stdenv/generic/make-derivation.nix:348:7

       … while evaluating attribute 'buildInputs' of derivation 'nix-shell'

         at /nix/store/im9ymqc98dwazijqagl057yalg78lxx6-source/pkgs/stdenv/generic/make-derivation.nix:395:7:

          394|       depsHostHost                = elemAt (elemAt dependencies 1) 0;
          395|       buildInputs                 = elemAt (elemAt dependencies 1) 1;
             |       ^
          396|       depsTargetTarget            = elemAt (elemAt dependencies 2) 0;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: attribute 'cxxdev' missing

       at /nix/store/rgrwzc3qyq0rlv6q7df3jdzaww4bzqgm-source/distros/noetic/cv-bridge/default.nix:19:42:

           18|   checkInputs = [ python3Packages.numpy rostest ];
           19|   propagatedBuildInputs = [ boost opencv opencv.cxxdev python3 python3Packages.opencv4 rosconsole sensor-msgs ];
             |                                          ^
           20|   nativeBuildInputs = [ catkin ];
```

with the nix shell example, it pulls the packages and compiles successfully but I am unable to run the demo.  I let it sit for a few minutes and nothing happened before I ctrl+c'd the process. Again I'm not sure if this is a problem on my end or not.

```bash
[prem@morokei:~]$ nix-shell \
  -I nix-ros-overlay=https://github.com/lopsided98/nix-ros-overlay/archive/master.tar.gz \
  --option extra-substituters 'https://ros.cachix.org' \
  --option trusted-public-keys 'cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' \
  '<nix-ros-overlay/examples/turtlebot3-gazebo.nix>'
# If not on NixOS, nixGL (https://github.com/guibou/nixGL) is needed for OpenGL support
roslaunch turtlebot3_gazebo turtlebot3_world.launch 

[nix-shell:~]$ roslaunch turtlebot3_gazebo turtlebot3_world.launch
... logging to /home/prem/.ros/log/a6decd14-adc6-11ee-ba96-98597abc5705/roslaunch-morokei-87316.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

^CTraceback (most recent call last):
  File "/nix/store/5s4yz3x6cjavpbhj2kz24bw019nvbg0v-ros-noetic-roslaunch-1.16.0-r1/bin/.roslaunch-wrapped", line 36, in <module>
    roslaunch.main()
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/roslaunch/parent.py", line 301, in start
    self.logger.info("starting roslaunch parent run")
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1489, in info
    self._log(INFO, msg, args, **kwargs)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1622, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rosgraph/roslogging.py", line 67, in findCaller
    filename = os.path.normcase(co.co_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 52, in normcase
KeyboardInterrupt
^CException ignored in atexit callback: <function _ros_atexit at 0x7f09eb271a80>
Traceback (most recent call last):
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rospy/core.py", line 615, in _ros_atexit
    signal_shutdown('atexit')
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rospy/core.py", line 560, in signal_shutdown
    _logger.info("signal_shutdown [%s]"%reason)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1489, in info
    self._log(INFO, msg, args, **kwargs)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1622, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rosgraph/roslogging.py", line 67, in findCaller
    filename = os.path.normcase(co.co_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 52, in normcase
KeyboardInterrupt: 

[nix-shell:~]$ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
... logging to /home/prem/.ros/log/c50f2d30-adc7-11ee-b75b-98597abc5705/roslaunch-morokei-88538.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

^CTraceback (most recent call last):
  File "/nix/store/5s4yz3x6cjavpbhj2kz24bw019nvbg0v-ros-noetic-roslaunch-1.16.0-r1/bin/.roslaunch-wrapped", line 36, in <module>
    roslaunch.main()
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/roslaunch/parent.py", line 301, in start
    self.logger.info("starting roslaunch parent run")
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1489, in info
    self._log(INFO, msg, args, **kwargs)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1622, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rosgraph/roslogging.py", line 64, in findCaller
    while hasattr(f, "f_code"):
          ^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
^CException ignored in atexit callback: <function _ros_atexit at 0x7f46095e1a80>
Traceback (most recent call last):
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rospy/core.py", line 615, in _ros_atexit
    signal_shutdown('atexit')
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rospy/core.py", line 560, in signal_shutdown
    _logger.info("signal_shutdown [%s]"%reason)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1489, in info
    self._log(INFO, msg, args, **kwargs)
  File "/nix/store/qp5zys77biz7imbk6yy85q5pdv7qk84j-python3-3.11.6/lib/python3.11/logging/__init__.py", line 1622, in _log
    fn, lno, func, sinfo = self.findCaller(stack_info, stacklevel)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/8280mkj35v5hmdr75mq9hn0l5gfb1l45-ros-env/lib/python3.11/site-packages/rosgraph/roslogging.py", line 67, in findCaller
    filename = os.path.normcase(co.co_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 52, in normcase
KeyboardInterrupt: 
```

lopsided98, 2024-01-09:
Probably the locked nixpkgs version is too old.

cc @hacker1024

hacker1024, 2024-01-09:
Ah yep, sorry about that!

We should maybe start using `lib.getOutput` when using specific outputs, as that handles missing outputs gracefully.

I'm not sure how that can be implemented in rosdistro, though - we might just need to split on the period in Super Flore.

hacker1024, 2024-01-09:
A silly workaround for immediate use would be to add an `opencv` overlay like so:

```nix
self: super:

{
  opencv = super.opencv.override ({ passthru ? { }, ... }: {
    passthru = passthru // { cxxdev = self.opencv; };
  });
}
```

crthilakraj, 2024-01-10:
@hacker1024 I assume its `super.opencv.overrideAttrs ` , since passthru is an attribute and not the argument.

purepani, 2024-02-08:
By chance, is it possible to easily fix this? If i just override opencv, my computer has to rebuild a lot of packages.

lopsided98, 2024-02-09:
I updated nixpkgs. Probably will break 10 new things though...

lopsided98, 2024-02-10:
Build is fixed now, but Gazebo just seems to hang without ever opening a window when I tested it.

hacker1024, 2024-02-11:
> Build is fixed now, but Gazebo just seems to hang without ever opening a window when I tested it.

I've seen this happen when the pinned Nixpkgs gets too old in comparison to the revision used to build NixOS.

grnnja, 2024-02-11:
Is it related to this? https://github.com/lopsided98/nix-ros-overlay/issues/323

I patched my devshell with this and it stopped roscore from hanging, not sure if it is the same issue.

purepani, 2024-02-15:
> Is it related to this? #323
> 
> I patched my devshell with this and it stopped roscore from hanging, not sure if it is the same issue.

It looks like this is in fact the issue. It originates from this issue in ros_comm
https://github.com/ros/ros_comm/issues/2352

which is fixed by this pull request:
https://github.com/ros/ros_comm/pull/2353

which is where the patch in https://github.com/lopsided98/nix-ros-overlay/issues/323 originates from.

MatthewCroughan, 2024-03-06:
@grnnja can you share the last working commit of this repo (nix-ros-overlay) that you were using before making this issue?

taylorpool, 2024-03-12:
Hi all, having the same issue with roscore hanging with the following versions:
nixpkgs:
  branch: nixpkgs-unstable
  hash: 0fbcc4b2e8571f4af39be41752581ea09dd9ab06
nix-ros-overlay:
  branch: master
  hash: 5f5fc569845b81487033d553c748e140e2ef4ca2

I can also provide a minimal example if that's helpful

grnnja, 2024-03-12:
@MatthewCroughan @taylorpool I don't have the exact commit where nix-ros-overlay worked for me, but I have it working right now after applying the rosgraph patch I mentioned here https://github.com/lopsided98/nix-ros-overlay/issues/343#issuecomment-1937419031. 

Here is a minimal flake and the patch.

`flake.nix`:
```nix
{

  description = "A basic flake with a shell";
  inputs =
    {
      nixpkgs.url = "github:lopsided98/nixpkgs/nix-ros";
      flake-utils.url = "github:numtide/flake-utils";
      ros-flake.url = "github:lopsided98/nix-ros-overlay";
    };

  outputs = { self, nixpkgs, flake-utils, ros-flake }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        # fixes roscore hanging
        # https://github.com/lopsided98/nix-ros-overlay/issues/323
        fix-ros-logging = (final: prev: {
          rosPackages.noetic = prev.rosPackages.noetic.overrideScope (rosSelf: rosSuper: {
            rosgraph = rosSuper.rosgraph.overrideAttrs ({ patches ? [ ], ... }: {
              patches = patches ++ [
                ./rosgraph.patch
              ];
            });
          });
        });
        pkgs = import nixpkgs
          {
            inherit system;
            overlays = [
              ros-flake.overlays.default
              fix-ros-logging
            ];
          };
        ros = pkgs.rosPackages.noetic;
      in
      {
        devShell = pkgs.mkShell
          {
            buildInputs = with pkgs; [
              ros.ros-core
            ];
          };

      });
  # I had to add myself as a trusted user in my nixos config for this to work
  # https://discourse.nixos.org/t/nix-build-trying-to-build-basic-tools-for-all-packages-from-scratch/11097/4
  nixConfig = {
    substituters = "https://cache.nixos.org https://ros.cachix.org";
    trusted-public-keys = "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY= ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=";
  };
}
```

`rosgraph.patch`:
```patch
diff --git a/ros_comm-release-release-noetic-rosgraph-1.16.0-1/src/rosgraph/roslogging.py b/ros_comm-release-release-noetic-rosgraph-1.16.0-1-patched/src/rosgraph/roslogging.py
index 9ecc121..5adc95f 100644
--- a/src/rosgraph/roslogging.py
+++ b/src/rosgraph/roslogging.py
@@ -69,6 +69,8 @@ class RospyLogger(logging.getLoggerClass()):
                 break
             if f.f_back:
                 f = f.f_back
+            else: # Reached the last stack frame and found no matching one.
+                raise ValueError("Could not find function [%s] on the framestack"%func_name)
 
         # Jump up two more frames, as the logger methods have been double wrapped.
         if f is not None and f.f_back and f.f_code and f.f_code.co_name == '_base_logger':
```

taylorpool, 2024-03-16:
That worked! Thank you so much!

lopsided98, 2024-04-11:
e239e71fd2ce40ce1f7499328a4c92e9473fec88

closed 2024-04-11 by lopsided98