lopsided98/nix-ros-overlay #353: CI is broken ?

movefasta, 2024-02-13:
Hello!

About several months CI doesn't work on all distros because of

(for examble [this](https://github.com/lopsided98/nix-ros-overlay/actions/runs/7856578661/job/21439502610) action)
Cached Failed Build JOB:
`Cache not found for input keys: failed-builds-rolling-x86_64-linux-c8afb864b67d2c77b4a523953be4cb73cb8f4d4a, failed-builds-rolling-x86_64-linux-`
Run ./.github/actions/nix-ros-build-action JOB:
 `Warning: Failed to load failed builds cache: Error: ENOENT: no such file or directory, open 'failed-builds/failed-builds.json'`.

@lopsided98 how we can fix it?

lopsided98, 2024-02-14:
That warning is normal with no cache (because none of the builds have succeeded recently). I suspect some number of packages are being built because it runs for a few hours before failing, but then something goes wrong. Build log output is not printed because it would produce an absurd amount of interleaved output. It is supposed to be dumped in an organized manner at the end, but something causes the action to fail before that. Even if all builds fail, the action is still supposed to succeed.

movefasta, 2024-02-21:
> That warning is normal with no cache (because none of the builds have succeeded recently). I suspect some number of packages are being built because it runs for a few hours before failing, but then [...]

Thank you for explanation. it's very good that binary cache still uploads. But I was just alarmed that when the continuous integration system log constantly returns an error, then we may miss something really important in the future.

lopsided98, 2024-03-05:
I'm keeping this open because there is definitely something wrong with the CI. I just enabled debug logging so maybe that will give some hints.

lopsided98, 2024-03-23:
It is working correctly again as far as I can tell. It was running out of memory while evaluating and there was a bug uploading to cachix.

closed 2024-03-23 by lopsided98