lopsided98/nix-ros-overlay #360: Cachix not working?

Thieso, 2024-02-25:
Hello, 

I am trying to install a ros environment and as far as I understand I should not need to build all the packages when using the binary cache. However, when I use the `flake.nix`:

```
{
  description = "ROS Test";

  nixConfig = {
    substituters = [ 
      "https://nix-community.cachix.org"
	  "https://cache.nixos.org/"
      "https://ros.cachix.org"
     ];
    trusted-public-keys = [
	  "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY="
      "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
      "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo="
    ];
  };

  inputs = {
    flake-utils.url = "github:numtide/flake-utils";
    ros-flake.url = "github:lopsided98/nix-ros-overlay";
  };

  outputs = { self, nixpkgs, flake-utils, ros-flake }: 
      let
        system = "x86_64-linux";
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ros-flake.overlays.default];
        };
      in
      {
        devShells.x86_64-linux.default = pkgs.mkShell {
          name = "ros devenv";
          buildInputs = with pkgs; with pkgs.rosPackages.humble; [
            ros-core
            colcon
            geometry-msgs
            sensor-msgs
            rviz2
          ];
        };
      };
}
```

and then start the shell with 

`nix develop`

it starts building >100 packages which is not what I would expect. Am I doing something wrong?

Thanks for the help

wentasah, 2024-02-25:
It's likely because you try build the ROS packages against a "random" version of nixpkgs and not the version, that was used for populating the cache. Try adding

    nixpkgs.follows = "ros-flake/nixpkgs";

to the inputs of your flake.

BTW, this was the reason why I submitted #359 yesterday :-) However, if I do the change, I still have to compile some packages. I'm not sure why. Maybe, it has something to do with #353.

Thieso, 2024-02-25:
Ah perfect thank you. Yes #359 is fully supported by me. I struggled a bit and created my example from the `flake.nix` files in the issues. 
Thanks for the help.

Thieso, 2024-03-04:
So after all, this does not fix the issue. It creates a different amount of packages to build, such that I assume the new package versions are used but it still builds >100 packages. Can someone confirm that the Cachix is working for them for a simple example?

lopsided98, 2024-03-05:
There's definitely something wrong with the GitHub actions build, so probably many packages aren't being built.

Thieso, 2024-03-05:
Ah I see. I will try to take a look but sadly I am still a beginner in Nix.

Thieso, 2024-03-12:
Ok that works now again, Thank you

closed 2024-03-12 by Thieso