lopsided98/nix-ros-overlay #362: ROS2 humble joint-state-publisher not found after installing

Thieso, 2024-02-27:
Hello, I am trying to install the `joint-state-publisher` package for ros2 humble but when I run `ros2 run joint_state_publisher joint_state_publisher` there is no package found. 

The `flake.nix` I am using is:

```nix
{
    description = "ROS Test";

    nixConfig = {
        substituters = [ 
            "https://nix-community.cachix.org"
            "https://cache.nixos.org/"
            "https://ros.cachix.org"
        ];
        trusted-public-keys = [
            "cache.nixos.org-1:6NCHdD59X431o0gWypbMrAURkbJ16ZPMQFGspcDShjY="
            "nix-community.cachix.org-1:mB9FSh9qf2dCimDSUo8Zy7bkq5CX+/rkCWyvRCYg3Fs="
            "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo="
        ];
    };

    inputs = {
        flake-utils.url = "github:numtide/flake-utils";
        ros-flake.url = "github:lopsided98/nix-ros-overlay";
        nixpkgs.follows = "ros-flake/nixpkgs";
    };

    outputs = { self, nixpkgs, flake-utils, ros-flake }: 
        let
        system = "x86_64-linux";
    pkgs = import nixpkgs {
        inherit system;
        overlays = [ros-flake.overlays.default];
    };
    in
    {
        devShells.x86_64-linux.default = pkgs.mkShell {
            LD_LIBRARY_PATH = "${pkgs.stdenv.cc.cc.lib}/lib";
            name = "ros devenv";
            buildInputs = with pkgs; with pkgs.rosPackages.humble; [
                ros-core
                ros2run
                ros2launch
                colcon
                geometry-msgs
                sensor-msgs
                rviz2
                xacro
                robot-state-publisher
                joint-state-publisher

                python311Packages.pip
                gcc
                glibc.static
                stdenv
            ];
        };
    };
}
```

closed 2024-03-04 by Thieso

TheDelus, 2024-04-19:
I have the same issue. #342 did not fix this issue for me. I am using this `flake.nix`:

```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = with pkgs.rosPackages.humble; [
            pkgs.colcon
            ros-core
            joint-state-publisher
            robot-state-publisher
          ];
        };
      });
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

I am running:

```
$ ros2 run joint_state_publisher joint_state_publisher
Package 'joint_state_publisher' not found
$ echo $AMENT_PREFIX_PATH | grep --color joint_state_publisher
(nothing)
```

lopsided98, 2024-04-20:
It appears this happens because the hooks to setup `AMENT_PREFIX_PATH` are installed by colcon for `ament_python` packages. For `ament_cmake` packages, they are installed through CMake even if you don't use colcon. This is another reason I should probably try to use colcon to build everything.

For now though, it should work if you use `buildEnv`.