lopsided98/nix-ros-overlay #363: Progress of MacOS support?

etherswangel, 2024-02-29:
[RoboStack](https://github.com/RoboStack/ros-humble) is working quite well on aarch64-darwin but I still prefer nix over conda. I wonder what's the current progress and how can I help (if you want)?

lopsided98, 2024-03-03:
I don't use or have access to a macOS machine, so I'm not going to do it myself. I'd accept patches for macOS support as long as they aren't too hard to maintain. In particular, if an upstream package maintainer isn't interested in macOS support, then I won't apply downstream patches to add support.

hegde-atri, 2024-06-18:
@lopsided98 I've got access to a macOS machine, and would like to package this for macOS systems. Completely new to nix, just about learning nix flakes at the moment. Is there any advice or resources you can point me to, to make this happen.

Any help is appreciated!

hegde-atri, 2024-06-20:
Thanks for your advice on approaching this @wentasah I've opened #419 for this.

etherswangel, 2024-06-23:
Thank you for your work @hegde-atri ! I’ll have some time for help starting from July, you could assign me some packages then if needed :D

hegde-atri, 2024-06-28:
Looking forward to your support @etherswangel :P

wentasah, 2026-08-29:
We now have basic MacOS support and CI, so I'm closing this issue. If anyone has problems with Darwin (there are still many), open new issue.

closed 2026-08-29 by wentasah