lopsided98/nix-ros-overlay #369: Is the turtlebot3 example supposed to be graphical?

MatthewCroughan, 2024-03-06:
I have little experience with ROS, but I wanted to check out one of your examples, so I did the following

### Terminal 1

```
user: matthew nix-ros-overlay on  develop 
❯ nix develop .#example-turtlebot3-gazebo

user: matthew nix-ros-overlay on  develop via ❄️  impure (nix-shell-env) 
❯ roslaunch turtlebot3_gazebo turtlebot3_world.launch
... logging to /home/matthew/.ros/log/8ea6a3bd-dba6-11ee-8edd-f875a479a5f8/roslaunch-t480-652383.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
```

### Terminal 2

```
user: matthew nix-ros-overlay on  develop 
❯ nix develop .#example-turtlebot3-gazebo

user: matthew nix-ros-overlay on  develop via ❄️  impure (nix-shell-env) 
❯ roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
... logging to /home/matthew/.ros/log/99595d15-dba6-11ee-b417-f875a479a5f8/roslaunch-t480-652481.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
```

But nothing graphical appears? Am I supposed to use rviz2? 

If I add rviz2 to the devshell, I end up with the following aborted core

```
rviz2 
[INFO] [1709723102.578189725] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1709723102.578322409] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ERROR] [1709723102.612805321] [rviz2]: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at /build/rviz-release-release-humble-rviz_ogre_vendor-11.2.11-1/build/ogre-v1.12.1-prefix/src/ogre-v1.12.1/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 257)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
Aborted (core dumped)
```

MatthewCroughan, 2024-03-06:
Maybe relates to https://github.com/lopsided98/nix-ros-overlay/issues/343#issuecomment-1937104594

lopsided98, 2024-04-11:
e239e71fd2ce40ce1f7499328a4c92e9473fec88

closed 2024-04-11 by lopsided98