lopsided98/nix-ros-overlay #373: can't compile flake example

henrispriet, 2024-03-18:
I tried running the flake example like in the readme, but cmake is failng somehow.
```
Change Dir: '/build/ignition-cmake_0.6.1-source/build/CMakeFiles/CMakeScratch/TryCompile-GLJQ5p'
    
    Run Build Command(s): /nix/store/jacf2kn4dfj99c3ywbvfyg6w53xvxsfl-cmake-3.27.8/bin/cmake -E env VERBOSE=1 /nix/store/05sqpqfnha0pmb5aia3gz968im7n806v-gnumake-4.4.1/bin/make -f Makefile cmTC_fe38f/fast
    /nix/store/05sqpqfnha0pmb5aia3gz968im7n806v-gnumake-4.4.1/bin/make  -f CMakeFiles/cmTC_fe38f.dir/build.make CMakeFiles/cmTC_fe38f.dir/build
    make: /bin/sh: No such file or directory
    make: *** [Makefile:127: cmTC_fe38f/fast] Error 127
```

The commands I ran and the full output below:
```
$ nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo
do you want to allow configuration setting 'extra-substituters' to be set to 'https://ros.cachix.org' (y/N)? y
do you want to permanently mark this value as trusted (y/N)? n
do you want to allow configuration setting 'extra-trusted-public-keys' to be set to 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' (y/N)? y
do you want to permanently mark this value as trusted (y/N)? n
warning: ignoring untrusted substituter 'https://ros.cachix.org'
error: builder for '/nix/store/xaj576mnh461ahp2x2xmb3896297ygc2-ignition-cmake-0.6.1.drv' failed with exit code 1;
       last 10 log lines:
       >
       >
       >
       >
       >   CMake will not be able to correctly generate this project.
       > Call Stack (most recent call first):
       >   CMakeLists.txt:6 (project)
       >
       > 
       > -- Configuring incomplete, errors occurred!
       For full logs, run 'nix log /nix/store/xaj576mnh461ahp2x2xmb3896297ygc2-ignition-cmake-0.6.1.drv'.
error: 1 dependencies of derivation '/nix/store/j83h2a1x82d1yd13xb3clac81pnnnxr4-gazebo-11.14.0.drv' failed to build
error: 1 dependencies of derivation '/nix/store/0hxdpiffymal5j65acshw1h5iwy507bs-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/qj7ckd7g5ydlmh97szi9gp9yf77dvmqk-nix-shell-env.drv' failed to build
~> nix log nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo                                                                                                      1 03/18/24 19:48:48 PM
error: unexpected argument 'develop'
Try 'nix --help' for more information.
```
```
$ nix log /nix/store/xaj576mnh461ahp2x2xmb3896297ygc2-ignition-cmake-0.6.1.drv
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/xaj576mnh461ahp2x2xmb3896297ygc2-ignition-cmake-0.6.1.drv^*'
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/hwns493bz81if40gf8lbp97xzbcajghq-ignition-cmake_0.6.1-source
source root is ignition-cmake_0.6.1-source
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
applying patch /nix/store/lbqn0ixzn0j3d2h6kd23y1881648hagq-fe3100f11073a82a8faf63eb629de9f77fe2b331.patch
patching file cmake/IgnPackaging.cmake
patching file cmake/pkgconfig/ignition.pc.in
patching file cmake/JoinPaths.cmake
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "configurePhase" }
Running phase: configurePhase
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDI>
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /nix/store/khkhbch4p1wjfl1g89gw1mszvvr7bzv0-gcc-wrapper-13.2.0/bin/gcc
-- Check for working C compiler: /nix/store/khkhbch4p1wjfl1g89gw1mszvvr7bzv0-gcc-wrapper-13.2.0/bin/gcc - broken
CMake Error at /nix/store/jacf2kn4dfj99c3ywbvfyg6w53xvxsfl-cmake-3.27.8/share/cmake-3.27/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/nix/store/khkhbch4p1wjfl1g89gw1mszvvr7bzv0-gcc-wrapper-13.2.0/bin/gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/build/ignition-cmake_0.6.1-source/build/CMakeFiles/CMakeScratch/TryCompile-GLJQ5p'
    
    Run Build Command(s): /nix/store/jacf2kn4dfj99c3ywbvfyg6w53xvxsfl-cmake-3.27.8/bin/cmake -E env VERBOSE=1 /nix/store/05sqpqfnha0pmb5aia3gz968im7n806v-gnumake-4.4.1/bin/make -f Makefile cmTC_fe38f/fast
    /nix/store/05sqpqfnha0pmb5aia3gz968im7n806v-gnumake-4.4.1/bin/make  -f CMakeFiles/cmTC_fe38f.dir/build.make CMakeFiles/cmTC_fe38f.dir/build
    make: /bin/sh: No such file or directory
    make: *** [Makefile:127: cmTC_fe38f/fast] Error 127
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
```

lopsided98, 2024-03-18:
This looks like a problem with your system. Can you build other Nix packages?

henrispriet, 2024-03-19:
Yes I can

henrispriet, 2024-03-25:
I've noticed that every time I run these commands (nix develop and then nix log) the directory it tries to change to changes.
Some script somewhere tries to cd to `/build/ignition-cmake_0.6.1-source/build/CMakeFiles/CMakeScratch/TryCompile-xxxxxx` where `xxxxxx` seem to be randomly generated

henrispriet, 2024-03-25:
It was indeed a problem with my installation (I'm sorry to have doubted you :P).
I [uninstalled](https://nixos.org/manual/nix/unstable/installation/uninstall.html) and reinstalled with the [determinate installer](https://github.com/DeterminateSystems/nix-installer).

closed 2024-03-25 by henrispriet