lopsided98/nix-ros-overlay #379: ros-env fails to evaluate: empy_3 no longer exists

kjeremy, 2024-03-27:
https://github.com/lopsided98/nix-ros-overlay/blob/1b9585ae8f529b2b392203f2dda6fb96311621a9/distros/distro-overlay.nix#L9 is causing me some trouble.

I get a `error: attribute 'empy_3' missing`

wentasah, 2024-03-27:
Which command/Nix expression are you trying to run/evaluate? My projects based on humble work fine even with the latest `develop` branch.

kjeremy, 2024-05-09:
I think this is a local issue

kjeremy, 2024-05-24:
Okay I think I'm overriding python incorrectly. If I add the following to an overlay then I get the error about `empy_3`. This worked a long time ago so maybe there's a different way to override python now?

```nix
python3 = super.python3.override {};
```

PR #466 (2024-09-14): Allow changing the Python version in ROS distros
This PR reworks the Python overlay system to make changing the Python version used for a ROS distro easier.

When the top-level `rosPython` package is defined in a distro scope, it will be used instead of `python3`.

For example:

```nix
self: super:

{
  rosPackages = super.rosPackages // {
    jazzy = super.rosPackages.jazzy.overrideScope (rosSelf: rosSuper: {
      rosPython = self.python311;
    });
  };
}
```

Closes #413.
Closes #379.
closed 2024-09-14 by lopsided98