lopsided98/nix-ros-overlay #382: package franka-hw failed to build

henrispriet, 2024-03-31:
below is the nix log output
```
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/r4jdhjain6f995ydhga6x0z40wfv2wm2-0.10.1-1.tar.gz
source root is franka_ros-release-release-noetic-franka_hw-0.10.1-1
setting SOURCE_DATE_EPOCH to timestamp 1663333969 of file franka_ros-release-release-noetic-franka_hw-0.10.1-1/test/main.cpp
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "configurePhase" }
Running phase: configurePhase
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCM
AKE_INSTALL_LOCALEDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-
noetic-franka-hw-0.10.1-r1/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/fmmd8jg6mrwd9y
ajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/share/doc/franka_hw -DCMAKE_INSTALL_INFODIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/share/info -DCMA
KE_INSTALL_MANDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noet
ic-franka-hw-0.10.1-r1/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/include -DCMAKE_INSTALL_SBINDIR=/nix/store/fmmd8jg6m
rwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/bin -DCMAKE_INSTALL_NAME_DI
R=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-hw-0.10.1-r1/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/s
tore/qhpw32pz39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0/bin/strip -DCMAKE_RANLIB=/nix/store/qhpw32pz39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0/bin/ranlib -DCMAKE_AR=/nix/store/qhpw32p
z39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/fmmd8jg6mrwd9yajsw5s8viha8rs7j0c-ros-noetic-franka-
hw-0.10.1-r1  -DCATKIN_ENABLE_TESTING=OFF -DSETUPTOOLS_DEB_LAYOUT=OFF -DCATKIN_BUILD_BINARY_PACKAGE=ON 
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/qhpw32pz39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/qhpw32pz39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /build/franka_ros-release-release-noetic-franka_hw-0.10.1-1/build/devel
-- Using CMAKE_PREFIX_PATH: /nix/store/zsqcm0ssk6il5f0jk3maki3dzlfih4fj-ros-noetic-catkin-0.8.10-r1;/nix/store/9walp7knvqvmnpp9pqwjsdpnx2524ack-cmake-3.28.3;/nix/store/in0nn2h0gk07dixsb
2sv09h3jzq4ylxg-catkin-setup-hook;/nix/store/zz3vpam6m1bj81ss81crdb23mr2igvm9-wrap-python-hook;/nix/store/0g7fmf187mbjdjrwlyqhhd2gj4kxa2c1-make-shell-wrapper-hook;/nix/store/4zdzik0m1wg
q9yxad32k68lggdxlxwyg-die-hook;/nix/store/zbbp7rpnlv7jk8ynj07zjrqsli9jm7y6-gtest-1.14.0-dev;/nix/store/3bjj50wk1kw5angrm9kwnwp2yhcpvg8i-gtest-1.14.0;/nix/store/z9l762xi10nqk128w4jsysr9c
0p3lfic-python3.11-catkin_pkg-0.5.2;/nix/store/kl3yqkx4qydciyllnm0hl3pjzdk4kba8-python3.11-python-dateutil-2.8.2;/nix/store/ybf9yscf5yw09r70sgb1gqh5rraxc9va-python3.11-six-1.16.0;/nix/s
tore/3v2ch16fkl50i85n05h5ckss8pxx6836-python3-3.11.8;/nix/store/mfihnifp4kcydmjasxjmcjwk6pz3lq4k-python3.11-docutils-0.20.1;/nix/store/rpjaxwvz4v3klw3gh3ybqrxd9f7ky6zn-python3.11-pypars
ing-3.1.1;/nix/store/bvqadys5w6dallgim0y8y98h05b7kfn7-python3.11-setuptools-69.0.3;/nix/store/aj3p5ac7pjq09wa6arm5id5djqi05b8y-python3.11-empy-4.0.1;/nix/store/kf1g3jwl86fgps4zv62yk8wrv
czwjvfi-python3.11-nose-1.3.7;/nix/store/w5l59ivm9kh9l3gsb566xa5y9qpx04pa-python3.11-coverage-7.3.2;/nix/store/38q7flpba3qvndx07nc6z3hdhbq3bfsa-python3.11-cffi-1.16.0;/nix/store/vl4i7c2
09j460iyhdxabzsv6qwb95gqj-python3.11-pycparser-2.21;/nix/store/3125ahv429pk8sxnfhj5l3f2ph7jllrk-patchelf-0.15.0;/nix/store/zfgzsn57hhv547x3qmxgpf9y8a320mjr-update-autotools-gnu-config-s
cripts-hook;/nix/store/qhpw32pz39y6i30b3vrbw5fw6zv5549f-gcc-wrapper-13.2.0;/nix/store/7hhd9smnjspppk4k6n47bkw64fdbgbrz-binutils-wrapper-2.41;/nix/store/0qh5h72fvampz8z8ys8g8rrjwx46v0j6-
ros-noetic-message-generation-0.4.1-r1;/nix/store/qnahpns5ig59wiv54n51igcazlnbnayq-ros-noetic-gencpp-0.7.0-r1;/nix/store/054xi1p0vdk7bi7h458b4bmafrj85w50-ros-noetic-genmsg-0.6.0-r1;/nix
/store/57k87h5lgl9imwnd4hlfa9c1p169nda4-ros-noetic-geneus-3.0.0-r1;/nix/store/4wsgvwfqs5whmzc7g2piy9lb2y6n0vfm-ros-noetic-genlisp-0.4.18-r1;/nix/store/xy7dq5fyh7hhyllsd1nlym4ad6aps5sm-r
os-noetic-gennodejs-2.0.2-r1;/nix/store/146zk1w4hwa0hia5hz91a8c0vkgkw74x-ros-noetic-genpy-0.6.15-r1;/nix/store/40cx6kcq46a14h4sc5zxc3fdjzi41nx8-python3.11-pyyaml-6.0.1;/nix/store/sbdndh
v1j6ypwgiyw21n0i29kwq2izsd-ros-noetic-actionlib-1.14.0-r1;/nix/store/1dw0mf7krwsdvdjdk83m6sqmpvnvsz3a-ros-noetic-actionlib-msgs-1.13.1-r1;/nix/store/576pm5pisjgl9l87zhk4gla67apf947v-ros
-noetic-message-runtime-0.4.13-r1;/nix/store/i5sf1ja1rccfka55wga9z7r01vsp75yn-ros-noetic-cpp-common-0.7.3-r1;/nix/store/56x7vks6w9p7rp7hfznri8h9hkl4i389-boost-1.81.0-dev;/nix/store/pn9j
9wk9bnda8vz4x6qxqzndjmnvvzys-boost-1.81.0;/nix/store/k2lh9bhc0s1k5isypgnsznlg5l2z4yx4-console-bridge-1.0.2;/nix/store/24hqvm4aizz3f0w6nv0g8pjw4sy0chj4-ros-noetic-roscpp-serialization-0.
7.3-r1;/nix/store/yx0lnsrbvj6q9nxny89k5rqx8jbw7pkd-ros-noetic-roscpp-traits-0.7.3-r1;/nix/store/4kh2msjgpinz0rahcp431k51s5y3gi0n-ros-noetic-rostime-0.7.3-r1;/nix/store/vpgrww2956gafprnz
pvnpnniqnwzxh9s-ros-noetic-std-msgs-0.5.13-r1;/nix/store/sw31klvm0p3vjjc013bq64wnszr8m7yr-ros-noetic-roscpp-1.16.0-r1;/nix/store/2ky6ap3adaqf8zyjql1g1005wn5a676h-ros-noetic-rosconsole-1
.14.3-r1;/nix/store/y0ai9n48fj3w6qv9smy27j95zp9b37dx-apr-1.7.4-dev;/nix/store/m4aqkncsklp5zdhjyig2zfi6vvl7ya1x-apr-1.7.4;/nix/store/nwwhj04pmwva76fj694panb33q5k4fcx-log4cxx-1.2.0;/nix/s
tore/1lwipn0jj6xxxrw244d1z4nz70zdnqf9-ros-noetic-rosbuild-1.15.8-r1;/nix/store/f9knbyc14s2j9gm8azh59315cdm9w402-ros-noetic-rosgraph-msgs-1.11.3-r1;/nix/store/a8mc1r9hq5ckla6xyzqmlv8nsya
ijk8m-ros-noetic-xmlrpcpp-1.16.0-r1;/nix/store/s4xyb988g01hqxdz33j5qjdzhvy2l2ks-ros-noetic-rospy-1.16.0-r1;/nix/store/y8k1qdx0lf8f4rli72jwb3qjvshlxjvk-python3.11-numpy-1.26.4;/nix/store
/sp6vbh42pb0pgkzgsc9g5z4nsm96ybqc-python3.11-rospkg-1.4.0;/nix/store/c86wammj6kcx2j5fmk6nx6y2x6z7jk7l-python3.11-distro-1.8.0;/nix/store/f0rx64gp2xy1mwa0gx4fvfygimn0al1y-ros-noetic-rosg
raph-1.16.0-r1;/nix/store/bfk94qsfmcbg3138sh7ax995ah261sqr-python3.11-netifaces-0.11.0;/nix/store/c3gbmnfjqr80i5wqpy5vw9apxnmy2g38-ros-noetic-roslib-1.15.8-r1;/nix/store/cxcpr9armg46lc3
g4ix10wgx3wjd1558-ros-noetic-ros-environment-1.3.2-r1;/nix/store/4k2jdxx432pw2jhslch2l9116h0bryih-ros-noetic-rospack-2.6.2-r1;/nix/store/a3752l19hc76dmzh1wgrxbsxh6lfx67z-pkg-config-wrap
per-0.29.2;/nix/store/hm7siz11hmv3rgm900b0ygwzmd91xv04-python3.11-rosdep-0.22.1;/nix/store/rnpxv2d3h48pkcbk917b8fbjf6rd983n-python3.11-rosdistro-0.9.0;/nix/store/qivzarwcfinpsjhi7rh278j
mk4gqbrsb-tinyxml2-10.0.0;/nix/store/0vfz2viap2w88887mnmz6n7yhcn2r4l5-ros-noetic-rostest-1.16.0-r1;/nix/store/1rz8b2zkgmh1dzxky9y1fhsnfizqa8mn-ros-noetic-roslaunch-1.16.0-r1;/nix/store/
[... 488 lines cut ...]
                 from include/boost/thread/condition_variable.hpp:16,
[... cut ...]
                 from /build/franka_ros-release-release-noetic-franka_hw-0.10.1-1/src/franka_combined_hw.cpp:3:
include/boost/bind.hpp:36:1: note: '#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp
> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.'
make[1]: *** [CMakeFiles/Makefile2:1147: CMakeFiles/franka_hw.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

lopsided98, 2024-03-31:
Looks like you just need to add `#include <cstdint>` to `resource_helpers.h`. Many packages have needed this fix recently. If you make an upstream PR, I will add the patch to the overlay.

henrispriet, 2024-03-31:
upstream PR is submitted

henrispriet, 2024-04-07:
I tried to fix it in my flake with an overlay on top of nix-ros-overlay. I got franka-hw to build, but then franka-ros would (I think) still try to build with the unpatched version. I ended up just forking nix-ros-overlay, adding the patch there, and then using the fork as my flake input instead of nix-ros-overaly. @lopsided98 would it be useful to submit a PR to add the patch to nix-ros-overlay?

commit 674adb5b (2024-04-07): noetic: add patch to fix franka-hw

Fixes #382