lopsided98/nix-ros-overlay #383: How does the installation process work?

Guelakais, 2024-04-03:
To be honest, either I don't understand it or it's not clearly described. I've read the readme and looked at the examples and unfortunately it's not clear to me. Do you know what I need to put in my config to make the ros2 installation process via `sudo nixos-rebuild switch` work?
```
{ config, pkgs, builtins,stdenv, ... }:
let
  nixvim = import (builtins.fetchGit {url = "https://github.com/nix-community/nixvim";});
in
let 
  rosDocker=import (builtins.fetchGit{url="https://gitlab.com/docker5348/ros2_main_dev_container";});
in
{
  imports = [
    /etc/nixos/hardware-configuration.nix
  ];
  nix={
    gc = {
      automatic = true;
      dates = "daily";
    };
    settings.auto-optimise-store=true;
  };

  # paste your boot config here...
  boot.loader = {
    systemd-boot.enable = true;
    efi.canTouchEfiVariables = true;
    };

  virtualisation.docker.enable = true;
  networking = {
    firewall = {
      enable = true;
      allowedTCPPorts = [];
      allowedUDPPorts = [];
    };
    hostName = "Azrael";
    networkmanager.enable = true;
  };

  # edit as per your location and timezone
  time.timeZone = "Europe/Berlin";

  i18n = {
    defaultLocale = "en_GB.UTF-8";
    extraLocaleSettings = {
      LC_ADDRESS = "de_DE.UTF-8";
      LC_IDENTIFICATION = "de_DE.UTF-8";
      LC_MEASUREMENT = "de_DE.UTF-8";
      LC_MONETARY = "de_DE.UTF-8";
      LC_NAME = "de_DE.UTF-8";
      LC_NUMERIC = "de_DE.UTF-8";
      LC_PAPER = "de_DE.UTF-8";
      LC_TELEPHONE = "de_DE.UTF-8";
      LC_TIME = "de_DE.UTF-8";
      LC_CTYPE="en_US.utf8"; # required by dmenu don't change this
    };
  };

  sound.enable = true;

	services = {

	  printing.enable = true;

	  xserver = {
	    xkb={
	      layout = "eu";
	      variant = "";
	      };
      libinput.enable = true;
	    enable = true;
	    windowManager.i3 = {
	      enable = true;
	      };
      desktopManager = {
		    xterm.enable = true;
		    xfce = {
		      enable = true;
		      noDesktop = true;
		      enableXfwm = false;
		    };
	    };
	    displayManager = {
		    lightdm.enable = true;
		    defaultSession = "xfce+i3";
	    };
	  }; # xserver
	    gvfs.enable = true;
	    gnome.gnome-keyring.enable = true;
	    blueman.enable = true;
	    pipewire = {
	      enable = true;
	      alsa = {
		      enable = true;
		      support32Bit = true;
	      };
	      pulse.enable = true;
	    };
	  };
	  nixpkgs = {
	    config = {
	      allowUnfree = true;
	      pulseaudio = true;
	    };
	  };

	  users.users.azrael = {
	    isNormalUser = true;
	    description = "azrael";
	    extraGroups = [ "networkmanager" "wheel" "docker" ];
	  }; # services

	environment = {
	  variables= {
	    SRC_ENDPOINT = "https://sourcegraph.example.com";
	    SRC_SRC_ACCESS_TOKEN="'cat ${config.users.users.azrael.home}/sourcegraphToken'";
	  };
	  systemPackages = with pkgs; [
	    alacritty
      catimg
      curl
	    cargo
	    clang
	    dmenu
	    docker
	    firefox
	    fuse
	    gcc
      git
      gimp
	    gnome.gnome-keyring
	    gr-framework
	    i3
	    i3status
	    inkscape
	    jabref
	    julia
	    just
	    kate
	    lazygit
	    libclang
	    nerdfonts
      networkmanagerapplet
	    neovim
	    nitrogen
	    pasystray
	    obsidian
	    openjdk
	    openscad-unstable
	    picom
	    pkg-config
	    polkit_gnome
	    pulseaudioFull
	    python311
	    python311Packages.polars
	    python311Packages.numpy
      python311Packages.scipy
      python311Packages.pip
      rofi
	    roboto
	    rustc
	    rustup
	    steam
	    tectonic
	    thunderbird
	    unrar
	    unzip
	    ucs-fonts
	    vcstool
	    vim
	    vscode
	    wget
	    xclip
	    xorg.xhost
	    xorg.libX11
	    x11docker
	    zip
	    (vscode-with-extensions.override {
        vscodeExtensions = with vscode-extensions; [
          bbenoist.nix
          ms-azuretools.vscode-docker
          ms-python.python
          ms-toolsai.jupyter
          ms-toolsai.jupyter-keymap
          ms-toolsai.jupyter-renderers
          ms-toolsai.vscode-jupyter-slideshow
          ms-toolsai.vscode-jupyter-cell-tags
          ms-vscode-remote.remote-ssh
          serayuzgur.crates
          tamasfe.even-better-toml
          ] ++ pkgs.vscode-utils.extensionsFromVscodeMarketplace [{
            name = "remote-ssh-edit";
            publisher = "ms-vscode-remote";
            version = "0.47.2";
            sha256 = "1hp6gjh4xp2m1xlm1jsdzxw9d8frkiidhph6nvl24d0h8z34w49g";
          }];
      })
	  ];
  }; # environment

	programs = {
	  bash.shellAliases = {
	    sudo="sudo ";
	    sshEval="eval $(ssh-agent -s)";
	    nixos_list_generations="sudo nix-env --list-generations --profile /nix/var/nix/profiles/system";
	    nixosRebuild="sudo nixos-rebuild switch";
	    dockerRmContainters="docker ps -a | awk '{print $NF}' | while read -r container_name; do docker rm $container_name; done";
	    dockerRmImages="docker image ls --quiet --no-trunc | while read -r image_id; do docker image rm $image_id; done";
	  };
	  dconf.enable = true;
	  firefox={
	    enable=true;

	  };
	  git.enable = true;
	  neovim = {
	    enable = true;
	    defaultEditor = true;
	  };
	  steam = {
	    enable = true;
	    remotePlay.openFirewall = true;
	    dedicatedServer.openFirewall = true;
	  };
    thunar.enable = true;
	};

	security = {
    polkit.enable = true;
	  rtkit.enable = true;
  };

  systemd={
    tmpfiles.settings = {
      "ros2_docker_folders" = {
        "${config.users.users.azrael.home}/ros2_ws/src" = {d.mode = "0777";};
        "${config.users.users.azrael.home}/microros_ws/src" = {d.mode = "0777";};
        "${config.users.users.azrael.home}/ros2_debug/src"= {d.mode = "0777";};
      };
      "projects_folders"={
        "${config.users.users.azrael.home}/projects"= {d.mode = "0777";};
        "${config.users.users.azrael.home}/projects/rust"= {d.mode = "0777";};
        "${config.users.users.azrael.home}/projects/julia"= {d.mode = "0777";};
        "${config.users.users.azrael.home}/projects/python"= {d.mode = "0777";};
        "${config.users.users.azrael.home}/projects/latex"= {d.mode = "0777";};
        "${config.users.users.azrael.home}/projects/java"= {d.mode = "0777";};
      };
    };
    user.services.polkit-gnome-authentication-agent-1 = {
      description = "polkit-gnome-authentication-agent-1";
      wantedBy = [ "graphical-session.target" ];
      wants = [ "graphical-session.target" ];
      after = [ "graphical-session.target" ];
      serviceConfig = {
        Type = "simple";
        ExecStart =
          "${pkgs.polkit_gnome}/libexec/polkit-gnome-authentication-agent-1";
        Restart = "on-failure";
        RestartSec = 1;
        TimeoutStopSec = 10;
      };
    };
  };
  
  hardware.bluetooth.enable = true;

  # Don't touch this
  system={
    activationScripts={
    script.text =
    ''
       echo "Udo!"
    '';
    };
    stateVersion = "unstable";
    };
}
```

lopsided98, 2024-04-20:
If you are just doing development or running ROS processes manually, you don't put anything in your NixOS config. The NixOS modules are only needed if you want to start ROS processes at boot (i.e. you are running NixOS on your robot).

Otherwise, you just use a shell file like one of the examples and enter the shell using `nix-shell` or `nix develop` if using flakes.

Guelakais, 2024-04-20:
To be honest, it's too complicated for me. I haven't dealt with flakes yet and run my ros2 applications using a [Docker container](https://gitlab.com/docker5348/ros2_main_dev_container) that I have developed. I will look into it when the standard ros2 packages, like [gazebo](https://gazebosim.org/home), [rviz](https://github.com/ros-visualization/rviz) or the [ros2 command line tool](https://github.com/ros2/ros2cli), can be included in the config as normal.

closed 2024-07-06 by lopsided98