lopsided98/nix-ros-overlay #385: package gazebo failed to build

henrispriet, 2024-04-07:
[Patched it in my fork](https://github.com/henrispriet/nix-ros-overlay/commit/c2ac06b43ad0673cdbd276dd36c71cc6cd5a4e2d).
Not entirely sure that it's correct (I don't know if there is some super subtle difference between C++'s and C's booleans), but it builds and I personally don't need to run it.

lopsided98, 2024-04-07:
Builds fine for me, are you using a different nixpkgs version?

kjeremy, 2024-04-19:
I'm seeing the same thing using this in my `flake.lock`:

```json
 "nix-ros": {
      "inputs": {
        "flake-utils": [
          "flake-utils"
        ],
        "nixpkgs": [
          "nixpkgs"
        ]
      },
      "locked": {
        "lastModified": 1712797796,
        "narHash": "sha256-SkCWsnJ6H+LbXt7PLroXCPlb0k52+EG7WKEXtWjuNbI=",
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "rev": "e239e71fd2ce40ce1f7499328a4c92e9473fec88",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "type": "github"
      }
    },

"nixpkgs": {
      "locked": {
        "lastModified": 1712608508,
        "narHash": "sha256-vMZ5603yU0wxgyQeHJryOI+O61yrX2AHwY6LOFyV1gM=",
        "owner": "lopsided98",
        "repo": "nixpkgs",
        "rev": "4cba8b53da471aea2ab2b0c1f30a81e7c451f4b6",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "nix-ros",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
```

kjeremy, 2024-04-19:
The error I am seeing is:

```
[ 66%] Building CXX object gazebo/sensors/CMakeFiles/gazebo_sensors.dir/SensorTypes.cc.o
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp: In member function 'QGVNode* QGVSubGraph::addNode(const QString&)>
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:47:53: error: 'TRUE' was not declared in this scope
   47 |     Agnode_t *node = agnode(_sgraph->graph(), NULL, TRUE);
      |                                                     ^~~~
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp: In member function 'QGVSubGraph* QGVSubGraph::addSubGraph(const Q>
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:69:56: error: 'TRUE' was not declared in this scope
   69 |             ("cluster_" + _name).toLocal8Bit().data(), TRUE);
      |                                                        ^~~~
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:72:71: error: 'TRUE' was not declared in this scope
   72 |         sgraph = agsubg(_sgraph->graph(), _name.toLocal8Bit().data(), TRUE);
```

I haven't tried the patch yet.

lopsided98, 2024-04-19:
That's the latest version of the nix-ros branch, but not the pinned version in `flake.lock`. That version has various issues, notably #365 and the fact that freeimage (a Gazebo dependency) is marked as insecure, which is why I haven't updated the lock file yet.

In fact, gazebo just fails for me at eval time due to freeimage, so I'm assuming you have allowed insecure packages in order to get that build failure.

henrispriet, 2024-04-22:
> In fact, gazebo just fails at eval time for me due to freeimage, so I'm assuming you have allowed insecure packages in order to get that build failure.

Oh yeah, I totally forgot about that.

In regards to the nixpkgs version: i changed my nixpkgs input in my flake to `nixpkgs.follows = "nix-ros-overlay/nixpkgs";` and it builds now. I find it odd though, because I thought all the packages in `nix-ros-overlay` would use their locked nixpkgs input, instead of the separate nixpkgs input that I've defined. I thought that was the whole point of flakes?

henrispriet, 2024-04-22:
Found at the end of [this section of the nixos manual](https://nixos.org/manual/nix/stable/command-ref/new-cli/nix3-flake.html#flake-inputs):
> It is worth noting, however, that it is generally not useful to eliminate transitive nixpkgs flake inputs in this way. **Most flakes provide their functionality through Nixpkgs overlays or NixOS modules, which are composed into the top-level flake's nixpkgs input; so their own nixpkgs input is usually irrelevant.**

Should have read the README more closely, oops.

closed 2024-04-22 by henrispriet