lopsided98/nix-ros-overlay #386: panda-moveit-config causes collision when included in ros-env (buildEnv)

henrispriet, 2024-04-07:
`panda-moveit-config` builds fine on its own, but when it is included in ros-env, it breaks.
```nix
with pkgs.rosPackages.noetic;
pkgs.mkShell {
    packages = [
        (buildEnv {
            paths = [ panda-moveit-config ];
        })
    ];
}
```

Error message:
```
@nix { "action": "setPhase", "phase": "buildPhase" }
Running phase: buildPhase
error: collision between `/nix/store/fylj4985hk0d9ck4jldqzy2jbi9h27na-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py' and `/nix/store/k5hy73mfzdyw9dl9ljg05rzzmf2waqg1-ros-noetic-python-qt-binding-0.4.4-r1/share/python_qt_binding/cmake/sip_configure.py'
```

henrispriet, 2024-04-07:
I believe the issue arises due to [this patch](https://github.com/lopsided98/nix-ros-overlay/blob/1b9585ae8f529b2b392203f2dda6fb96311621a9/distros/ros1-overlay.nix#L144-L181), which overrides some part of `python-qt-bingings` in `rviz`.

lopsided98, 2024-04-07:
Yes, see https://github.com/lopsided98/nix-ros-overlay/issues/301 for more discussion of this problem.

henrispriet, 2024-04-07:
Alright, closing since it's a duplicate.

closed 2024-04-07 by henrispriet