lopsided98/nix-ros-overlay #387: How do I include launch files from a package i packaged myself?

henrispriet, 2024-04-07:
I tried to write a derivation for [this package](https://github.com/rail-berkeley/serl_franka_controllers) myself, using the packages in this repo as an example. But, the `.lauch` files, don't seem to be getting copied into the ros-env when I build it. With other packages (for example realsense2-camera), this does work. What am I missing?

Here is my derivation (FIXME comment is related to #386):
```nix
{ lib
, buildRosPackage
, fetchFromGitHub
, catkin
, message-generation
, eigen
, message-runtime
, controller-interface
, dynamic-reconfigure
, eigen-conversions
, franka-hw
, franka-gripper
, geometry-msgs
, hardware-interface
, tf
, tf-conversions
, libfranka
, pluginlib
, realtime-tools
, roscpp
, franka-control
, franka-description
, panda-moveit-config
, rospy
}:
buildRosPackage rec {
  pname = "serl_franka_controllers";
  version = "0.1.1";

  src = fetchFromGitHub {
    owner = "rail-berkeley";
    repo = "serl_franka_controllers";
    rev = "${version}";
    hash = "sha256-s5lGg93ChHIC/dTUJbm4uBsBfO3v6xZ7DB7qZVnxz2M=";
  };

  buildType = "catkin";
  buildInputs = [
    catkin
    message-generation
    eigen

    message-runtime
  ];
  propagatedBuildInputs = [
    controller-interface
    dynamic-reconfigure
    eigen-conversions
    franka-hw
    franka-gripper
    geometry-msgs
    hardware-interface
    tf
    tf-conversions
    libfranka
    pluginlib
    realtime-tools
    roscpp

    franka-control
    franka-description
    message-runtime
    # FIXME: causes "error: collision between ... and ..." when included in ros-env
    # panda-moveit-config
    rospy
  ];
  nativeBuildInputs = [ catkin ];

  meta = {
    description = "Carteasian impedance controller with reference limiting for Franka Emika Robot";
    license = with lib.licenses; [ mit ];
  };
}
```

lopsided98, 2024-04-07:
That package has the launch file installation commented out for some reason: https://github.com/rail-berkeley/serl_franka_controllers/blob/55f7563d15544caa95828b2968ebd32abb0d3fe4/CMakeLists.txt#L91-L93

henrispriet, 2024-04-07:
Wow, that's strange, thanks for the catch!

closed 2024-04-07 by henrispriet