lopsided98/nix-ros-overlay #393: Sophus doesn't build

martiege, 2024-04-15:
I'm trying to add the Sophus library to my package, which doesn't seem to build for certain ROS distros. In my case, it is specifically the noetic version that doesn't build, but several others seem not to work either. 

I've done some rudimentary testing, and have these results: 

foxy: fails
humble: fails
iron: fails
noetic: fails
rolling: builds

It seems that the Sophus package builds the tests and examples, and this is where it crashes. Is there any way of disabling these?

martiege, 2024-04-15:
`nix log /nix/store/y4jqk0fvm398y2dmff2gbwcsx8v6fcdk-ros-foxy-sophus-1.2.1-r1.drv`:
```
  \\  \\ //      martin@wsdesktop
 ==\\__\\/ //    os     NixOS 24.05 (Uakari)
   //   \\//     host   P10S WS
==//     //==    kernel 6.6.24
 //\\___//       uptime 3h 36m
// /\\  \\==     pkgs   1535
  // \\  \\      memory 9474M / 32028M

~ 
❯ cd /tmp 

/tmp 
❯ mkdir test-ros          

/tmp 
❯ cp ~/dev/test-ros/flake.nix                          
cp: missing destination file operand after '/home/martin/dev/test-ros/flake.nix'
Try 'cp --help' for more information.

/tmp 
✗ cp ~/dev/test-ros/flake.nix.
cp: missing destination file operand after '/home/martin/dev/test-ros/flake.nix.'
Try 'cp --help' for more information.

/tmp 
✗ cp ~/dev/test-ros/flake.nix .
'/home/martin/dev/test-ros/flake.nix' -> './flake.nix'

/tmp 
❯ vim flake.nix      

/tmp took 23s 
❯ nix develop                                    
error (ignored): error: end of string reached
error:
       … while fetching the input 'path:/tmp'

       error: file '/tmp/.ICE-unix/1318' has an unsupported type

/tmp 
✗ gs
fatal: not a git repository (or any of the parent directories): .git

/tmp 
✗ ls
 dbus-10AbgGKi          scoped_dirUz4hhX                                                                    test-ros
 dbus-CGWRqjMY          systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-clamav-daemon.service-WEItow       zellij-1000
 flake.nix              systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-cups.service-nkr1wA               
 graveyard-martin       systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-nscd.service-QCIqJQ               
 hypr                   systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-polkit.service-JNE6Jn             
 migrate_db             systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-syncthing.service-1ngy3n          
 model.model.ce94127b   systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-systemd-logind.service-o1lJ1s     
 nvim.martin            systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-systemd-oomd.service-tqleR6       
 preamble-08aae8.pch    systemd-private-c8d8d2245e774bcc9d57dc64bf29ebfc-systemd-timesyncd.service-TsLwgF  
 preamble-6a43f3.pch    Temp-bdb4ff00-c5b9-4803-ac84-c4ce823c92b4                                          

/tmp 
❯ mv flake.nix test-ros        

/tmp 
❯ cd te              
direnv: loading ~/projects/navtool/dev/tightly_coupled_lidar_camera_odom/.envrc
direnv: using flake
warning: Git tree '/home/martin/projects/navtool/dev/tightly_coupled_lidar_camera_odom' is dirty
error: interrupted by the user
direnv: error signal: killed

tightly_coupled_lidar_camera_odom/aloam_ouster on  master [✘!+] via △ 
[... 79 lines cut ...]
    inlined from 'void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename enable_if<evaluator_assume_aliasing<Src>::value, void*>::type) [with Dst = Eigen::Block<Eigen::Matrix<float, 7, 1, 0, 7, 1>, -1, 1, false>; Src = Eigen::Product<Eigen::Matrix<float, 3, 3>, Eigen::Matrix<float, 3, 1>, 0>; Func = assign_op<float, float>]' at /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/AssignEvaluator.h:851:27,
    inlined from 'void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Block<Eigen::Matrix<float, 7, 1, 0, 7, 1>, -1, 1, false>; Src = Eigen::Product<Eigen::Matrix<float, 3, 3>, Eigen::Matrix<float, 3, 1>, 0>]' at /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/AssignEvaluator.h:836:18,
    inlined from 'Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Product<Eigen::Matrix<float, 3, 3>, Eigen::Matrix<float, 3, 1>, 0>; Derived = Eigen::Block<Eigen::Matrix<float, 7, 1, 0, 7, 1>, -1, 1, false>]' at /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/Assign.h:66:28,
    inlined from 'Sophus::Sim3Base<Derived>::Tangent Sophus::Sim3Base<Derived>::log() const [with Derived = Sophus::Sim3<float>]' at /build/sophus-release-release-foxy-sophus-1.2.1-1/sophus/sim3.hpp:160:23:
/nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/lib/gcc/x86_64-unknown-linux-gnu/13.2.0/include/xmmintrin.h:940:23: error: array subscript '__m128_u[0]' is partly outside array bounds of 'Eigen::internal::plain_matrix_type<Eigen::Product<Eigen::Matrix<float, 3, 3>, Eigen::Matrix<float, 3, 1>, 0>, Eigen::Dense>::type [1]' {aka 'Eigen::Matrix<float, 3, 1> [1]'} [-Werror=array-bounds=]
  940 |   return *(__m128_u *)__P;
      |                       ^~~
In file included from /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/Core:278:
/nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In member function 'Sophus::Sim3Base<Derived>::Tangent Sophus::Sim3Base<Derived>::log() const [with Derived = Sophus::Sim3<float>]':
/nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/AssignEvaluator.h:850:41: note: object 'tmp' of size 12
  850 |   typename plain_matrix_type<Src>::type tmp(src);
      |                                         ^~~
[ 66%] Linking CXX executable test_geometry
[ 66%] Built target test_geometry
cc1plus: all warnings being treated as errors
make[2]: *** [test/core/CMakeFiles/test_sim3.dir/build.make:76: test/core/CMakeFiles/test_sim3.dir/test_sim3.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:404: test/core/CMakeFiles/test_sim3.dir/all] Error 2
[ 70%] Linking CXX executable test_se3
[ 70%] Built target test_se3
[ 75%] Linking CXX executable test_rxso3
[ 75%] Built target test_rxso3
[ 79%] Linking CXX executable test_so3
[ 79%] Built target test_so3
make: *** [Makefile:146: all] Error 2
```

martiege, 2024-04-15:
`nix log /nix/store/y4jqk0fvm398y2dmff2gbwcsx8v6fcdk-ros-foxy-sophus-1.2.1-r1.drv`:
```
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/lz8pxzzhj8626h48b8xqh042b1p8mi6y-1.2.1-1.tar.gz
source root is sophus-release-release-foxy-sophus-1.2.1-1
setting SOURCE_DATE_EPOCH to timestamp 1595736626 of file sophus-release-release-foxy-sophus-1.2.1-1/test/core/tests.hpp
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "configurePhase" }
Running phase: configurePhase
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/share/doc/Sophus -DCMAKE_INSTALL_INFODIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/share/info -DCMAKE_INSTALL_MANDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/include -DCMAKE_INSTALL_SBINDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/bin -DCMAKE_INSTALL_NAME_DIR=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/strip -DCMAKE_RANLIB=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/ranlib -DCMAKE_AR=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/hx43h0870mwmh4yarlbr82pk6wrdfraj-ros-foxy-sophus-1.2.1-r1  
[0mCMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[0m
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3 (Required is at least version "3.3.0") 
-- Configuring done (1.6s)
-- Generating done (0.1s)
[33mCMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTING
    CMAKE_POLICY_DEFAULT_CMP0025

[0m
-- Build files have been written to: /build/sophus-release-release-foxy-sophus-1.2.1-1/build
cmake: enabled parallel building
cmake: enabled parallel installing
@nix { "action": "setPhase", "phase": "buildPhase" }
Running phase: buildPhase
build flags: -j8 SHELL=/nix/store/a1s263pmsci9zykm5xcdf7x9rv26w6d5-bash-5.2p26/bin/bash
[  4%] [32mBuilding CXX object test/core/CMakeFiles/test_common.dir/test_common.cpp.o[0m
[  8%] [32mBuilding CXX object test/core/CMakeFiles/test_so2.dir/test_so2.cpp.o[0m
[ 12%] [32mBuilding CXX object test/core/CMakeFiles/test_sim2.dir/test_sim2.cpp.o[0m
[ 16%] [32mBuilding CXX object test/core/CMakeFiles/test_rxso2.dir/test_rxso2.cpp.o[0m
[ 20%] [32mBuilding CXX object test/core/CMakeFiles/test_se2.dir/test_se2.cpp.o[0m
[ 25%] [32mBuilding CXX object test/core/CMakeFiles/test_rxso3.dir/test_rxso3.cpp.o[0m
[ 29%] [32mBuilding CXX object test/core/CMakeFiles/test_so3.dir/test_so3.cpp.o[0m
[ 33%] [32mBuilding CXX object test/core/CMakeFiles/test_se3.dir/test_se3.cpp.o[0m
[ 37%] [32m[1mLinking CXX executable test_common[0m
[ 37%] Built target test_common
[ 41%] [32mBuilding CXX object test/core/CMakeFiles/test_sim3.dir/test_sim3.cpp.o[0m
[ 45%] [32m[1mLinking CXX executable test_so2[0m
[ 45%] Built target test_so2
[ 50%] [32mBuilding CXX object test/core/CMakeFiles/test_velocities.dir/test_velocities.cpp.o[0m
In file included from [01m[K/nix/store/a3mmr5jmrd0sjvmnc9vqvs388ppq1bnf-gcc-13.2.0/lib/gcc/x86_64-unknown-linux-gnu/13.2.0/include/emmintrin.h:31[m[K,
                 from [01m[K/nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:346[m[K,
[... cut ...]
make[2]: *** [test/core/CMakeFiles/test_sim3.dir/build.make:76: test/core/CMakeFiles/test_sim3.dir/test_sim3.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:404: test/core/CMakeFiles/test_sim3.dir/all] Error 2
[ 70%] [32m[1mLinking CXX executable test_se3[0m
[ 70%] Built target test_se3
[ 75%] [32m[1mLinking CXX executable test_rxso3[0m
[ 75%] Built target test_rxso3
[ 79%] [32m[1mLinking CXX executable test_so3[0m
[ 79%] Built target test_so3
make: *** [Makefile:146: all] Error 2
```

martiege, 2024-04-15:
`nix log /nix/store/lvvscf1yq95k82jaf4rkb0i53av4j2x2-ros-humble-sophus-1.3.2-r1.drv`:
```
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/rrx0y0am5ryfikbzd3xdkpz0lgvqmlq1-1.3.2-1.tar.gz
source root is sophus-release-release-humble-sophus-1.3.2-1
setting SOURCE_DATE_EPOCH to timestamp 1710767256 of file sophus-release-release-humble-sophus-1.3.2-1/test/core/tests.hpp
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "configurePhase" }
Running phase: configurePhase
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/share/doc/Sophus -DCMAKE_INSTALL_INFODIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/share/info -DCMAKE_INSTALL_MANDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/include -DCMAKE_INSTALL_SBINDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/bin -DCMAKE_INSTALL_NAME_DIR=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_OSX_SYSROOT= -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/strip -DCMAKE_RANLIB=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/ranlib -DCMAKE_AR=/nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/izjxixcq54nsfrf9ddj2anv7yznhfd8n-ros-humble-sophus-1.3.2-r1  
[0mCMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

[0m
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/ac1hb5dc2z4biwgy8mjrhlifffkkrvdq-gcc-wrapper-13.2.0/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMake version: 3.28.3
-- Found Eigen3: /nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3 (Required is at least version "3.3.0") 
[33mCMake Warning at CMakeLists.txt:61 (find_package):
  By not providing "Findfmt.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "fmt", but
  CMake did not find one.

  Could not find a package configuration file provided by "fmt" with any of
  the following names:

    fmtConfig.cmake
    fmt-config.cmake

  Add the installation prefix of "fmt" to CMAKE_PREFIX_PATH or set "fmt_DIR"
  to a directory containing one of the above files.  If "fmt" provides a
  separate development package or SDK, be sure it has been installed.

[0m
-- Turning basic logging ON
-- Configuring done (0.5s)
-- Generating done (0.0s)
[33mCMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTING
    CMAKE_POLICY_DEFAULT_CMP0025

[0m
-- Build files have been written to: /build/sophus-release-release-humble-sophus-1.3.2-1/build
cmake: enabled parallel building
cmake: enabled parallel installing
@nix { "action": "setPhase", "phase": "buildPhase" }
Running phase: buildPhase
build flags: -j8 SHELL=/nix/store/a1s263pmsci9zykm5xcdf7x9rv26w6d5-bash-5.2p26/bin/bash
[  3%] [32mBuilding CXX object test/core/CMakeFiles/test_common.dir/test_common.cpp.o[0m
[... 207 lines cut ...]
    inlined from '[01m[KPacketType Eigen::internal::evaluator<Eigen::PlainObjectBase<Derived> >::[01;32m[Kpacket[m[K(Eigen::Index, Eigen::Index) const [35m[K[with int LoadMode = 0; PacketType = __vector(4) float; Derived = Eigen::Matrix<float, 3, 1>][m[K[m[K' at [01m[K/nix/store/d2gm661bgmg42x7q642i74m38sknndsf-eigen-3.4.0/include/eigen3/Eigen/src/Core/CoreEvaluators.h:238:42[m[K,
[... cut ...]
[01m[K/build/sophus-release-release-humble-sophus-1.3.2-1/test/core/test_rxso3.cpp:130:39:[m[K [01;36m[Knote: [m[Kobject '[01m[Kso3_tangent[m[K' of size 12
  130 |       const typename SO3Type::Tangent [01;36m[Kso3_tangent[m[K =
      |                                       [01;36m[K^~~~~~~~~~~[m[K
cc1plus: all warnings being treated as errors
make[2]: *** [test/core/CMakeFiles/test_rxso3.dir/build.make:76: test/core/CMakeFiles/test_rxso3.dir/test_rxso3.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:434: test/core/CMakeFiles/test_rxso3.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

wentasah, 2024-04-16:
See #394.

wentasah, 2026-08-29:
Sophus can be compiled now. Closing.

closed 2026-08-29 by wentasah