lopsided98/nix-ros-overlay #396: Can't use `rmw_cyclonedds_cpp` with ros2 `humble` because path to `*.so` can't be resolved

zatevakhin, 2024-04-23:
Trying to list topics for example, or use any `ros2` command:
```
# ivan @ nixos in /projects/work/nix-ros [12:53:18] 
$ nix develop   
You use 'humble' with 'rmw_cyclonedds_cpp' on domain '1'.

[ivan@nixos:/projects/work/nix-ros]$ ros2 topic list
[ERROR] [1713873209.974130528] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_cyclonedds_cpp'), with error message 'failed to load shared library 'librmw_cyclonedds_cpp.so' due to dlopen error: librmw_cyclonedds_cpp.so: cannot open shared object file: No such file or directory, at /build/rcutils-release-release-humble-rcutils-5.1.5-1/src/shared_library.c:99, at /build/rmw_implementation-release-release-humble-rmw_implementation-2.8.2-1/src/functions.cpp:65', exiting with 1., at /build/rcl-release-release-humble-rcl-5.3.7-1/src/rcl/rmw_implementation_identifier_check.c:139
```

Here my `flake.nix` which I use:
```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        ROS_DISTRO = "humble";
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = with pkgs.rosPackages.${ROS_DISTRO}; [
            pkgs.colcon
            ros-core
            rmw-cyclonedds-cpp
          ];
          
          RMW_IMPLEMENTATION = "rmw_cyclonedds_cpp";
          ROS_DOMAIN_ID = 1;
          shellHook = ''
          echo "You using '$ROS_DISTRO' with '$RMW_IMPLEMENTATION' on domain '$ROS_DOMAIN_ID'."
          '';
        };
      });

  nixConfig = {
    trusted-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

I found the `librmw_cyclonedds_cpp.so` exists in the Nix store, but it seems it can't be linked, mb something wrong with `nix-ld`.
```
# ivan @ nixos in /projects/work/nix-ros [13:00:37] C:1
$ locate librmw_cyclonedds_cpp.so
/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/lib/librmw_cyclonedds_cpp.so
```

zatevakhin, 2024-04-23:
Checked my `env` and path to `/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/` is present in next vars:
```
AMENT_PREFIX_PATH= ... :/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1: ...
CMAKE_PREFIX_PATH= ... :/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1: ...
XDG_DATA_DIRS= ... :/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/share: ...
NIX_LDFLAGS=-rpath /projects/work/nix-ros/outputs/out/lib  ... -L/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/lib ... -L/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/lib 
nativeBuildInputs=/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1 ...
CMAKE_LIBRARY_PATH=... :/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/lib: ... :/nix/store/nvgv329v69kmb2j5bvwzmir9xyjy9kyd-ros-humble-rmw-cyclonedds-cpp-1.3.4-r1/lib: ...
```

wentasah, 2024-04-27:
Wrap your ROS packages to `buildEnv` like this:
```nix
          packages = with pkgs.rosPackages.${ROS_DISTRO}; [
            pkgs.colcon
            (buildEnv {
              paths = [
                ros-core
                rmw-cyclonedds-cpp
              ];
            })
          ];
```
This ensures that `LD_LIBRARY_PATH` is set correctly.

We should fix the example `flake.nix` as many people seem to experience similar problems. I'll do it.

zatevakhin, 2024-04-29:
@wentasah Thanks! Works perfectly for me. Now is no need to set `LD_LIBRARY_PATH` manually.

PR #402 (2024-07-20): README: Update example flake.nix to use buildEnv
Currently, quite a few things are broken when ROS packages are used directly without buildEnv.

Closes #396 #401
Related #362
commit ab4c9ba0 (2024-07-20): README: Update example flake.nix to use buildEnv

Currently, quite a few things are broken when ROS packages are used
directly without buildEnv.

Closes #396 #401
Related #362
closed 2024-04-29 by zatevakhin