lopsided98/nix-ros-overlay #401: RQt fail to run

fromtheeast710, 2024-04-27:
I install RQt from ROS-Humble but it failed to run with the following error:

``` $ rqt
Traceback (most recent call last):
  File "/nix/store/86shmvdxbvjdka85bjhi0vcyy0zz1yqw-python3.11-ros-humble-rqt-gui-1.1.7-r1/bin/.rqt-wrapped", line 34, in <module>
    sys.exit(load_entry_point('rqt-gui==1.1.7', 'console_scripts', 'rqt')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/86shmvdxbvjdka85bjhi0vcyy0zz1yqw-python3.11-ros-humble-rqt-gui-1.1.7-r1/lib/python3.11/site-packages/rqt_gui/main.py", line 94, in main
    sys.exit(Main().main())
             ^^^^^^^^^^^^^
  File "/nix/store/86shmvdxbvjdka85bjhi0vcyy0zz1yqw-python3.11-ros-humble-rqt-gui-1.1.7-r1/lib/python3.11/site-packages/rqt_gui/main.py", line 62, in main
    return super(Main, self).main(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/9b9wrlfdvi9a1dih0bwgb5j8vk0hbxz0-ros-humble-qt-gui-2.2.3-r2/lib/python3.11/site-packages/qt_gui/main.py", line 455, in main
    app = self.create_application(argv)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/86shmvdxbvjdka85bjhi0vcyy0zz1yqw-python3.11-ros-humble-rqt-gui-1.1.7-r1/lib/python3.11/site-packages/rqt_gui/main.py", line 71, in create_application
    rqt_gui_path = get_package_path('rqt_gui')
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/9b9wrlfdvi9a1dih0bwgb5j8vk0hbxz0-ros-humble-qt-gui-2.2.3-r2/lib/python3.11/site-packages/qt_gui/ros_package_helper.py", line 35, in get_package_path
    _, package_path = get_resource('packages', name)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/74vayia0ypi5z4spac48xqkki1lk7ih8-python3.11-ros-humble-ament-index-python-1.4.0-r2/lib/python3.11/site-packages/ament_index_python/resources.py", line 86, in get_resource
    raise LookupError(
LookupError: Could not find the resource 'rqt_gui' of type 'packages'
```

Here's my `flake.nix`

```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs"; # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = with pkgs.rosPackages.humble; [
            ros-core
            rviz2
            # turtlesim
            gazebo-dev
            gazebo-ros
            ackermann-msgs
            python-qt-binding
            qt-gui
            rqt
            rqt-gui
            rqt-gui-py
            rqt-gui-cpp
            rqt-plot
            rqt-action
            rqt-common-plugins
            yaml-cpp-vendor
            ignition-cmake2-vendor
            tf2
            tf2-ros

            python3Packages.rosdep
            python3Packages.pandas
            python3Packages.pillow
            python3Packages.matplotlib
            python3Packages.scipy

            pkgs.colcon
            pkgs.gazebo
            pkgs.qt5ct
            pkgs.qt5.qtwayland
            pkgs.ogre
          ];
      
          shellHook = ''
            # . /home/theeast/.bashrc
            . /home/theeast/Desktop/Robo/Nix/sim/install/setup.bash
          '';
        };
      });
  
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

bjsowa, 2024-04-27:
Had the same isue. I think the problem is that `ament_python` packages are not added to `AMENT_PREFIX_PATH` variable (see [this answer](https://github.com/lopsided98/nix-ros-overlay/issues/362#issuecomment-2067777696)). The solution for now is to use `buildEnv` like in the examples to put all of the ROS packages into a single path.

fromtheeast710, 2024-04-28:
Thank you for the suggestion, it works now.

closed 2024-04-28 by fromtheeast710