lopsided98/nix-ros-overlay #423: mavros-extras build error

drawnwren, 2024-07-06:
With the example flake.nix and `mavros-extras`, I get the following build error:
```
/build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/mag_calibration_status.cpp:64:22: error: 'bitset' is not a member of 'std'
   64 |                 std::bitset<8> compass_calibrating = mp.cal_mask;
      |                      ^~~~~~
/build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/mag_calibration_status.cpp:14:1: note: 'std::bitset' is defined in header '<bitset>'; did you forget to '#include <bitset>'?
   13 | #include <mavros_msgs/MagnetometerReporter.h>
  +++ |+#include <bitset>
   14 | namespace mavros {
/build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/mag_calibration_status.cpp:64:32: error: 'compass_calibrating' was not declared in this scope
   64 |                 std::bitset<8> compass_calibrating = mp.cal_mask;
      |                                ^~~~~~~~~~~~~~~~~~~
In file included from include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from include/boost/smart_ptr/shared_ptr.hpp:29,
                 from include/boost/shared_ptr.hpp:17,
                 from /nix/store/3wabmag91va0i32iyyrgq2j9s065r1xc-ros-noetic-roscpp-1.16.0-r1/include/ros/forwards.h:37,
                 from /nix/store/3wabmag91va0i32iyyrgq2j9s065r1xc-ros-noetic-roscpp-1.16.0-r1/include/ros/node_handle.h:31,
                 from /nix/store/bjmc08cdif7yjbnfdiq17id719qa8sdx-ros-noetic-diagnostic-updater-1.11.0-r1/include/diagnostic_updater/diagnostic_updater.h:42,
                 from /nix/store/6zpp1dqwv10j0kqnkmj3k7nps2wpkp8x-ros-noetic-mavros-1.18.0-r1/include/mavros/mavros_plugin.h:23,
                 from /build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/vision_speed_estimate.cpp:18:
include/boost/bind.hpp:36:1: note: '#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.'
make[2]: *** [CMakeFiles/mavros_extras.dir/build.make:300: CMakeFiles/mavros_extras.dir/src/plugins/mag_calibration_status.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/trajectory.cpp: In member function 'void mavros::extra_plugins::TrajectoryPlugin::handle_trajectory(const mavlink::mavlink_message_t*, mavlink::common::msg::TRAJECTORY_REPRESENTATION_WAYPOINTS&)':
/build/mavros-release-release-noetic-mavros_extras-1.18.0-1/src/plugins/trajectory.cpp:353:52: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
  353 |                         tr_desired->point_valid[i] = true;
In file included from /nix/store/3wabmag91va0i32iyyrgq2j9s065r1xc-ros-noetic-roscpp-1.16.0-r1/include/ros/message.h:38,
                 from /nix/store/3wabmag91va0i32iyyrgq2j9s065r1xc-ros-noetic-roscpp-1.16.0-r1/include/ros/publisher.h:33,
                 from /nix/store/3wabmag91va0i32iyyrgq2j9s065r1xc-ros-noetic-roscpp-1.16.0-r1/include/ros/node_handle.h:32:
include/boost/array.hpp:62:11: note: at offset 5 into destination object 'boost::array<unsigned char, 5>::elems' of size 5
make[1]: *** [CMakeFiles/Makefile2:2380: CMakeFiles/mavros_extras.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

drawnwren, 2024-07-06:
The offending flake:
```
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = with pkgs.rosPackages.noetic; [
            pkgs.colcon
            ros-core
            mavros
            mavros-extras
          ];
        };
      });
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

wentasah, 2024-07-07:
This should be fixed by #426.

PR #426 (2024-07-07): noetic.mavros-extras: Add gcc-13 patch
Fixes #423.
commit 06c257eb (2024-07-07): noetic.mavros-extras: Add gcc-13 patch

Fixes #423.
closed 2024-07-07 by lopsided98