lopsided98/nix-ros-overlay #424: ros-humble-rosbag2 failed to build

fromtheeast710, 2024-07-07:
My flake was working until I updated and then it fails with the following log:
```
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/dvvl6g36wdaasmaxd1kgrvsb3camsg9k-0.15.11-1.tar.gz
source root is rosbag2-release-release-humble-rosbag2_compression-0.15.11-1
setting SOURCE_DATE_EPOCH to timestamp 1716395495 of file rosbag2-release-release-humble-rosbag2_compres>
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
applying patch /nix/store/awmpidljk3y90q4vxqdn4vhcv1basf43-65c889e1fa55dd85a148b27b8c27dadc73238e67.patch
patching file include/rosbag2_compression/compression_options.hpp
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file include/rosbag2_compression/compression_options.hpp.rej
```
Below is my flake:
```nix
{
  description = "FSDS Simulator setup";

  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs"; # IMPORTANT!!!
  };

  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
        msgpack-old = { buildPythonPackage, fetchPypi }:
          buildPythonPackage rec {
            pname = "msgpack-python";
            version = "0.5.6";
            src = fetchPypi {
              inherit pname version;
              hash = "sha256-N4zIptNUW1Mt/RSdpxWrrk/aKjrbbXTlJdDV5R9GkJs=";
            };
            doCheck = false;
          };
        msgpack-rpc = { buildPythonPackage, fetchPypi }:
          buildPythonPackage rec {
            pname = "msgpack-rpc-python";
            version = "0.4.1";
            src = fetchPypi {
              inherit pname version;
              hash = "sha256-unGIEp8rqV+5cx+QaqgqWr1YJ91U7PxxUAj1ziMUU18=";
            };
            doCheck = false;
          };
        tornado4 = { buildPythonPackage, fetchPypi }:
          buildPythonPackage rec {
            pname = "tornado";
            version = "4.5.3";
            src = fetchPypi {
              inherit pname version;
              hash = "sha256-bRTkfqsOFXmc883MhrC5gnnaaFIsqs4r185kQodoXwo=";
            };
            doCheck = false;
          };
      in with pkgs; {
        devShells.default = mkShell {
          name = "FSDS Simulator";

          packages = [
            colcon
            # colcon-clean
            cmake
            opencv
            clang_12
            clang-tools_12
            yaml-cpp
            unzip
            curl
            qt5ct

            (with rosPackages.humble; buildEnv {
              paths = [
                ros-core
                cv-bridge
                image-transport
                joy
                tf2
                tf2-ros
                rqt-gui
                rqt-common-plugins
                foxglove-bridge
                foxglove-msgs

                ament-cmake-core
                python-cmake-module
              ] ++ (with python3Packages; [ 
                (callPackage msgpack-rpc { })
                (callPackage msgpack-old { })
                (callPackage tornado4 { })

                numpy opencv4 matplotlib
              ]);})
          ];

          shellHook = ''
          . Formula-Student-Driverless-Simulator/ros2/install/setup.bash 
          '';
        };}
    );
}
```

wentasah, 2024-07-07:
#425 should fix it.

lopsided98, 2024-07-07:
The real problem is that you should be using the master branch if you don't want to hit issues like this. Whenever the overlay is updated, a large number of these trivial problems occur, but I can't find them until I let my Hydra instance build everything.

I could switch the default branch on GitHub to master, but then everyone makes PRs against master instead of develop. Perhaps this is an easier problem to solve though.

closed 2024-07-07 by lopsided98

fromtheeast710, 2024-07-08:
@lopsided98 Could you add the branch to the example flake's input then? Like:

```nix
inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
};
```

PR #425 (2024-07-07): rosbag2-compression: Remove patch, which is now in a released version
The patch causes build failure.

Fixes #424.
commit a91bd836 (2024-07-07): rosbag2-compression: Remove patch, which is now in a released version

The patch causes build failure.

Fixes #424.