lopsided98/nix-ros-overlay #427: gazebo build issue

drawnwren, 2024-07-07:
```
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:47:53: error: 'TRUE' was not declared in this scope
   47 |     Agnode_t *node = agnode(_sgraph->graph(), NULL, TRUE);
      |                                                     ^~~~
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp: In member function 'QGVSubGraph* QGVSubGraph::addSubGraph(const QString&, bool)':
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:69:56: error: 'TRUE' was not declared in this scope
   69 |             ("cluster_" + _name).toLocal8Bit().data(), TRUE);
      |                                                        ^~~~
[ 70%] Built target KeysToJointsPlugin_autogen_timestamp_deps
/build/gazebo-11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:72:71: error: 'TRUE' was not declared in this scope
   72 |         sgraph = agsubg(_sgraph->graph(), _name.toLocal8Bit().data(), TRUE);
  ```

It looks like https://github.com/gazebosim/gazebo-classic/commit/660ae15f4af1f5ea8d9d50d4a24e4d91a94f9c2d is the fix, but I am having trouble getting `self.gazebo_11.overrideAttrs` to patch the library correctly.

My attempt: 
```
  gazebo = self.gazebo_11.overrideAttrs ({
    patches ? [], ...
  }: {
    # Fix build with boost 1.77
    patches = patches ++ [ (self.fetchpatch {
      url = "https://github.com/gazebosim/gazebo-classic/commit/660ae15f4af1f5ea8d9d50d4a24e4d91a94f9c2d.patch";
      sha256 = "sha256-00000000000000000000000000000000000000000000";
      stripLen = 1;
    }) ];
  });
```

drawnwren, 2024-07-07:
Turns out I was patching the wrong thing. Added a pr for the correct patch.

kjeremy, 2024-07-12:
This can be closed now.

closed 2024-07-13 by lopsided98