lopsided98/nix-ros-overlay #430: Turtlebot3 packages on linux fail to build (ROS2 Humble)

hegde-atri, 2024-07-11:
Fails to build cartographer after failing to find `cairo` - [output here](https://pastebin.com/iCxjwJHc).

Not sure how I could fix this as `cairo` seems to exist in `propagatedBuildInputs`.

dylan-gonzalez, 2024-07-28:
I think you have to override the `cartographer` package to add `pkg-config` to its `nativeBuildInputs`. But then there are a bunch of C++ parsing errors...

hegde-atri, 2024-07-31:
> I think you have to override the `cartographer` package to add `pkg-config` to its `nativeBuildInputs`. But then there are a bunch of C++ parsing errors...

How would I go about doing this? Would I have to do this override in `humble/turtlebot3-cartographer/default.nix`?

dylan-gonzalez, 2024-07-31:
on the command line: 
```
nix-build -E 'with import <nix-ros-overlay>; pkgs.rosPackages.humble.cartographer.overrideAttrs ({ nativeBuildInputs ? [ ], ...}: 
    nativeBuildInputs = nativeBuildInputs ++ [ pkgs.pkg-config ];
)'
```

and in a file, you would do a similar thing but in an overlay. But I'd have to take a deeper look to see where to go from here.

liarokapisv, 2024-08-02:
It seems just defining the package in an overlay is not enough, packages most likely keep direct dependencies the previous package. It is probably best to directly `override` cartographer when invoking the package in the `buildEnv`. Other than that it seems that the cartographer package depends on a very old version (2018) of `abseil` which complicates things further.

hegde-atri, 2024-08-16:
> I could temporarily bypass this issue by directly `override` cartographer when passing turtlebot in the `buildEnv`.

@liarokapisv How would I go about doing this in a flake?

wentasah, 2025-04-02:
Can somebody check whether #609 fixes this problem?

PR #609 (2025-04-09): Fix cartographer for ROS 2
With this, cartographer builds at least for Humble, Jazzy and Rolling.

Closes #430
commit 215f4895 (2025-04-09): Fix cartographer for ROS 2

With this, cartographer builds at least for Humble, Jazzy and Rolling.

Closes #430
closed 2025-04-09 by lopsided98