lopsided98/nix-ros-overlay #438: Provide ROS environment on boot

IamTheCarl, 2024-07-29:
I am experimenting to see how I like using NixOS for developing and deploying ROS. I'm familiar with ROS, new to ROS2, and exceptionally new to Nix.
[This repository](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/main/base-config.nix) is currently home to my proof of concept. It's damn ugly. I plan to refine it as I better understand Nix. Feedback is welcome but I may be too new to effectively use it.

I have a few objectives with this proof of concept:
1. Be able to generate a hard drive image for the robot (this is working)
2. Be able to deploy updates to the robot once it's running over SSH using the same configuration that would be used to regenerate the image (this is working)
3. [Install modules with system configuration so I can start them on boot](https://github.com/lopsided98/nix-ros-overlay/issues/383#issuecomment-2067407609) (what I need help with)
    * Ideally I'd like to provide that shell environment with a file like `robot-shell.nix`. I'd be able to run `nix-shell robot-shell.nix` to get a development to replicate the robot's environment on my development machine.

I've successfully created a `shell.nix` file that I can load with `nix-shell` and get a working ros2 environment on my desktop. It's just a copy of [this example](https://github.com/lopsided98/nix-ros-overlay/blob/develop/examples/ros2-basic.nix) with minor modifications. I've been trying to figure out if there's a way to make that the environment my robot provides on boot, and then later be able to add a systemd unit to automatically launch the robot's autonomy.

In the meantime I just tried [dumbly adding the ROS packages to the list of system packages](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/645154891afbfd1fb33d235adb64f49c9f733bd0/base-config.nix#L31-L34). This *somewhat* works. Colcon is available, ros2 is not. I thought that maybe I needed to call a `setup.bash` somewhere but running `find / -name setup.bash` does not reveal the location of any such script.

I appreciate the effort you've put into this repository. I am very impressed and was quite happy with how simple it was to get a usable development environment on my development machine. I understand that my use-case may be niche. Thank you for this.

wentasah, 2024-07-30:
> In the meantime I just tried [dumbly adding the ROS packages to the list of system [...]

I think you need to wrap the ros2 packages (at least the ros2cli) in the ROS `buildEnv` as in the example. This wraps the executable commands and sets the needed environment variables.

IamTheCarl, 2024-07-30:
Thank you for your help.
That makes sense, so much sense that I actually tried it last night in another branch. Unfortunately it didn't work out for me. I'm getting identical results (as far as I can tell). I'm so new to Nix that I'm confident I'm doing something wrong.
Could you give this a look-over?
https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/buildEnv/base-config.nix#L5

IamTheCarl, 2024-07-30:
I feel quite silly. I discovered that I had made a mistake and have corrected it.
I forgot to include the `ros2cli` package in that list.

It's now in there, and I have made progress with that!
The `ros2` command is available but almost all of its subcommands (node, topic, service, etc) are missing.
```
[root@rosetta:~]# ros2
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

options:
  -h, --help            show this help message and exit
  --use-python-default-buffering
                        Do not force line buffering in stdout and instead use the python default buffering, which might be affected by PYTHONUNBUFFERED/-u and depends on whatever stdout is interactive or not

Commands:
  daemon  Various daemon related sub-commands

  Call `ros2 <command> -h` for more detailed usage.
```

I could go installing all of the subcommand packages but that feels like I'm doing something wrong, since that wasn't needed in [the example](https://github.com/lopsided98/nix-ros-overlay/blob/develop/examples/ros2-basic.nix). As a matter of fact, ros2cli wasn't even mentioned in the example, which makes it feel even more strange that I needed to mention ros2cli at all.

wentasah, 2024-07-30:
Yeah, the example is quite old; from times when not many ROS 2 packages worked, so many useful packages are not there. If you add ros-core (which you had in the original code), it should pull in all the needed extensions via ros2cli-common-extensions. If not, then something is broken. Maybe try mentioning ros2cli-common-extensions directly instead of just ros2cli.

Also, are you sure that you use ROS-specific version of buildEnv, i.e. `ros.rosPackages.humble.buildEnv`?

I'm sorry for giving you fuzzy answers. I'm currently on holiday with limited internet connection, so testing your code is a bit problematic for me right know.

IamTheCarl, 2024-07-31:
That was it! I didn't realize that there was a ROS specific version of buildEnv. I swapped that in there and it instantly started behaving just like the nix-shell environment.

Enjoy your vacation, I respect open source maintainers for taking it easy. I know how thankless it can be.

I'm not sure how much you care about what your projects get used for but I'm excited to get going with this. I want to try and integrate ROS with Home Assistant the way ESPHome integrates Arduino with Home Assistant, hopefully opening up a lot of opportunity for DIY smart home robotics. I learned from my last company that starting a robotics project without a good plan for deployment or a convenient way to bring up developer machines is a recipe for disaster, so I was hoping Nix would be a good solution.

So far it's a smidge lacking on documentation, but feature wise it looks like the best solution I've seen yet.

closed 2024-07-31 by IamTheCarl