lopsided98/nix-ros-overlay #439: gz-*-vendor fails on Rolling

khaninger, 2024-07-31:
Problem: The gz-*-vendor packages fail to build, blocking `ur-description`, `rviz2`, and many other basic packages.

Apparent cause: The vendored packages fetch from github if the non-vendored packages are not found by cmake (see, [e.g. here](https://github.com/gazebo-release/gz_math_vendor/blob/a3ba73041b3aca8f04d6cb42decfea35b469d02d/CMakeLists.txt#L42)), which is of course not compatible with the Nix approach.

If the gz-* package is found by cmake, the github pull gets skipped, so I've solved it (as needed for `ur-description`) by overriding as follows:
```
      rospkgs = pkgs.rosPackages.rolling.overrideScope (final: prev: {
        gz-cmake = prev.gz-cmake-vendor.overrideAttrs (old: {
          pname = "gz-cmake3";
          version = "3.5.3";
          src = pkgs.fetchgit {
            url = "https://github.com/gazebosim/gz-cmake";
            rev = "gz-cmake3_3.5.3";
            sha256 = "sha256-fnN3Fmp7F5W0JixJUEp2v/OnXzmRidS5ujmSYxIRWto=";
          };
        });
        gz-cmake-vendor = prev.gz-cmake-vendor.overrideAttrs (old: {
          propagatedBuildInputs = (old.propagatedBuildInputs or []) ++ [final.gz-cmake];
        });
        gz-utils = prev.gz-utils-vendor.overrideAttrs (old: {
          pname = "gz-utils2";
          version = "2.2.0";
          src = pkgs.fetchgit {
            url = "https://github.com/gazebosim/gz-utils";
            rev = "gz-utils2_2.2.0";
            sha256 = "sha256-dNoDOZtk/zseHuOM5mOPHkXKU7wqxxKrFnh7e09bjRA=";
          };
          propagatedBuildInputs = (old.propagatedBuildInputs or []) ++ [final.gz-cmake-vendor]; 
        });
        gz-utils-vendor = prev.gz-utils-vendor.overrideAttrs (old: { 
          propagatedBuildInputs = (old.propagatedBuildInputs or []) ++ [final.gz-utils];
        });
        gz-math = prev.gz-math-vendor.overrideAttrs (old: {
          pname = "gz-math7";
          version = "7.4.0";
          src = pkgs.fetchgit {
            url = "https://github.com/gazebosim/gz-math";
            rev = "gz-math7_7.4.0";
            sha256 = "sha256-W+LAe9hn/LhecxFmmxihv/lNAGvze8w2BDj4rRVfzJo=";
          };
          propagatedBuildInputs = (old.propagatedBuildInputs or []) ++ [
            final.gz-cmake-vendor
            final.gz-utils-vendor
          ]; 
        });
        gz-math-vendor = prev.gz-math-vendor.overrideAttrs (old: {
          propagatedBuildInputs = (old.propagatedBuildInputs or []) ++ [final.gz-math];
        });
      });

```

I'm happy to set up a PR for this but I'm not a Nix expert and am not sure where to best place this in the code base here.

wentasah, 2024-07-31:
This is being handled in #422. However, after fixing the gz-*-vendor packages, there are other problems to solve. And that's why that PR is not yet ready for merge.

I think that if you apply #422 on top of master (or maybe some older commit), it will work. develop branch is broken due to update to newer nixpkgs.

khaninger, 2024-07-31:
Excellent, that worked for me.  Thanks for the fast response and your maintainer efforts here, this repo is the only way I can stay sane with ROS2 build environments

closed 2024-07-31 by khaninger