lopsided98/nix-ros-overlay #444: Provide normal ROS environment on boot

IamTheCarl, 2024-08-03:
Hello, it's me again. This is a continuation of https://github.com/lopsided98/nix-ros-overlay/issues/438 but a different problem.
So I'm now using your `ros.rosPackages.humble.buildEnv` and it does a great job making ROS packages available in shells created with mkShell and installed on my Raspberry Pi with systemPackages, but now I have the opposite issue.

Non-ROS packages installed through `ros.rosPackages.humble.buildEnv` don't seem to properly show up on the target machine where the environment was installed with systemPackages. I can find them in the nix store but they aren't referenced in the PATH or other related environment variables. Interestingly, shells created using `ros.pkgs.mkShell` make a perfect environment with all the packages and environment variables as expected.

For some examples of packages that are not available on the target machine, here I try to install [vim](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/piper/environment.nix#L7).
I also depend on [piper and alsa utils](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/b3c12cbe29b155aadadf60c42e61d99b676b58ff/nodes/piper/default.nix#L14-L15). I also build a package [right out of my repository](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/piper/nodes/piper/default.nix#L21).

All of the ROS packages ([piper](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/piper/nodes/piper/package.xml) and [piper_interface](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/piper/nodes/piper_interface/package.xml)) show up on the Raspberry Pi and are discovered by ROS.

IamTheCarl, 2024-08-03:
I'm trying to dig into this more.
I found out that I can list the dependencies of a package in the nix store by printing the contents of `/nix/store/<package hash and name>/nix-support/propagated-build-inputs`.
So I decided to check out the difference between my desktop's shell session and what's on the Raspberry Pi.

ROS env dependencies on desktop:
```
/nix/store/wvg9979bmqdpcbchrgsqnwz6dx134259-vim-9.1.0148
/nix/store/3jcaxk9g6z0d3s8aj1sq90z6ii3cy7qb-piper-2023.11.14-2
/nix/store/55mczxbwil9xp5mc4zg7sf66vjzvapmk-alsa-utils-1.2.10
/nix/store/2f81xsqb3p049agp1pvf7jdzij8y60hf-python3-3.11.8-env
/nix/store/jq2x9hywgbrcqnfbaqrdc048a01qfhrg-en_US-hfc_male-medium-1.0.0
/nix/store/jacf2kn4dfj99c3ywbvfyg6w53xvxsfl-cmake-3.27.8
/nix/store/jp89dszyb61j89s40rd1v6k8b2kv1njb-gtest-1.12.1-dev
/nix/store/95388cxxjjxl2dfj3rfn9bxrv1wfxvif-console-bridge-1.0.2
/nix/store/4yan2r4vq8s4hv02y8wqcvgp79p8gazp-ament-cmake-core-setup-hook
/nix/store/v710l549nycprz3mplwbfkj3b956i3p6-tinyxml2-10.0.0
/nix/store/zzbxb8sxsi0kinz2iqsaqnh0x8715yx5-libyaml-0.2.5-dev
/nix/store/lxgc60sjs9fvz94gkmq70mkanaig8091-cppcheck-2.13.3
/nix/store/8vf56va5f848yf39cj0ikdpbycm3gzia-libxml2-2.12.3-unstable-2023-12-14-dev
/nix/store/2ibf0354y37ynsq2nvq4zixhcr27hxzm-spdlog-1.12.0-dev
/nix/store/a4q53h7cfszyxk3wwvwzm3hi7wrfvzvw-uncrustify-0.78.1
/nix/store/j7yb0yjmrmjx7a8lyh8yqniywm0l1jwr-openssl-3.0.12-dev
```

ROS env dependencies on Pi:
```
/nix/store/846vnc2hp5kmxvfb1jq00xj27wgfqqhy-vim-9.1.0148
/nix/store/36crnfwic43n8aj973r75d7y5m5nb1da-piper-2023.11.14-2
/nix/store/ns5l3xn4sdljz64ddyis9aqd1s23xmdd-alsa-utils-1.2.10
/nix/store/1z85j93x62n2dg2lhfy2kcj89y60snk4-python3-3.11.8-env
/nix/store/hnilq4y1d0k3ag0rls71pvzdhj33k51n-en_US-hfc_male-medium-1.0.0
/nix/store/m16f4yj9i3fgq5dkx7amcf2v9wp80kc7-cmake-3.27.8
/nix/store/h6w7fc5w16zsiya354b0xw6rg8js1w91-gtest-1.12.1-dev
/nix/store/xsqmjd7b9y00fqhq91zndb9s9vdzrjlh-console-bridge-1.0.2
/nix/store/9dz097vn6s1760r4w2k145hsqk15qylx-ament-cmake-core-setup-hook
/nix/store/0zbqngmfl5cvy4m3wcxg3agbv3hjmihl-tinyxml2-10.0.0
/nix/store/wd2kgn4xdvhfx58in7a81drikcpbdrbi-libyaml-0.2.5-dev
/nix/store/rkm5xdbgw9fi6iq181gccbyrx7nfrw0p-cppcheck-2.13.3
/nix/store/al3fb13zb62d1041cqq5shnkdzzxm05z-libxml2-2.12.3-unstable-2023-12-14-dev
/nix/store/3d4g76i9zhzlvjhyxfdcfra12vh3963z-spdlog-1.12.0-dev
/nix/store/4qnpdr16vbfk6w2f780hrf6kyq5my8nj-uncrustify-0.78.1
/nix/store/gmns25zaak7gmlp3n61b2x0r69vixfc3-openssl-3.0.12-dev
```

Huh, so everything's there. The hashes are different, which mostly make sense since these were compiled for different architectures. The only one that seems odd for having a different hash is `en_US-hfc_male-medium-1.0.0`. That's just a voice model for Piper. There's no difference in the content of the version on the Pi and the version on my desktop. I am wondering if the version of Nix used to build the package influences the hash.

As an extra step I decided to check if all the same setup hooks were being sourced. None of them have to do with the missing packages but between both environments, they are the same.
My tower:
```
$ cat /nix/store/mp5vrm18hzif7zhrz8c2yl1z7mjb5h7h-ros-env/nix-support/setup-hook 
. /nix/store/w4fvvhkzb0ssv0fw5j34pw09f0qw84w8-python3-3.11.7/nix-support/setup-hook
. /nix/store/yjwcmiq2nv1a36h9g02lmj5zpyrdcwks-python3.11-pytest-7.4.3/nix-support/setup-hook
. /nix/store/4w29klx4bfm6nznrvfi12cipy6s1bab5-ros-humble-python-cmake-module-0.10.0-r2-dev/nix-support/setup-hook
. /nix/store/fpy4zdwayygx96jmmb5b105lf8afca8p-ros-humble-rosidl-generator-py-0.14.4-r1/nix-support/setup-hook
. /nix/store/1vq2i3aykn5npiclifj6qxyyvkbg1ari-python3.11-rospkg-1.4.0/nix-support/setup-hook
```

The Pi:
```
$ cat /nix/store/4pgmxq0cyrqcxlpvjzpc00r7s95l657q-ros-env/nix-support/setup-hook 
. /nix/store/dz8lm4h0ivibad5kfc0ya3p3zqyd2fyf-python3-3.11.7/nix-support/setup-hook
. /nix/store/2czpd8x98dzk7ww2nbm24rk9mjn128d8-python3.11-pytest-7.4.3/nix-support/setup-hook
. /nix/store/l4f03nlbl7da750bxsspv9v3xvngnygq-ros-humble-python-cmake-module-0.10.0-r2-dev/nix-support/setup-hook
. /nix/store/16iynv7ivp93phjp1dyh457kxamac5m8-ros-humble-rosidl-generator-py-0.14.4-r1/nix-support/setup-hook
. /nix/store/4digdwj0w7w4bwk8lrckmpvv4x79qs8q-python3.11-rospkg-1.4.0/nix-support/setup-hook
```

Actally that's something interesting. Python3 has successfully made it onto the Pi.
It was probably brought in by a package that's not mine, but I should look into why that was successful.

IamTheCarl, 2024-08-11:
Well I gave it some more work this weekend. Made some discoveries.
The piper node itself can discover the piper-tts program. It cannot discover the [voice model](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/tree/piper/nodes/piper/voice_models/en_US-hfc_male-medium) though because `XDG_DATA_DIRS` isn't being defined for it.

hacker1024, 2024-08-11:
The ROS `buildEnv` isn't really great for non-shell usage. I have made https://github.com/hacker1024/nix-ros-workspace for this use case.

IamTheCarl, 2024-08-12:
Reading through your documentation, it sounds like exactly what I need.
I am new to Nix and unfamiliar with how things should be done. Is this meant to be used specifically with Flakes? I'm not using them because the documentation made it sound like they were far from stabilized, but I really like the concept behind them so it wouldn't be hard to convince me to try.

Currently my attempt looks like this and I'm not able to use my piper node in a nix-shell or on my remote pi. I suspect I've set something up wrong. I experimented with moving my piper node from `devPackages` to `prebuiltPackages` but that didn't make a difference in either environment.

Still, thank you for your help. I would love to get this working.
https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/blob/wip-trying_nix_ros_workspace/environment.nix

IamTheCarl, 2024-08-15:
I finally figured it out.
I had to convert the package with my voice model in it to be a [ROS package](https://github.com/IamTheCarl/ROS-on-Raspberry-Pi-with-NixOS-POC/tree/main/nodes/piper/voice_models/en_US-hfc_male-medium). I suspect something about buildRosPackage is not properly sourcing my packages setup-hook, preventing this from working on the Pi as expected.

closed 2024-08-15 by IamTheCarl