lopsided98/nix-ros-overlay #465: How do I install rosdep too

conner-replogle, 2024-09-05:
I tried adding it as a ros package but to no avail.
I keep getting 
bash: rosdep: command not found

wentasah, 2024-09-05:
Why  do you need to install `rosdep`? With Nix, it's almost not needed, because the packages in this overlay are already generated with the help of `rosdep` and depend on correct nixpkgs dependencies. Running `rosdep install` is therefore not needed on NixOS and would not be useful on non-NixOS distros.

If you really need it, you can try `nix-env -f . -i -A python3Packages.rosdep`.

conner-replogle, 2024-09-05:
I'm using this to build realsense-ros are you saying there's a way to just make the flake hold all the dependencies it needs instead of rosdep?

wentasah, 2024-09-20:
Sorry for delayed answer. Yes. It's possible. I was working on a tool that allows doing just that and uses `rosdep` under the hood. It's called [ros2nix](https://github.com/wentasah/ros2nix/) and now it's probably in state usable by others.

I tried it on realsense-ros:
```
git clone https://github.com/IntelRealSense/realsense-ros
ros2nix --output-as-nix-pkg-name --fetch --flake $(find -name package.xml)
nix build .#humble.realsense2-camera
```
And it complains about missing RealSense SDK. If you have a Nix expression for it, you can run:
```
ros2nix --output-as-nix-pkg-name --fetch --flake --extra-build-inputs realsense-sdk $(find -name package.xml_
```
and add your expression to the generated `overlay.nix`. It might then work.