lopsided98/nix-ros-overlay #469: Error in basic ros2-foxy flake

BalderHolst, 2024-09-13:
Hello i am trying to use ros2 foxy. I made a simple flake with the example from the readme. Changing the `pkgs.rosPackages.humble` to `pkgs.rosPackages.foxy` results in a strange error. Here is the flake.nix contents:

``ǹix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = [
            pkgs.colcon
            # ... other non-ROS packages
            (with pkgs.rosPackages.foxy; buildEnv {
                paths = [
                    ros-core
                    # ... other ROS packages
                ];
            })
          ];
        };
      });
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

Running `nix develop` results in the following error:

```
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:9:12:
            8|
            9|   strict = derivationStrict drvAttrs;
             |            ^
           10|

       … while evaluating derivation 'Example-project'
         whose name attribute is located at /nix/store/snp5f95xhkz89kdw9703kana3m6xfd0a-source/pkgs/stdenv/generic/make-derivation.nix:352:7

       … while evaluating attribute 'nativeBuildInputs' of derivation 'Example-project'
         at /nix/store/snp5f95xhkz89kdw9703kana3m6xfd0a-source/pkgs/stdenv/generic/make-derivation.nix:396:7:
          395|       depsBuildBuild              = elemAt (elemAt dependencies 0) 0;
          396|       nativeBuildInputs           = elemAt (elemAt dependencies 0) 1;
             |       ^
          397|       depsBuildTarget             = elemAt (elemAt dependencies 0) 2;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: evaluation aborted with the following error message: 'lib.customisation.callPackageWith: Function called without required argument "ros2doctor" at /nix/store/hbi3lvgz26n8pv4gqc0kbj8xh2yvzhfq-source/distros/foxy/ros-core/default.nix:5'
```

I hope that someone has a fix for this problem, this is such an awesome repo!

wentasah, 2024-09-13:
Hi, the problem is that in foxy, ros-core depends on ros2doctor, which is not available via nix, because it has dependencies unavailable in nixpkgs. 

One solution is not to include ros-core in your flake, but just its dependencies except ros2doctor.

Another solution is to add a dummy ros2doctor package to foxy. I had a code for that, but I'll have to find it if want to see it.

lopsided98, 2024-09-13:
Foxy has been EOL for over a year. I didn't delete it as a courtesy to @hacker1024, but I'm not going to provide any fixes for it.

BalderHolst, 2024-09-18:
Alright that is totally fair. Thank you for your all your work:)