lopsided98/nix-ros-overlay #501: Building Rust nodes

IamTheCarl, 2024-10-08:
I'm interested in building a ROS node written in Rust and am exploring this idea. I have not been able to find evidence of anyone else trying this.

Here is my current attempt: https://github.com/IamTheCarl/ros_assistant/blob/rosetta_robot/packages/create_bridge/default.nix

So I was thinking that in theory I could just add ros-core as a dependency and that would make the `ROS_DISTRO` and other necessary environment variables available, but I get the following error when running `nix-build`:
```
error: failed to run custom build command for `r2r_rcl v0.9.0`

Caused by:
  process didn't exit successfully: `/build/create_bridge/target/release/build/r2r_rcl-f66cac5910a687c4/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO

  --- stderr
  thread 'main' panicked at /build/create_bridge-0.1.0-vendor.tar.gz/r2r_common/src/lib.rs:144:51:
  ROS_DISTRO not set: Source your ROS!
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

So clearly the environment variable isn't passing through to it.

I should note, I can build the node in a Nix Shell: https://github.com/IamTheCarl/ros_assistant/blob/rosetta_robot/packages/create_bridge/shell.nix

I'm wondering if I need to modify [buildRosPackage](https://github.com/lopsided98/nix-ros-overlay/blob/develop/distros/build-ros-package/default.nix) to support Rust. I made an attempt at that earlier today but without success. I clearly don't understand how that function works and would love an explanation of it.

wentasah, 2024-10-08:
I'm using Rust as well as r2r. I think that `buildRosPackage` should work, but you must add `ament-cmake` or at least `ament-cmake-core` to `nativeBuildInputs`. This makes `ROS_*` variables available if you also depend on `ros-environment`, which is a part of `ros-core`.

I ended up compiling r2r apps with [crate2nix](https://github.com/nix-community/crate2nix). The reason is that r2r compiles messages as part of the main r2r crate and it is quite slow. And the messages are recompiled whenever e.g. AMENT_PREFIX_PATH changes, which happen quite often with Nix. The advantage of `crate2nix` is that it compiles each crate as a separate derivation and thus recompiles the r2r messages only when the messages changes, not when anything else changes. When only the r2r app (`main.rs`) changes, r2r crate is reused and recompilation is fast. I use the following expression to compile my apps with the help of `crate2nix`:
```nix
{ flexdev-interfaces-2
, buildRustCrate
, rustPlatform
, callPackage
, ros-core
, cf-jcc-interfaces
, eigen
, debug-interfaces
, nn-lines-interfaces
, tf2-msgs
}:

let
  messages = [ flexdev-interfaces-2 cf-jcc-interfaces tf2-msgs debug-interfaces nn-lines-interfaces ];

  addDeps = list: { ... }: { nativeBuildInputs = list ++ [ rustPlatform.bindgenHook ]; };
  customBuildRustCrateForPkgs = pkgs: buildRustCrate.override {
    defaultCrateOverrides = pkgs.defaultCrateOverrides // {
      r2r_rcl = addDeps [ ros-core ];
      r2r_msg_gen = addDeps ([ ros-core ] ++ messages);
      r2r_actions = addDeps [ ros-core ];
      r2r = addDeps ([ ros-core ] ++ messages);
      gradient_planner = addDeps [ eigen ];
    };
  };
  generatedBuild = callPackage ./Cargo.nix {
    buildRustCrateForPkgs = customBuildRustCrateForPkgs;
  };
in
generatedBuild.rootCrate.build.overrideAttrs ({postInstall, ...}: {
  postInstall = ''
    install -v -D -t $out/launch launch/*
  '';
})
```

It's not perfect, but it's sufficient for what I currently need.

IamTheCarl, 2024-10-08:
Thank you for your help! This was the exact hint I needed to figure this out.
I had started to experiment with crate2nix but didn't figure out how to add dependencies to the individual crates.

This is a really nice setup, since I was worried re-compiling the whole project with each deployment could get tedious.

IamTheCarl, 2024-10-09:
If I could ask you one more question:
Currently I can install my node onto my robot and manually launch it by calling the executable from the command terminal.

The `ros2` command is unaware of my package (probably because it wasn't built with ament cmake and therefore missing necessary files). This makes it so that tools like ros launch or this repository's NixOS integration cannot launch my node.

Do you have a workaround for that or has that not been done before?

IamTheCarl, 2024-10-09:
I've come up with a working solution. It's very manual at the moment so I'd like to automate it more.
My current solution is to build the Rust binary as usual, but instead of installing that derivation directly on the robot, have a ROS package built in another derivation that symbolically links to the one with the ROS binary.

https://github.com/IamTheCarl/ros_assistant/blob/3380a4762340eb19a039b339cc73a78204c4a506/packages/create_bridge/default.nix#L32

I still consider myself new to Nix so I'm not sure if that's a bad practice.

IamTheCarl, 2024-10-11:
I have [created a tool](https://github.com/IamTheCarl/ros_assistant/blob/824d430efe96907d45276c1815f6b30da7382f73/packages/rosify_package/default.nix#L1) for converting any ROS node written in any language (not just Rust) into a ROS package.

You can see an example of how it's intended to be used [here](https://github.com/IamTheCarl/ros_assistant/blob/824d430efe96907d45276c1815f6b30da7382f73/packages/create_bridge/default.nix#L35).

I would be happy to contribute this tool to this repository if you think it would fit in. I would need some guidance of where it belongs in the file structure.

closed 2024-10-11 by IamTheCarl

wentasah, 2024-10-19:
Sorry for not responding earlier. I use the a very similar approach. I call the following function for my r2r packages:
```nix
{ rustPkg, lib, ament-cmake-ros, stdenv }:
let
  pkgName = lib.strings.removePrefix "rust_" (builtins.head (builtins.split "-" rustPkg.name));
in
stdenv.mkDerivation {
  name = "ros-${rustPkg.name}";
  unpackPhase = ''
    cat > CMakeLists.txt <<'EOF'
    cmake_minimum_required(VERSION 3.5)
    project(${pkgName})
    find_package(ament_cmake REQUIRED)
    install(PROGRAMS ${rustPkg}/bin/${pkgName} DESTINATION lib/${pkgName})
    install(DIRECTORY ${rustPkg}/launch DESTINATION share/${pkgName})
    ament_package()
    EOF

    cp ${rustPkg.src}/package.xml .
  '';
  nativeBuildInputs = [ ament-cmake-ros ];
}
```
However, `colcon` complains about those packages that `AMENT_PREFIX_PATH doesn't contain any 'local_setup.*' files`. I don't know (yet) why, because the files are there. Do you experience this too?

I also tried using `cargo-ament-build` to create ROS layout from Rust packages as follows:
```nix
# Run cargo-ament-build to create directory structure of ROS
# package from Rust crate compiled by crate2nix. We already have
# the crate built, so we prevent cargo-ament-build from actually
# running "cargo build" and instead instruct it (in a hackish way)
# to operate on the Nix-compiled binaries from rustPkg.
cargoAmentBuild = pkgs: rustPkg: pkgs.runCommand "ros-${rustPkg.name}" {} ''
  PATH=${pkgs.writeShellScriptBin "cargo" "true"}/bin:$PATH
  ${pkgs.cargo-ament-build}/bin/cargo-ament-build \
      --manifest-path ${rustPkg.src}/Cargo.toml \
      --install-base=$out \
      --target-dir=${rustPkg} \
      --profile bin
'';
```
It was possible to do that, however `cargo-ament-build` doesn't install the local_setup files at all.

I think the proper solution to many of these problems would be to use `colcon` to build the packages (see https://github.com/lopsided98/nix-ros-overlay/issues/362#issuecomment-2067777696).

IamTheCarl, 2024-10-21:
> However, colcon complains about those packages that AMENT_PREFIX_PATH doesn't contain any 'local_setup.*' files. I don't know (yet) why, because the files are there. Do you experience this too?

No, I do not get this warning, but I think that's because I'm not using colcon to build.

My [custom tool](https://github.com/IamTheCarl/ros_assistant/blob/824d430efe96907d45276c1815f6b30da7382f73/packages/rosify_package/default.nix#L1) calls `buildRosPackage`. Looking at [the implementation](https://github.com/lopsided98/nix-ros-overlay/blob/develop/distros/build-ros-package/default.nix) for that, I don't see it invoking colcon. It looks like it just invokes cmake itself. That would explain why I don't see the warning.