lopsided98/nix-ros-overlay #509: freeimage-unstable insecure

luc-spec, 2024-10-19:
Preface -- I'm brand new to Nix with ROS. Maybe this is widely known and I'm just catching on. 

My system:

- nixos-unstable 
- flakes enabled [x]

I tried running `nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo`

```bash
> nix develop --impure github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo
do you want to allow configuration setting 'extra-substituters' to be set to 'https://ros.cachix.org' (y/N)?
do you want to permanently mark this value as untrusted (y/N)?
warning: ignoring untrusted flake configuration setting 'extra-substituters'.
Pass '--accept-flake-config' to trust it
do you want to allow configuration setting 'extra-trusted-public-keys' to be set to 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' (y/N)?
do you want to permanently mark this value as untrusted (y/N)?
warning: ignoring untrusted flake configuration setting 'extra-trusted-public-keys'.
Pass '--accept-flake-config' to trust it
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:34:12:
           33|
           34|   strict = derivationStrict drvAttrs;
             |            ^
           35|

       … while evaluating derivation 'nix-shell'
         whose name attribute is located at /nix/store/l3amk5lsakpc93him5kry24kax23sn4h-source/pkgs/stdenv/generic/make-derivation.nix:336:7

       … while evaluating attribute 'buildInputs' of derivation 'nix-shell'
         at /nix/store/l3amk5lsakpc93him5kry24kax23sn4h-source/pkgs/stdenv/generic/make-derivation.nix:383:7:
          382|       depsHostHost                = elemAt (elemAt dependencies 1) 0;
          383|       buildInputs                 = elemAt (elemAt dependencies 1) 1;
             |       ^
          384|       depsTargetTarget            = elemAt (elemAt dependencies 2) 0;

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: Package ‘freeimage-unstable-2021-11-01’ in /nix/store/l3amk5lsakpc93him5kry24kax23sn4h-source/pkgs/development/libraries/freeimage/default.nix:72 is marked as insecure, refusing to evaluate.

       Known issues:
        - CVE-2021-33367
        - CVE-2021-40262
        - CVE-2021-40263
        - CVE-2021-40264
        - CVE-2021-40265
        - CVE-2021-40266
        - CVE-2023-47992
        - CVE-2023-47993
        - CVE-2023-47994
        - CVE-2023-47995
        - CVE-2023-47996

       You can install it anyway by allowing this package, using the
       following methods:

       a) To temporarily allow all insecure packages, you can use an environment
          variable for a single invocation of the nix tools:

            $ export NIXPKGS_ALLOW_INSECURE=1

          Note: When using `nix shell`, `nix build`, `nix develop`, etc with a flake,
                then pass `--impure` in order to allow use of environment variables.

       b) for `nixos-rebuild` you can add ‘freeimage-unstable-2021-11-01’ to
          `nixpkgs.config.permittedInsecurePackages` in the configuration.nix,
          like so:

            {
              nixpkgs.config.permittedInsecurePackages = [
                "freeimage-unstable-2021-11-01"
              ];
            }

       c) For `nix-env`, `nix-build`, `nix-shell` or any other Nix command you can add
          ‘freeimage-unstable-2021-11-01’ to `permittedInsecurePackages` in
          ~/.config/nixpkgs/config.nix, like so:

            {
              permittedInsecurePackages = [
                "freeimage-unstable-2021-11-01"
              ];
            }
```

My workaround is intentionally moving past the insecure package with `NIXPKGS_ALLOW_INSECURE=1 nix develop --impure github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo`.

Is there a better workaround (i.e. removing or updating `freeimage-unstable-2021-11-01`, the package with the CVEs)?

lopsided98, 2024-10-19:
Gazebo depends on freeimage, and freeimage is unmaintained and has unpatched CVEs. There's not much that I can do about it unless upstream gets rid of the freeimage dependency.

The other issue is due to the upgrade to Python 3.12, which removes distutils from the standard library and is fixed by #506.

lopsided98, 2024-10-19:
It turns out there are a lot more packages that depend on distutils, so I'll have to figure out some way to dealing with them in bulk.

wentasah, 2025-03-27:
FYI: Some of the vulnerabilities were patched recently: https://github.com/NixOS/nixpkgs/pull/369766. Unfortunately, many still remain.

wentasah, 2026-08-29:
Closing this. We got rid of freeimage some months ago.

closed 2026-08-29 by wentasah