lopsided98/nix-ros-overlay #511: velodyne-pointcloud 2.4.0-r1 doesn't build

quantum9Innovation, 2024-10-20:
See: https://github.com/lopsided98/nix-ros-overlay/blob/master/distros/humble/velodyne-pointcloud/default.nix

## Steps to reproduce

1. Load the following into `flake.nix`:

```nix
{
  inputs = {
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
  };
  outputs =
    {
      self,
      nix-ros-overlay,
      nixpkgs,
    }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in
      {
        devShells.default = pkgs.mkShell {
          name = "Test";
          packages = [
            (
              with pkgs.rosPackages.humble;
              buildEnv {
                paths = [
   	              ros-core
	              velodyne-pointcloud
                  # add other ROS packages here
                ];
              }
            )
          ];
        };
      }
    );
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

2. `nix develop` gives:

```
error: builder for '/nix/store/qv2kmym5qwp6mfs2075sjr7ca1ypqbrj-ros-humble-velodyne-pointcloud-2.4.0-r1.drv' failed with exit code 1;
       last 25 log lines:
       > -- Found rmw_fastrtps_cpp: 6.2.6 (/nix/store/ma7myf6hai606m5iijwwivid92f4jaxi-ros-humble-rmw-fastrtps-cpp-6.2.6-r1/share/rmw_fastrtps_cpp/cmake)
       > -- Found OpenSSL: /nix/store/1l3a02nqq5b5v7rhchj89hi7plmbza5r-openssl-3.0.12/lib/libcrypto.so (found version "3.0.12")
       > -- Found FastRTPS: /nix/store/vp25xly0qsyvkimi1hmcjhi9fb4rnfk7-ros-humble-fastrtps-2.6.7-r1/include
       > -- Using RMW implementation 'rmw_fastrtps_cpp' as default
       > -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
       > -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
       > -- Found Threads: TRUE
       > CMake Error at CMakeLists.txt:16 (find_package):
       >   By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
       >   asked CMake to find a package configuration file provided by "Eigen3", but
       >   CMake did not find one.
       >
       >   Could not find a package configuration file provided by "Eigen3" with any
       >   of the following names:
       >
       >     Eigen3Config.cmake
       >     eigen3-config.cmake
       >
       >   Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
       >   "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
       >   provides a separate development package or SDK, be sure it has been
       >   installed.
       >
       > 
       > -- Configuring incomplete, errors occurred!
       For full logs, run 'nix log /nix/store/qv2kmym5qwp6mfs2075sjr7ca1ypqbrj-ros-humble-velodyne-pointcloud-2.4.0-r1.drv'.
error: 1 dependencies of derivation '/nix/store/y6qcjx17wjnqiay5qhnbcib2l166idpz-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/hy0c5d2n3x1g0nzqay427qvila4vh2da-Test-env.drv' failed to build
```

3. `nix log /nix/store/qv2kmym5qwp6mfs2075sjr7ca1ypqbrj-ros-humble-velodyne-pointcloud-2.4.0-r1.drv` gives:

```
warning: The interpretation of store paths arguments ending in `.drv` recently changed. If this command is now failing try again with '/nix/store/qv2kmym5qwp6mfs2075sjr7ca1ypqbrj-ros-humble-velodyne-pointcloud-2.4.0-r1.drv^*'
@nix { "action": "setPhase", "phase": "unpackPhase" }
Running phase: unpackPhase
unpacking source archive /nix/store/csy5aw7vk55lnczxn88y9awkyld43mp1-2.4.0-1.tar.gz
source root is velodyne-release-release-humble-velodyne_pointcloud-2.4.0-1
setting SOURCE_DATE_EPOCH to timestamp 1685193474 of file velodyne-release-release>
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "configurePhase" }
Running phase: configurePhase
fixing cmake files...
cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE>
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/khkhbch4p1wjfl1g89gw1mszvvr7bzv0-gcc-w>
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/khkhbch4p1wjfl1g89gw1mszvvr7bzv0-gcc>
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 1.3.8 (/nix/store/x52wsjvyrsppy332p3gj923d4y2dl8fh-ros-humbl>
-- Found Python3: /nix/store/w4fvvhkzb0ssv0fw5j34pw09f0qw84w8-python3-3.11.7/bin/p>
-- Found angles: 1.15.0 (/nix/store/rijx94rg0hpncz380mdqigia07mr55xi-ros-humble-an>
-- Found diagnostic_updater: 3.1.2 (/nix/store/1mirhmk01jj4bcnr5jrdavifhp3f0qq5-ro>
-- Found rosidl_generator_c: 3.1.5 (/nix/store/21q0vd4fdkixsbsykqmf9bm0kdb53pn7-ro>
-- Found rosidl_adapter: 3.1.5 (/nix/store/jndkqs8gv0zivjq66vmy2snq279v33z7-ros-hu>
-- Found rosidl_generator_cpp: 3.1.5 (/nix/store/gcwmrn7n1jdcffr38vdna47n58pgw5rz->
-- Using all available rosidl_typesupport_c: rosidl_typesupport_introspection_c;ro>
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_introspection_cp>
-- Found rmw_implementation_cmake: 6.1.1 (/nix/store/hhm2r5gfqkdgh9nzr5rshlrr9ypsw>
-- Found rmw_fastrtps_cpp: 6.2.6 (/nix/store/ma7myf6hai606m5iijwwivid92f4jaxi-ros->
-- Found OpenSSL: /nix/store/1l3a02nqq5b5v7rhchj89hi7plmbza5r-openssl-3.0.12/lib/l>
-- Found FastRTPS: /nix/store/vp25xly0qsyvkimi1hmcjhi9fb4rnfk7-ros-humble-fastrtps>
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

wentasah, 2024-10-20:
That's because that package forgot to mention eigen (and pkg-config) in their `package.xml` for ROS 2.

You can build the package by applying the following change to nix-ros-overlay:
```diff
diff --git a/distros/distro-overlay.nix b/distros/distro-overlay.nix
index 34bc6c297..af1bd5ca4 100644
--- a/distros/distro-overlay.nix
+++ b/distros/distro-overlay.nix
@@ -387,6 +387,15 @@ let
     };
 
     urdf = patchBoostPython rosSuper.urdf;
+
+    velodyne-pointcloud = rosSuper.velodyne-pointcloud.overrideAttrs ({
+      buildInputs ? [],
+      nativeBuildInputs ? [], ...
+    }: if distro == "noetic" then { } else {
+      /* For all ROS 2 distros: */
+      buildInputs = buildInputs ++ [ self.eigen ];
+      nativeBuildInputs = nativeBuildInputs ++ [ self.pkg-config ];
+    });
   };
 
   otherSplices = {
```
However, it would be better to fix that in their `package.xml`. Do you think, you could send them a PR?

quantum9Innovation, 2024-10-22:
Thank you for the fix! I will look into submitting a PR soon if everything works well.

quantum9Innovation, 2024-10-27:
It seems `eigen` was already added as a dependency many years ago: https://github.com/ros-drivers/velodyne/commit/b590d0060cd187d20f9b45aa20d833773382b081

It is specified here: https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/package.xml#L33

Not sure if it also needs to be specified somewhere else in `package.xml` or if I need to include/where I should include `pkg-config` in the `package.xml` file.

quantum9Innovation, 2024-10-27:
> ```diff
> velodyne-pointcloud = rosSuper.velodyne-pointcloud.overrideAttrs ({
> +      buildInputs ? [],
> +      nativeBuildInputs ? [], ...
> +    }: if distro == "noetic" then { } else {
> +      /* For all ROS 2 distros: */
> +      buildInputs = buildInputs ++ [ self.eigen ];
> +      nativeBuildInputs = nativeBuildInputs ++ [ self.pkg-config ];
> +    });
> ```

Changing this does solve the issue, but I'm not sure why the issue exists given that `eigen` is part of the `package.xml` dependencies specified by `velodyne-pointcloud`. Maybe it's only an issue with `pkg-config`? In that case, I'm not sure how to add this as a dependency in the `package.xml` file.

wentasah, 2024-10-28:
eigen was added only in the `master` branch, which is used only for ROS 1 releases (noetic). ROS 2 releases are on the `ros2` branch, and there [eigen is missing](https://github.com/ros-drivers/velodyne/blob/ros2/velodyne_pointcloud/package.xml).

quantum9Innovation, 2024-10-29:
Thanks for looking into this; I've submitted a PR here: https://github.com/ros-drivers/velodyne/pull/544. Hopefully it is merged soon (in the meantime I'm using this fork as my ROS overlay: https://github.com/ArcturusNavigation/nix-ros-overlay).

closed 2024-10-29 by quantum9Innovation

quantum9Innovation, 2024-10-31:
Should be fixed upstream now (closed in https://github.com/ros-drivers/velodyne/pull/545)