lopsided98/nix-ros-overlay #521: How does one create development shells for packages under development?

liarokapisv, 2024-10-30:
In normal situations, I can put specify a package under development within `mkShell`'s `inputsFrom` list and it will automatically fetch its dependencies and make them available without building the package itself. Because nix-ros-overlay provides the `buildEnv` utility, I don't see a good way to achieve the same. I guess one could directly flatten a given package's `(propagated)(native)buildInputs` and put those inside `buildEnv`. What is the recommended approach here?

kjeremy, 2024-10-30:
I'm not quite sure what you are asking but if I need to hack on a package that is in this repo (in this case `roscore`) I use something like `nix develop .noetic.roscore` and then run the `unpackPhase` and `patchPhase` before hacking on it.

wentasah, 2024-11-01:
I think that flattening the dependencies of needed packages is a way to go. Doing that would be useful for me too, but I don't have any code for it. So if you come with some solution, I'd appreciate if you can share it.

liarokapisv, 2024-11-03:
@kjeremy My usecase is maintaining a custom workspace with multiple packages under development. Ideally I would just list the derivation expressions for these in my shell's `inputFrom` in order to get a sane dev environment but this doesn't work due to the buildEnv. I don't think using `nix develop` is enough in this case otherwise `buildEnv` would not be needed.

RandomSpaceship, 2024-12-12:
I know I'm late to the party, but would the `buildROSWorkspace` function (and resultant `env` attribute for the shell) [nix-ros-workspace](https://github.com/hacker1024/nix-ros-workspace) work for you?

If you specifically need flattened dependencies, I'm currently developing on a flake-ified [fork](https://github.com/RandomSpaceship/nix-ros-workspace) which does [exactly that](https://github.com/RandomSpaceship/nix-ros-workspace/commit/a10de8651b000a54de61b24577445f9e06903cf2) for ROS dependencies in the `nix develop` shell, since I've had my own issues with it. Feel free to use that code if it works for you, though I wouldn't recommend using my fork directly for the next couple of weeks since I might need to make breaking changes.

liarokapisv, 2024-12-12:
Yes that one works, it's actually what I am currently using, just wanted to check that there is no any upstream alternative. I have no issues with `nix-ros-workspace` but generally it would be better to get first-class nix package support so that we don't need any `buildEnv` wrappers and just adding packages to `inputsFrom` directly would work directly.

lopsided98, 2024-12-12:
You don't have to use `buildEnv`. It primarily exists because the environment variables can become too long with lots of ROS packages.

closed 2025-09-14 by liarokapisv