lopsided98/nix-ros-overlay #525: nose has been removed since it has been deprecated

ckugel, 2024-11-15:
Hello I am trying to use nixos flakes to install ROS1 noetic for a universal robotics driver. I am a pretty new to this and have been getting an issue when I am trying to install noetic.ros-core that 

"nose has been removed since it has been deprecated and unmaintained and does not work on Python 3.12; palse switch to pytest or another test / framework."

It is quite possible that I am just dumb and doing something wrong. Here is my flake.nix that I am running sudo nix develop on:

```nix
{
  description = "Flake for Universal Robots Driver on ROS Noetic";

  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs";
    ros-overlay.url = "github:lopsided98/nix-ros-overlay";
    flake-utils.url = "github:numtide/flake-utils";
  };

  outputs = { self, nixpkgs, ros-overlay, flake-utils }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs { 
          inherit system; 
          overlays = [ ros-overlay.overlays.default ]; 
        };
      in {
        devShell = pkgs.mkShell {
          packages = with pkgs; [
            colcon
            git
            (with rosPackages.noetic; buildEnv {
              paths = [
                ros-core
                ];
            })
            gcc
            cmake
            pkg-config
            boost
            eigen
          ];

          # Environment setup
          shellHook = ''
            echo "Setting up Universal Robots ROS Noetic development environment"
            export ROS_VERSION="noetic"
            
            # Create catkin workspace if it doesn't exist
            if [ ! -d "catkin_ws" ]; then
              mkdir -p catkin_ws/src
              cd catkin_ws/src
              git clone -b melodic-devel https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git
              git clone -b noetic-devel https://github.com/ros-industrial/universal_robot.git
              cd ..
              colcon build
              cd ..
            fi

            # Source the workspace
            source catkin_ws/install/setup.bash
          '';
        };
      }
    );

  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}

```

lopsided98, 2024-11-15:
You're using a newer nixpkgs version than I have tested. I recommend pinning the nixpkgs version as shown in the example in the README:
```nix
nixpkgs.follows = "nix-ros-overlay/nixpkgs";
```

On the other hand, you are likely to still have problems because the removal of distutils from Python 3.12 broke a lot of noetic packages, and I haven't fixed them all yet.

ckugel, 2024-11-15:
Ok, perfect, thank you for the following suggestion. I got some packages to work but not all. Is there something I could do to get some working? Again, sorry if I sound very ignorant, I am very new at this.

lopsided98, 2024-11-15:
Try updating the overlay; I just fixed a bunch of packages. If it is still broken, provide more information.

taylorpool, 2024-12-11:
Getting the same issue. Simplified to the following `shell.nix`

```
{
  pkgs ? import <nixpkgs> { },
}:
let
  nixpkgs =
    import
      (pkgs.fetchzip {
        url = "https://github.com/NixOS/nixpkgs/archive/refs/tags/branch-off-24.11.tar.gz";
        hash = "sha256-CqCX4JG7UiHvkrBTpYC3wcEurvbtTADLbo3Ns2CEoL8=";
      })
      {
        overlays = [
          (import (
            (pkgs.fetchFromGitHub {
              owner = "lopsided98";
              repo = "nix-ros-overlay";
              rev = "e66f390525a62b16decb6a953508a9b3422594a2";
              hash = "sha256-ArHMycnUriIGZYPL4J4NCqkdYP/PJCGMpTlw7RM5SZg=";
            })
            + "/overlay.nix"
          ))
        ];
      };
in
with nixpkgs;
mkShell {
  packages = [
    rosPackages.noetic.catkin
  ];
}
```

wentasah, 2024-12-11:
Should be fixed by #539.

closed 2024-12-12 by lopsided98