lopsided98/nix-ros-overlay #536: "ROS" packages already in nixpkgs

nim65s, 2024-12-08:
Hi,

Could we override some packages from the overlay to directly use those provided by nixpkgs, instead of trying to re-declare and re-build things from ROS ?

Context: I'm the maintainer (upstream + nixpkgs + ROS) of a few things (eg. eigenpy / hpp-fcl / pinocchio): pinocchio is in better shape in nixpkgs than in ROS. For example in nixpkgs we can enable the casadi optional dependency, while in ROS that's harder since casadi is not packaged in ubuntu.

Also, since pinocchio requires lot of RAM and time to build, I think it would be very beneficial for your your build & cache infra :)

NB: I'm speaking of non-ROS-only packages, which are packaged in rosdistro. But maybe it make little or no difference for pure ROS packages.

nim65s, 2024-12-08:
Hum, wait, that's already the case, isn't it ?
You're wonderful, thank you.

nim65s, 2024-12-08:
Actually, that's half the case, this is why I was mislead in the first place:
```nix
nix repl
Lix 2.91.1
Type :? for help.
nix-repl> :lf github:lopsided98/nix-ros-overlay
Added 17 variables.

nix-repl> outputs.legacyPackages.x86_64-linux.humble.pinocchio                 
«derivation /nix/store/ka5slb3x93ky1k8s1i7710qdpbwlq1fi-ros-humble-pinocchio-2.6.21-r1.drv»

nix-repl> outputs.legacyPackages.x86_64-linux.humble.python3Packages.pinocchio
«derivation /nix/store/gac53bw2ryvinpyf40rfwc3b5mpkskgw-pinocchio-3.2.0.drv»

hacker1024, 2025-01-21:
This occurs because ROS packages are in one scope, while Python packages are in another. Using a ROS `callPackage` will shadow Nixpkgs packages from the scope above, but not packages in the Python scope.

i.e.

```nix
rosPackages.rolling.callPackage ({
  buildRosPackage
, pinocchio # Shadowed by the ROS definition
, python3Packages # Not defined by this overlay, so straight from Nixpkgs
}: buildRosPackage { /* ... */ }) { };
```

It would probably be a good idea to check for existing packages either in Superflore or in a step after running it.

---

rtabmap is another shadowed package that would benefit from this.

PR #538 (2025-03-01): inherit eigenpy, hppfcl, and pinocchio from nixpkgs
Hi again,

Here is a very simple attempt to fix #536
commit fd8d6a85 (2025-03-01): inherit eigenpy, hppfcl, and pinocchio from nixpkgs

fix #536
closed 2025-03-01 by lopsided98