lopsided98/nix-ros-overlay #540: Completely broken builds after python -> python3 rename

RandomSpaceship, 2024-12-12:
It looks like the python name change has completely broken the build. After adding the python2 package to the `permittedInsecurePackages` list, the new error was:
```
error: attribute 'python' missing
       at /nix/store/dj5fd5ifqjrq863s3iqx4rmqxzgy1fhv-source/distros/ros2-overlay.nix:125:14:
          124|   python-cmake-module = rosSuper.python-cmake-module.overrideAttrs ({ ... }: let
          125|     python = rosSelf.python;
             |              ^
          126|     libExt = self.stdenv.hostPlatform.extensions.sharedLibrary;
       Did you mean python3?
```
Fixing those typos (same thing in distro overrides) then gave:
```
       error: numpy-1.26.4 not supported for interpreter python2.7
```
Which indicates that Python 2 is being used in places it shouldn't be, instead of Python 3.

Running the full stack trace, the dependency order (for my application, anyway) appears to be:
`buildEnv (from this repo) -> ros-humble-nav2-bringup-1.1.17-r1 -> python3.12-ros-humble-launch-ros-0.19.8-r1 -> ros-humble-rclpy-3.3.15-r1 -> ros-humble-pybind11-vendor-2.4.2-r1 -> python2.7-pybind11-2.13.6`. Given that the error stems from `rclpy`, I would think that *any* ROS2 package which uses Python bindings is currently broken.

The root cause of the problem appears to be incorrect rosdistro files: A quick inspection reveals that a _large_ number of Python 3 only packages are still labelled as being python 2 packages (ie, using `pythonPackages` instead of `python3Packages`). [This search](https://github.com/search?q=repo%3Aros%2Frosdistro+path%3A%2F%5Erosdep%5C%2F%2F+%22nixos%3A+%5BpythonPackages.%22&type=code) seems to pick up most of them, but to properly fix you'd need to compare every `pythonPackages.*` listing against what's in `nixpkgs` to check whether or not it should be updated. A couple key ones I found with some quick searching:
- `argcomplete`
- `catkin-pkg`
- `pybind11` itself

Aside from the typos, which are easy to fix, I think this is probably entirely due to the incorrect rosdistro files, and will likely be fixed after they're updated and a new superflore generation is run.

Originally observed by @wentasah here: [538d757](https://github.com/lopsided98/nix-ros-overlay/commit/538d7578a8ab0951f48f06404e800248dde3a54b#commitcomment-150272436)

lopsided98, 2024-12-12:
Not your fault since I hadn't documented this anywhere until just now, but the `develop` branch is where I test changes that may change or break a lot of packages. It should not be relied on to always be working.

closed 2024-12-12 by lopsided98

lopsided98, 2024-12-13:
The only packages I can find that are relevant to ROS2 are black, pytest and pybind11. These are fixed in https://github.com/ros/rosdistro/pull/43895. For now, I have restored the `pythonPackages` override.

RandomSpaceship, 2024-12-13:
Thanks! If `develop` is unstable and potentially breaking, should `master` perhaps be the default branch instead, though? Happy to open a new issue if you'd like given that this one's original topic is resolved.

wentasah, 2024-12-13:
> If `develop` is unstable and potentially breaking, should `master` perhaps be the default branch instead, though?

This was discussed a bit here: https://github.com/lopsided98/nix-ros-overlay/issues/424#issuecomment-2212510812

RandomSpaceship, 2024-12-15:
That makes sense. Personally I think having a working default branch is more important, but I can see the merits to both arguments.